dresden-elektronik/deconz-rest-plugin #3325: Gateway resetting after relaunch

timpa82, 2020-09-27:
## Describe the question or issue you are having

Hi, I have an issue with deconz. After having it working for months the gateway (Phoscon app) is reset every time I stop and start the container.
It happened after a power failure and at the time power was back on I pulled lastest upgrades for deconz and Home Assistant using docker-containers, which ended up with .82 version of deconz. 
So I dont know wether there is a relation to latest version or power failure. But anyway it doesnt seem to save any settings. When i stop the deconz container and starting it, its acting as a new gateway, see the screenshots. It happens every time.
At first I couldnt find the 
The Home Assistant container is working as intended.

## Screenshots

[image]
[image]

## Environment

 - Host system: Intel NUC
 - Running method: Ubuntu server, Marthoc Docker container, Home Assistant with deconz integration (docker container)
 - Firmware version: 26580700
 - deCONZ version: 2.05.82
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs

Right now I can provide logs from Portainer when starting the deconz container.
Log from Portainer trying to restart the deconz container: https://pastebin.com/VGAVYfFC
Log from Portainer after rebooting the NUC and restarting the containers: https://pastebin.com/AQRNN5AK
docker-compose.yaml (haven't changed anything): https://pastebin.com/GsAcxrM6
## Additional context

Mimiix, 2020-09-28:
@timpa82 Can you try to flash the conbee again? Following this guide:  https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Use windows or anything else. Just not use a VM/Docker install. Native is the best.

timpa82, 2020-09-28:
> 
> 
> @timpa82 Can you try to flash the conbee again? Following this guide: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually
> 
> Use windows or anything else. Just not use a VM/Docker install. Native is the best.

Hi again. I first tried flashing through the marthoc way "Updating Conbee/RaspBee Firmware", with no difference.
I then flashed it through "Update in Windows" guide in the link. But the same problem persists, the added devices are gone when I take down the docker container and back up again. However this time it did save the username I gave it "deconz" and asked for password during login, which was wrong password (which it isnt). Then I hit the back button and click on the usb stick icon again, and then Im prompted the reset page with chosing a username and password, like the way it looks first time you start it.

Mimiix, 2020-09-28:
I really have no clue why it wouldn't save. This isn't something i've seen before with deCONZ. 

However, since you are saying Home-assistant: Are you using the Addon too? If so: is that running?

timpa82, 2020-09-28:
I am using the deconz integration. The devices are showing as unavailable. Worth degrading deconz from .82? I haven't tried that yet.

Mimiix, 2020-09-28:
The integration is not relevant here, only the addon or docker. 

I have no clue why it should not save. Could be a permission thing, or files that are corrupt.

closed 2020-11-01 by stale[bot]