dresden-elektronik/deconz-rest-plugin #3327: Soil moisture sensor compatible with Conbee ?

durandm70, 2020-09-27:
## Describe the question or issue you are having
Hi everyone,
I am looking for a soil moisture sensor compatible with the Conbee hub.
I found one which seems to be Zigbee compatible (https://spruceirrigation.com/sensors/) but it seems to be sold only in North America (I am from France).
Any idea (even DIY...) would be much appreciated.

## Screenshots
N/A

## Environment
 - Host system: NAS
 - Running method: Marthoc Docker container
 - Firmware version: 26490700
 - deCONZ version: 2.05.80
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs
N/A

## Additional context
Gardening.

wvuyk, 2020-10-19:
Maybe china? Check here: https://dutch.alibaba.com/g/zigbee-soil-moisture-sensor.html, it is the dutch site, but I am sure yo'll find the French site as well on this. They are not cheap, and you probably need to get support here for adding them.

rohrsh, 2022-05-16:
Mark me down as a +1 wishing this Spruce soil sensor was supported by Deconz. I still have 2 units if I can help with testing in any way.

rohrsh, 2022-07-26:
Good call. I have submitted a device request. 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6222