dresden-elektronik/deconz-rest-plugin #3328: IKEA five button remote multiple issues after pairing

tubalainen, 2020-09-27:
## Describe the bug

## Steps to reproduce the behavior

- Pairing of remote works as expected (adding a switch in Phoscon)
- Renaming the newly added switch (remote control)
- After renaming - BUG - The new name does not apply to the API - Restart of deconz needed.
- After restart of deconz - BUG - The right deconz_event is being sent to Home Assistant (in my case) but the automation does not trigger even if the "debug" clearly states the event data id is the new name I´ve set.
- There isnt any support for long press, long release and button hold. Just "short release" (x002) - Only x002 shows up in both home assistant and deconz logs.
- After several restarts of deconz and home assistant - it suddenly kicks to life and works, the deconz_event triggers and the automation just works - NOTHING changed, just restarts.

I have repeated this with three remotes (switches), IKEA on/off and IKEA 5 button round´s - all exactly the same behavior as above.

## Expected behavior

1.) The renamed entity (the remote control / switch) in this case should not require a restart - if a restart is needed, please prompt the user to do so.
2.) The functions should work, long press, short press and button hold - and no multiple restarts should be needed for the functionality to "kick in".

## Environment

 - Host system: Debian 9 under ESXi VM
 - Running method: Marthoc Docker container
 - Firmware version: (26370500)
 - deCONZ version: (.82)
 - Device: ConBee I
 - Do you use an USB extension cable: no

SAOPP, 2020-09-27:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2618#issuecomment-649822696

Kane610, 2020-09-27:
This looks like a mess of hass related and deconz related issues 😣

Mimiix, 2020-09-28:
@tubalainen Thanks for the extensive bug report. Could you make a different bug report for the renaming stuff? Now there's 2 things messed up in 1 issue.

Mimiix, 2020-09-28:
@tubalainen Ehh?

tubalainen, 2020-09-28:
> @tubalainen Ehh?

Do not know what happened here, can you please reopen? Did not mean to close it. Must have been a slip of a button. My bad.

Mimiix, 2020-09-28:
@tubalainen No problem 👍

tubalainen, 2020-09-28:
Will split it soon as you suggested. :)

tubalainen, 2020-09-28:
@Mimiix Here Dennis, the new renaming issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3332

MarcoStein, 2020-09-28:
I have a similar problem with the IKEA 5 button remote, but specifically in conjunction with the left and right buttons of the tradfri remote. I cannot use them to manage the IKEA lamps white tone (colder .. warmer)

tubalainen, 2020-09-28:
> I have a similar problem with the IKEA 5 button remote, but specifically in conjunction with the left and right buttons of the tradfri remote. I cannot use them to manage the IKEA lamps white tone (colder .. warmer)

I had a similar problem in my bed room. One remote to control two lights. Dont remember how I solved it. I used one of the options below. :)
1.) Try to add the lights in question to a Phoscon group with the remote control and it should work.
2.) If that does not work, first join the remote to Phoscon - Then join it to the light (being very close to the bulb and hold down the join button on the remote).
[image]

closed 2020-11-01 by stale[bot]