dresden-elektronik/deconz-rest-plugin #3330: Turn lights on/off using traditional wall switches

axlthorell, 2020-09-28:
## Feature request type

Use old school switches to turn lights on and off via power cycling.

## Description

If the Power On/Off state of a light was set to opposite of last scene the use of a traditional switch to cycle the power to the light would make it possible to turn the light on and off. If the switch is NC with a spring to only open the circuit for a short moment the light looses power and when power is restored (release of button) it takes the new power state.

## Considered alternatives
It's already possible to set the light to Power On/Off state as On, Off or Previous. These three options are not the solution.
- Would it be possible to have deconz alter the default state between On and Off every time the light is powered on?
- Does Zigbee support the feature of the state Opposite-previous?

ebaauw, 2020-09-28:
No; no.  You really want to be using wireless switches for this kind of functionality.

You might try a series of gateway rules that toggle the light on a _Device Announcement_ (using a `dx` condition on `lastannounced` and a CLIPGenericFlag sensor to remember the previous light state).  I haven’t stress tested `lastannounced`, but initial tests are hopeful, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2590#issuecomment-643266440.  Note that the light will send a _Device Announcement_ on every firmware boot / reset, not just on a power cycle, so this could lead to some ghost behaviour.

discodogge, 2020-10-06:
@ebaauw 
Do you know if the IKEA lights can be configured to power on like the Philips ones after you power toggle twice? I have mine set to previous and that works a bit too well. If the gateway is offline I can't turn the lights on again this is of course by my own choosing but it feels a bit safer the way Philips does it. That could also be a workaround for @axlthorell I have tried comparing the cluster configs but can't find the answer myself. I still have lots to learn about the app

ebaauw, 2020-10-06:
Dunno.  The Trådfri lights I have expose the poweron attributes, but don’t seem to have implemented them.

discodogge, 2020-10-07:
@ebaauw wait are you talking about 0x4003 powerOn onoff? The one you can set to on or off or previous? That one works with the latest firmware, i have been able to select previous on almost all of them and they take that state when power returns. So that one works now as expected. some 1gen candlestick e14 doesn't work but everyone else does. But that is where they differ from hue. If a hue bulb is set to previous and was off when it loses power it returns when power comes back in the off state. But if I via the wall switch toggles the power on and off 2 times it will change to state on. Where the Ikea bulbs just remain in the off state until your gw is back online and you change there. or you reset them :) 
Or maybe its one of the others you are referring to? The description in some of them aren't crystal clear to me. 
I haven't figured out which one should be the one to configer just for this scenario.

ebaauw, 2020-10-07:
> That one works with the latest firmware

Cool.  I need to check my lights (I only use them for testing), but they’re rather old, and I don’t think IKEA has released new firmware for all older models.

We reverse engineered the poweron attributes, they’re not mentioned in the publicly available ZHA or ZLL specs.  Afaik the turning on to default value after two power cycles in quick succession is a feature exclusive to the Hue firmware, irrespective of the poweron setting.

discodogge, 2020-10-08:
Ah ok. You are doing a fantastic Jobb just so that's said. Too bad it's hue specific but also something to take into account going forward.
Thanks for taking the time.

ebaauw, 2020-10-13:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2590#issuecomment-643266440.

marmitol, 2021-09-18:
@ebaauw 
The Hue bulb Model LTA001 works properly, until you turn the lights on and off with traditional wall switches. At that point I have to reboot the server to stop the Hue bulb Model LTA001 from flashing.
[image]
[image]

Homebridge Hue Configuration.

            "name": "Hue",
            "anyOn": true,
            "effects": true,
            "excludeSensorTypes": [
                "Geofence",
                "CLIP"
            ],
            "forceEveWeather": true,
            "groups": false,
            "group0": false,
            "heartrate": 5,
            "hosts": [
                "xxx.xxx.xxx.xxx:xx"
            ],
            "hueDimmerRepeat": true,
            "hueMotionTemperatureHistory": true,
            "lights": true,
            "linkButton": false,
            "lowBattery": 15,
            "nativeHomeKitLights": false,
            "nativeHomeKitSensors": true,
            "parallelRequests": 3,
            "resource": true,
            "rules": false,
            "sensors": true,
            "timeout": 5,
            "users": {
                "xxxxxxxxxxxxxxxx": "xxxxxxxxxxxxxxx"
            },
            "waitTimeResend": 300,
            "waitTimeUpdate": 20,
            **"wallSwitch": true,**
            "platform": "Hue"
        },

I have other hue bulbs and I don't have this problem. Please can you indicate a solution for the flashing bulb problem. Thank you very much.