dresden-elektronik/deconz-rest-plugin #3331: Phoscon group ends up as RGBWWZ

Plantje1978, 2020-09-28:
## Describe the question or issue you are having
I see IKEA Trädfri lights being correctly added to Phoscon. I can add them to a group there as well and dim the individual lights or the whole group. These are normal dimmable Trädfri lights, so no RGB nor WW.
If I then lookup these devices in Domoticz the group is interpreted as RGBWWZ group.

## Screenshots
[image]
[image]

## Environment
Version 2.05.77 / 22-5-2020
Firmware 26350500

 - Host system: Windows 10 PC
 - Running method: Windows (non virtual)
 - Firmware version: 26350500
 - deCONZ version:  2.05.77
 - Device: ConBee I
 - Do you use an USB extension cable: yes -- only relevant for ConBee I/II

### deCONZ Logs

## Additional context

Smanar, 2020-09-28:
Wrong github ^^
It s not deconz dependant but from domoticz, better use https://github.com/Smanar/Domoticz-deCONZ

I have answered too on domoticz forum.

Plantje1978, 2020-09-28:
Ah, I see, my bad. Sorry. Since you already responded on the Domoticz forum, there is no need to create a topic on your Github site

Mimiix, 2020-09-28:
So this can be closed 👍

closed 2020-09-28 by Mimiix