dresden-elektronik/deconz-rest-plugin #3333: Installation of deconz in OpenHab: Please keep special config

openschwall, 2020-09-28:
Hi,
when updating deconz in openhab, the updating processing resets my config to default  values.
However, due to a conflict with the sonos binding I need to alter the port of the deconz binding and deactivate upnp in /lib/systemd/system/deconz.service and /lib/systemd/system/deconz-gui.service after each major update by manually editing a specific line:
…
[Service]
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=8007 –upnp=0
…
Markus Storm as maintainer of openhabian told me that openHABian does not change these files.  I should better contact the maintainers of the deconz service package. "You might be able to use an override file like we do for the openhab2 service to get around that."

Would it be possible to keep these line between updates?

Regards,
Peter

Smanar, 2020-09-28:
Perhaps same use than here ? Try to use a override.conf file ?

openschwall, 2020-09-28:
yes, but according to Markus Storm and as far as I understand this matter the overriding file must be provided by the standard install procedure of deconz. Openhab does  not override the original config file but deconz.

Smanar, 2020-09-28:
It s for that I m speaking about the override.conf file, this file over ride the deconz config file, so event if deconz change the setting in the deconz-gui.service file or reset them, it don't change the setting in override.conf, so you can have your own setting and those one have priority on deconz one.

Mimiix, 2020-09-28:
@openschwall I've already requested to get in contact with their maintainer: https://github.com/openhab/openhab-addons/issues/8379

Could you please ask him to join our discord? This isn't something to be discussed here.