dresden-elektronik/deconz-rest-plugin #3336: After Update Firmware on Raspbee II - no communication

weider1, 2020-09-30:
Hello everybody,

after updating the Raspbee II with firmware deCONZ_RaspBeeII_0x26650700.bin.GCF.md5. Every light is accessable and online. But it wont turn on/off any lights / No motion detected via motion detectors etc.

Already rebooted, reinstalled....

after deleting a "old" light and Searching for new lights, the gateway found it and it´s possible to turn on/off...

Smanar, 2020-09-30:
So you have solved the problem ?

weider1, 2020-09-30:
I had to relearn all devices (38)
Manually.

Thought there would be an easier solution, unfortunately not.

Smanar, 2020-09-30:
You haven't try the hidden config ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues
Or just a backup ?

Mimiix, 2020-10-01:
Closing this issue. Please use the Issue templates.

closed 2020-10-01 by Mimiix