dresden-elektronik/deconz-rest-plugin #3339: Konke Kit Pro Multi Function Button

Doreinie, 2020-10-01:
Konke Kit Pro Multi Function Button

- Product name: Kit Pro Multi Function Button IC 23777-KPKEY
- Manufacturer: Konke.
- Model identifier: 3AFE280100510001
- Device type : - Switch 

## Screenshots
[image]
[image]
[image]
[image]
[image]

I hope this request is now accepbtable. Not that easy for a newbie to get this done. but Yeah! got so far and hope to have the buttons added some time soon. 
Thanks

Mimiix, 2020-10-01:
@SwoopX Can you check if this is correct?

Doreinie, 2020-10-11:
@Mimiix what is the netx step? Are you taking this item on as a candidate into deconz / Phoscon?

SwoopX, 2020-10-11:
Info should be complete from what I can tell. I can't take care of that since I got tons of local PRs waiting with significant dependencies.

Mimiix, 2020-10-12:
@Doreinie We need people like Swoop to integrate it. Or you have to do it yourself. They do this in their spare time so i really can't obligate them to do it.

Doreinie, 2020-10-12:
@Mimiix is there any sample on what is needed to do to integrate the Konke item by myself. Then I can try it

Mimiix, 2020-10-12:
@SwoopX Is best to explain that.

SwoopX, 2020-10-13:
What makes you think that?

Erik made a nice short overview on this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2061#issuecomment-568191933
However, point 2 and 3 are not necessary and more and got simplified by the entries in `button_maps.json`.

closed 2020-11-24 by stale[bot]

opechunka, 2021-08-29:
Any updates? It seems like it's not supported yet by the latest deconz.

rfilmyer, 2022-01-27:
This button is supported by [Zigbee2MQTT](https://www.zigbee2mqtt.io/devices/2AJZ4KPKEY.html), which may be useful to deconz as a possible reference implementation for the protocol. One complication to note is that Konke devices only operate on channels 15, 20, and 25.

Doreinie, 2023-04-22:
Hi guys, Did not follow up my issue here as I was busy with other projects. 
I still have 2 of these switches and would like to revisit the topic to get them work with deconz.
Maybe someone can help me or guide me how to do this?
thanks!

Mimiix, 2023-04-22:
> Hi guys, did not follow up my issue here as I was busy with other projects. I still have 2 of these switches and would like to revisit the topic to get them work with deconz. Maybe someone can help me or guide me how to do this? thanks!

It's been 3 years.

@Smanar can you help out ?

Doreinie, 2023-04-24:
yes
@Mimiix, it has been a while. Those days back, I was not able to figure out how to do this by myself, finally ran out of time. 
Maybe, now it is a bit easier? I´m a bit more patient with myself in learning how to get the switches integrated. 
With some help and guidance from @Smanar ?

Smanar, 2023-04-25:
Can try this DDF 
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Konke",
  "modelid": "3AFE280100510001",
  "product": "Konke Kit Pro Multi Function Button",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "awake": true,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
            "eval": "const vmin = 20; const vmax = 30; var bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 3600,
          "max": 3600
        }
      ]
    }
  ]
}
```

Then include the device, go in deconz logs (help/Debug view) enable flag "info" + "info_l2', and press button, you will have error message about missing button map, I just need them with the pressed button.

Doreinie, 2023-04-26:
Tried it:

Get this result in the browser.

I do not get any switch result in ioBroker when pressing the button.

Product name does not update.

Debug.txt file attached

 

18:43:49:080 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
18:43:49:104 /home/pi/.local/share/dresden-elektronik/deCONZ/config.ini exists and is writeable
18:43:49:104 /home/pi/.local/share/dresden-elektronik/deCONZ/zll.db exists and is writeable
18:43:49:279 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
18:43:49:279 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
18:43:49:490 DDF enabled for Gold status
18:43:50:112 parent process -bash
18:43:50:112 gw run mode: normal
18:43:50:112 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
18:43:50:113 DB sqlite version 3.27.2
18:43:50:115 DB PRAGMA page_count: 82
18:43:50:115 DB PRAGMA page_size: 4096
18:43:50:115 DB PRAGMA freelist_count: 5
18:43:50:115 DB file size 335872 bytes, free pages 5
18:43:50:115 DB PRAGMA user_version: 9
18:43:50:115 DB cleanup
18:43:50:116 DB create temporary views
18:43:50:117 DB view [0] created
18:43:50:117 DB view [1] created
18:43:50:117 DB view [2] created
18:43:50:118 DB view [3] created
18:43:50:118 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
18:43:50:163 sql exec SELECT key FROM config2
18:43:50:163 sql exec SELECT key,value FROM config2
18:43:50:163 Load config UTC: 2023-04-24T21:10:53 from db.
18:43:50:163 Load config announceinterval: 45 from db.
18:43:50:163 Load config announceurl: https://phoscon.de/discover from db.
18:43:50:163 Load config apiversion: 2.21.02 from db.
18:43:50:163 Load config bridgeid: 00212EFFFF050A2B from db.
18:43:50:163 Load config datastoreversion: 93 from db.
18:43:50:163 Load config dhcp: true from db.
18:43:50:163 Load config discovery: false from db.
18:43:50:163 Load config factorynew: false from db.
18:43:50:164 Load config fwneedupdate: false from db.
18:43:50:164 Load config fwupdatestate: idle from db.
18:43:50:164 Load config fwversion: 0x26780700 from db.
18:43:50:164 Load config gateway: 127.0.0.1 from db.
18:43:50:164 Load config group0: 65520 from db.
18:43:50:164 Load config groupdelay: 50 from db.
18:43:50:164 Load config gwpassword: $1$8282jdkm$lFVtkO9W8v1MOOfW6TQmH. from db.
18:43:50:164 Load config gwusername: delight from db.
18:43:50:164 Load config homebridge: not-managed from db.
18:43:50:164 Load config homebridge-pin:  from db.
18:43:50:164 Load config homebridgepin:  from db.
18:43:50:164 Load config homebridgeupdate: false from db.
18:43:50:164 Load config homebridgeupdateversion:  from db.
18:43:50:164 Load config homebridgeversion:  from db.
18:43:50:164 Load config ipaddress: 192.168.178.38 from db.
18:43:50:164 Load config linkbutton: false from db.
18:43:50:164 Load config localtime: 2023-04-24T23:10:53 from db.
18:43:50:164 Load config mac: 38:60:77:7c:53:18 from db.
18:43:50:164 Load config modelid: deCONZ from db.
18:43:50:164 Load config name: Phoscon-GW from db.
18:43:50:164 Load config netmask: 255.0.0.0 from db.
18:43:50:164 Load config networkopenduration: 180 from db.
18:43:50:164 Load config otauactive: true from db.
18:43:50:164 Load config otaustate: off from db.
18:43:50:164 Load config panid: 0 from db.
[... 902 lines cut ...]
18:43:50:253 Sqlite sensors: uniqueid = 00:15:8d:00:04:5e:b7:aa-01-0400
18:43:50:253 Sqlite sensors: swversion = 3000-0001
18:43:50:253 Sqlite sensors: state = {"dark":false,"daylight":false,"lastupdated":"2023-04-26T11:09:00.256","lightlevel":18866,"lux":77}
18:43:50:253 Sqlite sensors: config = {"battery":95,"on":true,"reachable":false,"tholddark":12000,"tholdoffset":7000}
18:43:50:254 Sqlite sensors: fingerprint = {"d":263,"ep":1,"in":[1024],"p":260}
18:43:50:254 Sqlite sensors: deletedState = normal
18:43:50:254 Sqlite sensors: mode = 1
18:43:50:254 Sqlite sensors: lastseen = 2023-04-26T16:40Z
18:43:50:254 DB found sensor Motion Sensor Wohnzimmer 1 23
18:43:50:254 DB skip loading sensor Motion Sensor Wohnzimmer 1 23, handled by DDF Aqara motion sensor RTCGQ11LM
18:43:50:254 Sqlite sensors: sid = 20
18:43:50:254 Sqlite sensors: name = Motion Sensor Küche Vorrat
18:43:50:254 Sqlite sensors: type = ZHAPresence
18:43:50:255 Sqlite sensors: modelid = lumi.sensor_motion.aq2
18:43:50:255 Sqlite sensors: manufacturername = LUMI
18:43:50:255 Sqlite sensors: uniqueid = 00:15:8d:00:04:5f:b7:ea-01-0406
18:43:50:255 Sqlite sensors: swversion = 3000-0001
18:43:50:255 Sqlite sensors: state = {"lastupdated":"2023-04-26T14:17:39.988","presence":false}
18:43:50:255 Sqlite sensors: config = {"battery":100,"duration":90,"on":true,"reachable":false,"temperature":2600}
18:43:50:255 Sqlite sensors: fingerprint = {"d":263,"ep":1,"in":[1030],"p":260}
18:43:50:255 Sqlite sensors: deletedState = normal
18:43:50:255 Sqlite sensors: mode = 1
18:43:50:255 Sqlite sensors: lastseen = 2023-04-26T16:40Z
18:43:50:255 DB found sensor Motion Sensor Küche Vorrat 20
18:43:50:25

Mimiix, 2023-04-26:
Can you remove this logs and please use pastebin for these large logs?

Doreinie, 2023-04-26:
Added the DDF:
Debug.txt in
https://pastebin.com/QPHVpwL4

Smanar, 2023-04-27:
```
18:44:12:172 [INFO] - No button map for: 3AFE280100510001, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001080, zclSeq: 102
18:44:48:553 [INFO] - No button map for: 3AFE280100510001, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001082, zclSeq: 106
18:45:05:767 [INFO] - No button map for: 3AFE280100510001, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001081, zclSeq: 107
```

Now the hard part, will be not possible if you have HA.

Need to add in the button_maps.json file.

```
        "konkeMap": {
            "vendor": "Konke",
            "modelids": "3AFE280100510001",
            "map": [
                [1, "0x01", "ONOFF", "ATTRIBUTE_REPORT", "0x80", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press 1"],
                [1, "0x01", "ONOFF", "ATTRIBUTE_REPORT", "0x81", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press 2"],
                [1, "0x01", "ONOFF", "ATTRIBUTE_REPORT", "0x82", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press 3"]
            ]
        },
```

But I have a doubt on the payload, I think I need the full one and not just 0x08

Can you try this DDF, your device is simple, I think it's possible without buttonmap

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Konke",
  "modelid": "3AFE280100510001",
  "product": "Konke Kit Pro Multi Function Button",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "awake": true,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
            "eval": "const vmin = 20; const vmax = 30; var bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0x0006",
            "cmd": "0x0A",
            "eval": "Item.val=(ZclFrame.at(4)-121)*1000+2"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 3600,
          "max": 3600
        }
      ]
    }
  ]
}

`payload: 00001081`
`"eval": "Item.val=(ZclFrame.at(4)-121)*1000+2"`

I m trying to convert the 81 to "1002" value

Doreinie, 2023-04-30:
Thank you for the guidance I hope I did this all correctly.
a) replaced the DDF with recent provided
b) added info to the button_maps.json file
[image]
c) I was not sure if I have  to add the payload info anywhere,
I do not see any status change e.g. in iobroker when pression a button.
[image]
d) what do you mean with HA which would be bad?
Do you need another log?
thanks

Smanar, 2023-05-01:
>what do you mean with HA which would be bad?

It's realy hard to edit the buttonmap file on HA because of the docker use.
How have you done that ? You have direclty edited the file on the docker ?
And with the last DDF you don't need to edit the buttonamp file ^^

Can you just enable log "info" "info_l2" "DDF" and press device button with the new DDF ?

Doreinie, 2023-05-01:
I´m not using docker. I have installed this directly on rasp pi 4b. 
now I think that with "HA" you mean "home assistant". I´m not using that. I use ioBroker.
So I edit the file using "nano" or file editor.
I did edit the DDF and the button_map file. This is how I understood the instructions. 
Is that wrong, only to change the DDF an not the button map file?
log pastebin here: https://pastebin.com/w7fLRQX6
thank you
Interesting is that the battery of the switches moved from green to red over night.

[image]

Smanar, 2023-05-01:
Ha nice ^^.
So yes it's easy on your OS, but this device use only 3 events and all on same command so easier using only DDF too

```
15:09:08:041 APS-DATA.indication from child 0x01EC
15:09:08:051 failed to evaluate expression for 14:b4:57:ff:fe:09:7c:cf-01-0006/state/buttonevent: Item.val=(ZclFrame.at(4)-121)*1000+2, err: RangeError: index out of range
15:09:08:056 [INFO] - No button map for: 3AFE280100510001, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001080, zclSeq: 13
```

So the buttonamp haven't worked, but I think we can avoid them
And the javascript expression have failed, I think it's because the first byte is 0 and not 1
`payload: 00001080`
0 > 00
1 > 00
2 > 10
3 > 80

So the code need to be 

`"eval": "Item.val=(ZclFrame.at(3)-121)*1000+2"`

Doreinie, 2023-05-01:
thanks, now I´m getting a button response.
in iobroker I see
single click response= 7002
double click response = 8002
long click upon release = 7002
little log: https://pastebin.com/tJNUbaWf

I found this info about this switch, hence maybe there is a way to capture the long hold?  https://www.zigbee2mqtt.io/devices/2AJZ4KPKEY.html

Smanar, 2023-05-02:
I don't have impact on phoscon, so you need to ask phoscon devs.

>Is there a possibility to capture as well the "buttonpressed"?

From that I m reading this device have only 3 events.
And I m using bad one ^^ https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#events-legend

Need to have

> single click response= 1002
> double click response = 1004
> long clic response = 1003

Can you try this eval code ?
`"eval": "const t={'0':2,'1':4,'2':3};if(ZclFrame.at(3) in t){Item.val=1000+t[(ZclFrame.at(3)-128)]}"`

Doreinie, 2023-05-02:
This is working. Great!
single click response= 1002
double click response = 1004
long click response = 1003
I have tried to see if there is any other output for mutiple button presses, but there seems nothing. It goes that far that if I do 3 or more clicks the entire input is going to be ignored.

Do you know who to connect with to show in the web app the product info, switch status?

[image]

thank you!

Doreinie, 2023-05-02:
Yes, it seems we have max 3 actions and we got this working. Thank you!

To have it nice and neat it would be nice to have this more complete implemented in the phoscon web app and mobile app.
I thought that this is done through the DDF but seems not. 

Functionality in iobroker is available and all other stuff in phoscon would just be a plus.

Smanar, 2023-05-03:
Do you want to make the PR or want I make it to include this device officialy ?
And yes, I understand, but phoscon is a closed application and IDK what happen for them.

Smanar, 2023-05-04:
You have it here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6946

Doreinie, 2023-05-07:
Thank you for taking this forward