dresden-elektronik/deconz-rest-plugin #3341: Conbee 2 crashing and switching ttyACM's

Goldwing1973, 2020-10-01:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: Raspberry Pi4 4gb and NUC
 - Running method: (Hassbian / Debian 10 / Home Assistent deCONZ Add-on / Virtual Machine vmware ESXI)
 - Firmware version: (26580700)
 - deCONZ version: (2.05.81)
 - Device: ConBee II (2x)
 - Do you use an USB extension cable: yes and no (tried both)
### deCONZ Logs

## Additional context

N5A, 2020-10-04:
Use the By-ID line for it instead, see if that helps. At least then if it drops in the night and comes back the connection won't change.

I couldn't even keep the CB2 stick stable on Win10 with a HASSOS VM, vbox passing it through. Had to move it to a Pi, which did allow me to better centralize its location to handle the house lights more reliably. It would connect/disconnect every few seconds. Which if you know windows and its tendency to chirp on that event.. will drive you bonkers quickly.

Goldwing1973, 2020-10-04:
> 
> 
> Use the By-ID line for it instead, see if that helps. At least then if it drops in the night and comes back the connection won't change.
> 
I tried that already, but then problem then is that the Deconz integration in Home Assistant needs to be restarted before working again.
I Also tried it on a Pi4 4gb, same problems.

I think i'll have to try some scripting, tail the log for "HTTP request response: [{'error': {'address': '/groups/2/action','description': 'Not connected' " and when it happens, 30 secs later give the deconz integration a restart command

But still, this isn't the way is supposed to be to keep it working.

kmplngj, 2020-10-05:
I have the same problem since some months. I had my Conbee II running 12 months without problems (EXSi Host, Fedora Guest, Deconz in Docker). Since some update the stick is unstable. Sometimes it is disconnected after some days, often after ~ 1 day, sometime afters some minutes. I always have to restart the VM running Deconz.

I also have many devices in the network but this should not be a problem. Since the Conbee II supports up to 200 devices.

I hope the devs could address the problem in the near future. I think it should be top priority.

Mimiix, 2020-10-05:
@kmplngj Please open a own issue.

@Goldwing1973 ESXI and after that Docker. Might be something off there. However, i am seeing a trend: Users reporting this behaviour almost **always** use that Z-wave stick.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3318
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1598#issuecomment-507020400

Also had some cases on Discord.

SwoopX, 2020-10-05:
If you feel funny, you can nail the Conbee to a dedicated tty symlink by adding a udev rule and insert

```
KERNEL=="ttyACM?", SUBSYSTEMS=="usb", DRIVERS=="usb", ACTION=="add", ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", SYMLINK+="ttyConBee2"
```

SaulGoodman1337, 2020-11-29:
same issue here

Goldwing1973, 2020-11-29:
Please reopen the issue, as it isn't solved.
Updated to the latest Deconz (2.05.88 / 15-10-2020) en latest firmware (26660700) and the ttyACM switching is still happening.

tkerstan, 2020-12-11:
I am having the same issue with my Conbee II. The device continuously disconnects every minute on Ubuntu 20.04 LTS. Everything worked fine until I updated the docker container from marthoc. After the update I got a segmentation fault when I started the container and the phoscon app wasn't working anymore. I then decided to install the native tools. Same result. Update to newest firmware didn't work either. Stick works fine in windows 10, but my home assistant installation is currently out of order. Any suggestions?

Mimiix, 2020-12-14:
You should select the full serial of the device in the HA addon config (not the TTYxxx).

@tkerstan you should open a seperate issuerequest.

Goldwing1973, 2020-12-14:
> 
> 
> You should select the full serial of the device in the HA addon config (not the TTYxxx).

I use /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2195XXX-xxxx in the HA addon config.
But when it "hops" ttyACM the addon doesn't see the Conbee2 anymore, only way to get it working again is restarting the addon.

dvbit, 2021-01-03:
after updating to the last firmware version I also have this issue.
Issue is NOT closed.

dvbit, 2021-01-03:
reopen this. Issue is still happening. Flashing is not possible with "jumping" port

Mimiix, 2021-01-03:
@dvbit I am not reopening on command.

Either way, flashing isn't possible on HassOS at all.

dvbit, 2021-01-04:
Nobody is asking to reopen on command .
Just supporting your product and honoring the warranty.
I have read ALL related GitHub and Reddit threads.
And applied all suggestions :
Other pc
No usb 3
No other devices connected
Usb cord extension 

What is left is trying to flash again from windows 10 and flash tool with timeout option .

This is NOT a rare issue and being not rare it should be either fixed OR mentioned as a known issue together with workaround (if they exist).

A very disappointed customer

Mimiix, 2021-01-04:
@dvbit this is not end user support.

Please open your own issue and lets have a look at your problem then. Maybe something else went wrong.

dvbit, 2021-01-04:
Thank You for the feedback.
I will open a issue with full detail and a ticket .
Could you kindly point me at the correct procedure to open a support ticket.
Thank you
Kind regards

Goldwing1973, 2021-01-04:
@Mimiix Please reopen, this issue still isn't resolved.

Mimiix, 2021-01-04:
@Goldwing1973 I am  happy to re-open. But i am wondering why you've let the Stale bot close it. It was inactive for a while, so i am uncertain where we are at right now.

@SwoopX Can you pinch in and give me a pointer on what to do?

dvbit, 2021-01-04:
Thank You @Goldwing1973 I have the feeling (more then that having read almost all the related issues) that most of the issues have been autoclosed after being stale. Not because of solution.
For me the solution was to:
On a debian 10 laptop with extension for usb
-  completely  disable deconz software (gui etc)
- unplug the dongle
- use the manual flash tool to flash the first release of conbee II firmware
- after launching it, insert the dongle
- flash it
- reenable deconz software
- access the dongle and reset coordinator
- disable deconz software
- remove dongle
- flash with latest firmware

Hope this helps

Goldwing1973, 2021-01-04:
> @Goldwing1973 I am  happy to re-open. But i am wondering why you've let the Stale bot close it. It was inactive for a while, so i am uncertain where we are at right now.
> 
> 
> 
> @SwoopX Can you pinch in and give me a pointer on what to do? 
> 
> 
> 
> @dvbit Check the support page @ phoscon.de for the contact methods listed there. This git is community support and not a "official" support channel.
> 
> 

First, I'm fairly new to GitHub and didn't expect the closure by the stale bot to be so definite.
Also I didn't reply anymore because I didn't have any new information to be added and thought it would be spamming to add a useless comment just to prevent the stale bot to close it.

But thank you for reopening it.

Goldwing1973, 2021-01-04:
> Thank You @Goldwing1973 I have the feeling (more then that having read almost all the related issues) that some of the issues have been autoclosed after being stale. Not because of solution.
> 
> For me the solution was to:
> 
> On a debian 10 laptop with extension for usb
> 
> -  completely  disable deconz software (gui etc)
> 
> - unplug the dongle
> 
> use the manual flash tool to flash the first release of conbee II firmware:
> 
> - after launching it, insert the dongle
> 
> - flash it
> 
> - reenable deconz software
> 
> - access the dongle and reset coordinator
> 
> - disable deconz software
> 
> - remove dongle
> 
> - flash with latest firmware
> 
> 
> 
> Hope this helps

My problem is not upgrading the firmware, I have 2 Conbee 2 sticks (one for backup) and upgrading is done using a windows PC and the current firmware on the stick I'm using in HA now is 26660700.
The other (backup stick) is one firmware version lower.
Both have the same ttyACM hopping and both sticks were used in different setups (NUC with hassos, NUC with ESXi and hassos VM, RPI4 with hassos and (still using) NUC with ESXi with Debian 10 VM with HA)
I must admit that the ttyACM switching is less frequent on this setup but it still occurs and the only way to fix it is by restarting the Deconz integration in HA.

Mimiix, 2021-01-04:
@dvbit this is exactly how the manually flashing page describes it.  Either way: Please stick to a own issue and do not hijack other issues that are unrelated. 

> First, I'm fairly new to GitHub and didn't expect the closure by the stale bot to be so definite.
> Also I didn't reply anymore because I didn't have any new information to be added and thought it [...]

No worries, that's fine. 

I am wondering why it hops. I'll ask Manuel to check this issue.

manup, 2021-01-04:
When the device name changes it means that the firmware does re-enumerate the USB port but the old device name is still registered in the Linux. This can be reproduced by quickly unplug and reattach the ConBee (or any serial USB device).

On a native system this isn't a problem since deCONZ would just reconnect to the new device name.

With Docker it's not that easy since new devices names are not automatically available inside the container.
There is a solution for this using Docker cgroups as described here: https://github.com/marthoc/docker-deconz/issues/298

To my knowledge, the HassOS deCONZ Docker container currently doesn't support this, but will hopefully in future.

However, what is strange in your case is that once the ConBee is connected it should stay connected, given you're already using an USB extension cable.

Goldwing1973, 2021-01-04:
> 
> 
> When the device name changes it means that the firmware does re-enumerate the USB port but the old device name is still registered in the Linux. This can be reproduced by quickly unplug and reattach the ConBee (or any serial USB device).
> 
> On a native system this isn't a problem since deCONZ would just reconnect to the new device name.
> 
> With Docker it's not that easy since new devices names are not automatically available inside the container.
> There is a solution for this using Docker cgroups as described here: [marthoc/docker-deconz#298](https://github.com/marthoc/docker-deconz/issues/298)
> 
> To my knowledge, the HassOS deCONZ Docker container currently doesn't support this, but will hopefully in future.
> 
> However, what is strange in your case is that once the Conbee is connected it should stay connected, given you're already using an USB extension cable.

With the USB extension cable the hops were more frequent but after using a cheap AliExpress USB2.0 hub (with only the Conbee 2 connected) it is way less frequent.
What also puzzles me, i have a lot of Zigbee devices (currently over 75 connected and 50+ still to be connected) but the problems started when i exceeded the 45-50 devices (i checked/tested this also on the second Conbee 2, below 45 devices no problems, above that the ttyACM switching started).
That makes me think that the firmware crashes and restarts somehow and when i restarts it request a ttyACM, the computer still thinks that the Conbee 2 is connected and gives out a new ttyACM.
And right after that the first ttyACM is released because a device with the same device ID is connected to the new ttyACM

Also, maybe a strange thought, but could it be that a Zigbee device is sending some bad information to the Conbee 2 and that the Conbee 2 crashes because of it? (like the Conbee sends a "hello" to a light-bulb and expects "hi" back but receives a "yes")

manup, 2021-01-04:
The number of devices shouldn't cause crashes, I'm currently running latest version 0x26680700 in some larger networks with up to 170 nodes.

The firmware can't be crashed by incoming commands, it's actually very basic modem like and doesn't know anything about specific devices – this is all managed at a higher layer in deCONZ. deCONZ itself had some crashes in past versions which where fixed as stated in the related changelog, but this was before 2.71. I think a deCONZ crash or restart should also be visible in the HA logs?

There are indeed quality differences with USB extension cables and we have also seen that some USB ports work better than others.

Goldwing1973, 2021-01-05:
> The firmware can't be crashed by incoming commands, it's actually very basic modem like and doesn't know anything about specific devices – this is all managed at a higher layer in deCONZ. deCONZ [...]

I will post a HA log as soon as it occurs again.
In my first post i already posted a log [HERE](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5313678/log.txt) 

> 
> There are indeed quality differences with USB extension cables and we have also seen that some USB ports work better than others.

I figured that, that's why i tried it on 3 different computers (2 NUC's and 1 RPI4) and with extension cable, without extension cable and with HUB, but on all the ttyACM's are switching

adampetrovic, 2021-01-08:
I can confirm this is also happening for me. I’ve had my Conbee II plugged into a different NUC (Ubuntu 18.04 + docker-deconZ) for well over a year. Since purchasing a new NUC (20.04.1 + docker-second) and flashing to the latest firmware I’ve started seeming this problem. 

For what it’s with I have 60 devices in my Zigbee network.

Goldwing1973, 2021-01-27:
The last weeks it happened a couple of times again, only all at night, so i wasn't able to capture the logs.
Until today, it just happened again, and this cluttered my deconz log

20:12:43:492 0x7CB03EAA00AB6E00 error APSDE-DATA.confirm: 0xA7 on task
20:12:44:292 0x7CB03EAA00AF0554 error APSDE-DATA.confirm: 0xA7 on task
20:12:45:092 0x00158D00054F71BB error APSDE-DATA.confirm: 0xA7 on task
20:12:45:892 0x00158D0005669B42 error APSDE-DATA.confirm: 0xA7 on task
20:12:46:692 0x0017880102B946BE error APSDE-DATA.confirm: 0xA7 on task
20:12:47:492 0x0017880100F1B035 error APSDE-DATA.confirm: 0xA7 on task
20:12:48:292 0x0017880100F1A8E3 error APSDE-DATA.confirm: 0xA7 on task
20:12:49:092 0xBC33ACFFFE6D8247 error APSDE-DATA.confirm: 0xA7 on task
20:12:49:892 0xEEEE030201036912 error APSDE-DATA.confirm: 0xA7 on task
20:12:50:692 0x0017880104A2A7A7 error APSDE-DATA.confirm: 0xA7 on task
20:12:51:492 0x0017880100ED297D error APSDE-DATA.confirm: 0xA7 on task
20:12:52:292 0x7CB03EAA00AAB1BB error APSDE-DATA.confirm: 0xA7 on task
20:12:53:092 0x0017880100E077BA error APSDE-DATA.confirm: 0xA7 on task
20:12:53:892 0x0017880102520124 error APSDE-DATA.confirm: 0xA7 on task
20:12:54:692 0x0017880102858A14 error APSDE-DATA.confirm: 0xA7 on task
20:12:55:492 0x001788010285503C error APSDE-DATA.confirm: 0xA7 on task
20:12:56:292 0x0017880100D60683 error APSDE-DATA.confirm: 0xA7 on task
20:12:57:092 0xEEEE030201036A7D error APSDE-DATA.confirm: 0xA7 on task
20:12:57:892 0x0017880106FB3ADD error APSDE-DATA.confirm: 0xA7 on task
20:12:58:692 0x001788011055073F error APSDE-DATA.confirm: 0xA7 on task
20:12:59:492 0x0017880100F1B0A4 error APSDE-DATA.confirm: 0xA7 on task
20:13:00:292 0x0017880100F1AEDA error APSDE-DATA.confirm: 0xA7 on task
20:13:01:092 0x0017880100F1AE9E error APSDE-DATA.confirm: 0xA7 on task
20:13:01:893 0x00178801062A5D32 error APSDE-DATA.confirm: 0xA7 on task
20:13:02:932 0x588E81FFFEDB88CE error APSDE-DATA.confirm: 0xA7 on task
20:13:03:732 0x7CB03EAA00AB8741 error APSDE-DATA.confirm: 0xA7 on task
20:13:04:612 0x00178801062B2295 error APSDE-DATA.confirm: 0xA7 on task
20:13:05:412 0x0017880106427F6E error APSDE-DATA.confirm: 0xA7 on task
20:13:06:212 0x0017880100C19BB7 error APSDE-DATA.confirm: 0xA7 on task
20:13:07:012 0x00178801067F88D7 error APSDE-DATA.confirm: 0xA7 on task
20:13:07:812 0x842E14FFFE1397E9 error APSDE-DATA.confirm: 0xA7 on task
20:13:08:612 0x7CB03EAA0A080D33 error APSDE-DATA.confirm: 0xA7 on task
20:13:09:412 0x0017880100CCC9F1 error APSDE-DATA.confirm: 0xA7 on task
20:13:10:212 0x0017880100E11D28 error APSDE-DATA.confirm: 0xA7 on task
20:13:11:012 0x001788010371AC3F error APSDE-DATA.confirm: 0xA7 on task
20:13:11:812 0x842E14FFFEDFA394 error APSDE-DATA.confirm: 0xA7 on task
20:13:12:612 0x842E14FFFE1397E5 error APSDE-DATA.confirm: 0xA7 on task
20:13:13:412 0x0017880110599FA7 error APSDE-DATA.confirm: 0xA7 on task
20:13:14:212 0x00178801102AA7E0 error APSDE-DATA.confirm: 0xA7 on task
20:13:15:012 0x00178801102E8408 error APSDE-DATA.confirm: 0xA7 on task
20:13:15:812 0x00178801104BE2F7 error APSDE-DATA.confirm: 0xA7 on task
20:13:16:612 0x0017880104194D23 error APSDE-DATA.confirm: 0xA7 on task

Hundreds and hundreds of these errors, and after restarting Deconz everything worked again.

Mimiix, 2021-01-28:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log#0xa7-no-application-ack-received

Some routing issues i guess.

Goldwing1973, 2021-01-28:
> 
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log#0xa7-no-application-ack-received
> 
> Some routing issues i guess.

And can i fix this in any way?

And it look like the description i gave earlyer...
"Also, maybe a strange thought, but could it be that a Zigbee device is sending some bad information to the Conbee 2 and that the Conbee 2 crashes because of it? (like the Conbee sends a "hello" to a light-bulb and expects "hi" back but receives a "yes")"

Goldwing1973, 2021-01-29:
And it happened again, but this time i also get permission errors on ttyACM2.
After a manual restart of Deconz it works again.

[errorlog.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5895247/errorlog.txt)

Goldwing1973, 2021-02-21:
Today i noticed something other strange in de Supervisor log from HA

21-02-21 10:50:28 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-21 11:09:28 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE serial hardware /dev/ttyACM1 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2195308-if00
21-02-21 11:09:28 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE usb hardware /dev/bus/usb/002/107
21-02-21 11:09:33 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD usb hardware /dev/bus/usb/002/109
21-02-21 11:09:33 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD serial hardware /dev/ttyACM2 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2195308-if00
21-02-21 11:09:34 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD usb hardware /dev/bus/usb/002/109
21-02-21 11:09:34 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD serial hardware /dev/ttyACM2 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2195308-if00
21-02-21 11:17:41 INFO (MainThread) [supervisor.host.info] Updating local host information
21-02-21 11:17:41 INFO (MainThread) [supervisor.host.services] Updating service information
21-02-21 11:17:41 INFO (MainThread) [supervisor.host.network] Updating local network information
21-02-21 11:17:42 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-02-21 11:17:42 INFO (MainThread) [supervisor.host] Host information reload completed
21-02-21 11:18:13 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE serial hardware /dev/ttyACM2 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2195308-if00
21-02-21 11:18:13 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE usb hardware /dev/bus/usb/002/109
21-02-21 11:18:18 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD usb hardware /dev/bus/usb/002/111
21-02-21 11:18:19 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD serial hardware /dev/ttyACM1 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2195308-if00
21-02-21 11:18:20 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD usb hardware /dev/bus/usb/002/111
21-02-21 11:18:20 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD serial hardware /dev/ttyACM1 - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2195308-if00

I did NOT remove the Conbee II stick or was even close to it.

Mimiix, 2021-02-22:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log#0xa7-no-application-ack-received

I am wondering what happens if you use a normal OS (instead of HassOS). Is there a way to test that?

Goldwing1973, 2021-02-22:
@Mimiix HA is running on Debian GNU/Linux 10 (buster) and not on HassOS.

Mimiix, 2021-02-22:
So why is the supervisor detecting removals? Without HassOS (supervised install) it doesn't see that.

Goldwing1973, 2021-02-22:
This is a supervised install on Debian 10

Goldwing1973, 2021-02-22:
[image]

Goldwing1973, 2021-02-22:
But i had these problems also with HassOS on a NUC, HassOSS in ESXi and on a Pi4 4GB
On the Debian 10 installation it doesn't happen as much as on the other installations.
I even bought a second Conbee II to rule out a defect in the Conbee II stick

Mimiix, 2021-02-22:
As long as we can't rule out HassOS / Addon /virtualisation involvement in terms of the running method we are shooting in the dark.

Goldwing1973, 2021-02-22:
@Mimiix So, what do you suggest?
I have all the hardware needed to perform testing, i have the know-how to set things up and the time to do it.

Mimiix, 2021-02-23:
Setup a native  with Debian or ubuntu and install deconz on there. No virtual stuff.

deviantintegral, 2021-03-04:
I am having similar problems that seem to be getting worse over time. In my case, the `/dev/ttyACM*` device isn't changing.

Normally, I pass the USB device through into a kvm machine running Home Assistant. To test, I:

1. Shut down the VM and turned off `autostart`.
2. Restarted the host.
3. Noted multiple resets of the USB device on the host via `dmesg`:

```
[   83.358519] usb 1-1.2: USB disconnect, device number 7
[   83.581366] usb 1-1.2: new full-speed USB device number 8 using ehci-pci
[   83.691518] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[   83.691520] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   83.691521] usb 1-1.2: Product: ConBee II
[   83.691522] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[   83.691522] usb 1-1.2: SerialNumber: DE2234239
[   83.691815] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[   87.197115] usb 1-1.2: USB disconnect, device number 8
[   87.420208] usb 1-1.2: new full-speed USB device number 9 using ehci-pci
[   87.533989] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[   87.533991] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   87.533992] usb 1-1.2: Product: ConBee II
[   87.533993] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[   87.533994] usb 1-1.2: SerialNumber: DE2234239
[   87.534408] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[  126.876899] usb 1-1.2: USB disconnect, device number 9
[  127.102092] usb 1-1.2: new full-speed USB device number 10 using ehci-pci
[  127.216848] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[  127.216851] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  127.216854] usb 1-1.2: Product: ConBee II
[  127.216855] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  127.216857] usb 1-1.2: SerialNumber: DE2234239
```

Note that deconz is not installed on the host - it's a pretty light weight install of Ubuntu 20.04, currently on `5.4.0-66-generic`.

I also swapped out the USB extension cable I was using with no effect. Finally, I pulled the other two USB devices on the host (a UPS and a USB->Serial adapter) and am still seeing resets.

After all this, I installed deconz on the host to check the firmware:

```
$ GCFFlasher_internal -l
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2234239  | ConBee II
```

Reflashing the firmware with the same version via `sudo GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF -d /dev/ttyACM0` didn't fix it.

I then flashed the other (newer?) firmware file: `sudo GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26680700.bin.GCF -d /dev/ttyACM0`. Same thing there.

Any further suggestions or next steps?

Mimiix, 2021-03-04:
@deviantintegral two things here: 

1: We can't rule out involvement of the VM.
2: Please open a own user question issue as your environment is different.

Goldwing1973, 2021-03-27:
Bump.
I’m still migrating from a VMWare ESXi HA installation to a NUC.
To rule out conflicts with the Aeotec Gen5 Z-Wave stick I started with moving from the Z-Wave 1.4 integration to Z-WaveJS with the Z-WaveJS server on a different computer so that the Z-Wave stick is not in the HA machine, only the Conbee2 will be in the HA machine.

dvbit, 2021-03-27:
I have aeon and connbee on the same raspberry pi. In docker using
zwavejs2mqtt and zigbee2mqtt no problem

Goldwing1973, 2021-03-30:
Migration from vmware ESXi to a local install on a NUC is complete.
Now the testing starts.
For the first test i have  plugged the ConbeeII directly in a USB3 port on the NUC (no USB2 cable or USB2 Hub used)

Goldwing1973, 2021-04-01:
Well, it happened again.
I had to restart te Deconz integration.

22:22:41:857 wait reconnect 5 seconds
22:22:42:858 wait reconnect 4 seconds
22:22:43:857 wait reconnect 3 seconds
22:22:44:857 wait reconnect 2 seconds
22:22:45:857 wait reconnect 1 seconds
22:22:45:967 device disconnected reason: 4, index: 0
22:22:46:857 wait reconnect 15 seconds
22:22:47:857 wait reconnect 14 seconds
22:22:47:858 Skip idle timer callback, too early: elapsed 912 msec
22:22:48:857 wait reconnect 13 seconds
22:22:49:857 wait reconnect 12 seconds
22:22:50:858 wait reconnect 11 seconds
22:22:51:857 wait reconnect 10 seconds
22:22:52:857 wait reconnect 9 seconds
22:22:53:857 wait reconnect 8 seconds
22:22:54:857 wait reconnect 7 seconds
22:22:55:858 wait reconnect 6 seconds
22:22:56:857 wait reconnect 5 seconds
22:22:57:857 wait reconnect 4 seconds
22:22:58:858 wait reconnect 3 seconds
22:22:59:857 wait reconnect 2 seconds
22:23:00:858 wait reconnect 1 seconds
22:23:00:968 device disconnected reason: 4, index: 0
22:23:01:857 wait reconnect 15 seconds
22:23:02:858 wait reconnect 14 seconds
22:23:02:860 Skip idle timer callback, too early: elapsed 910 msec

This is on a NUC with NO virtualization with the Deconz stick connected directly to a USB3 port and no other USB devices.

Mimiix, 2021-04-01:
Can you use a extension cable? Reason 4 is not able to join the network.

Goldwing1973, 2021-04-01:
I will connect it to a extension cable tomorrow and restart the host.

Goldwing1973, 2021-04-02:
USB extension cable connected and Deconz restarted.

Goldwing1973, 2021-04-07:
And same problem is back.

11:48:42:483 device disconnected reason: 4, index: 0
11:48:43:367 wait reconnect 15 seconds
11:48:44:367 wait reconnect 14 seconds
11:48:44:369 Skip idle timer callback, too early: elapsed 911 msec
11:48:45:366 wait reconnect 13 seconds
11:48:46:367 wait reconnect 12 seconds
11:48:47:367 wait reconnect 11 seconds
11:48:48:367 wait reconnect 10 seconds

JakeHertenstein, 2021-04-07:
I had the same problems. My “solution” was to migrate to a RaspBee - rock solid ever since. Seems to be USB related

Goldwing1973, 2021-04-07:
RasBee is for a Raspberry pi, I’m using a NUC

JakeHertenstein, 2021-04-07:
Yes, I know

Goldwing1973, 2021-04-08:
And this morning again, so frustrating

Mimiix, 2021-04-08:
@Goldwing1973  Are you running the Addon or native deconz?

Goldwing1973, 2021-04-08:
The HA addon

[image]

Mimiix, 2021-04-08:
I specifically asked you to use a native os with a native deconz installation. Ha uses docker in its background.

Goldwing1973, 2021-04-08:
> 
> 
> Setup a native with Debian or ubuntu and install deconz on there. No virtual stuff.

My mistake, i thought you mend without using ESXi, HA native on Debian or Ubuntu.
RPi4 4gb with de Deconz image from the Phoscon site (Homebridge-hue + Raspbian Buster ) is that Ok?

Goldwing1973, 2021-04-10:
It just happened again, but this time a was behind my computer and this was the last log before it crashed.

11:29:11:866 device state timeout ignored in state 3
11:29:14:866 device state timeout (handled)
11:29:14:966 device disconnected reason: 4, index: 0
11:29:14:968 device disconnected reason: 4, index: 1
11:29:15:010 0x0017880103529E3B error APSDE-DATA.confirm: 0xA7 on task
11:29:15:330 0x0017880104A2A7A7 error APSDE-DATA.confirm: 0xA7 on task
11:29:15:866 wait reconnect 15 seconds
11:29:15:869 Skip idle timer callback, too early: elapsed 916 msec
11:29:16:366 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
11:29:16:369 device state timeout ignored in state 1
11:29:16:866 wait reconnect 14 seconds
11:29:17:866 wait reconnect 13 seconds
11:29:18:866 wait reconnect 12 seconds
11:29:19:866 wait reconnect 11 seconds

[24 more replies]

closed 2021-06-16 by github-actions[bot]