dresden-elektronik/deconz-rest-plugin #3342: Some Fyrtur blinds don't respond sometimes when triggered by automation

parruda, 2020-10-01:
## Describe the bug
I am having the same problem described in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2809

I have 12 Fyrtur blinds in my house, all connected to deCONZ. I set an automation to open/close them automatically at some times of the day. I also have Ikea open/close remotes paired and added to the blind room groups.

They always respond when I push a button on the remote, but every once in a while one of the blinds won't open/close when triggered by the automation.
I have the `Current Position` attribute of `Window covering` set as follows for all blinds:
* Min Report Interval: 1
* Max Report Interval: 60
* Reportable Change: 1

I am able to get the offending blind working again by sending it some of the commands described at https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies#keyboard-shortcuts

Let me know if you need any more information. I will add the logs here next time it happens again.

## Steps to reproduce the behavior
I don't know how to reproduce the problem. It happens randomly, but often.

## Environment

 - Host system: Raspberry Pi
 - Running method: Homebridge Pi image
 - Firmware version: (264A0700)
 - deCONZ version: (2.05.81)
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs
I just enabled the logs and will update this issue once it happens again.

zigistar, 2020-10-02:
I had the same issue in the past. Sometimes homekit automation worked for both blinds, sometimes only for one. Open/close remote worked always.

I updated some days ago to the latest firmware version for conbee-1 and since that day automation worked every single time.

closed 2020-11-01 by stale[bot]

cosmo84, 2021-09-01:
May I ask what solved the issue. Since I have the same issue with the latest version of deconz as well as the latest ConbeeII firmware.

dusopn, 2021-09-29:
Hello I have the same issue, one of the blinds never reports state back, but both of them occasionally stop responding. My conbee 2 is up to date.

cosmo84, 2021-09-29:
Something interesting: I had two older bulbs from Osram in my network. In the last weeks both of them regularly rebooted and then turned on during the day. Since I could not get rid of this even after resetting and rejoining them, I finally replaced them with two IKEA bulbs. And since then, after removing the battery of all my 10 IKEA blinds once more, everything works perfect. Reliable reporting and no more issues with non-responding blinds. 

@manup Do you think that such issues are possible, caused by some not well behaving routers, Osram bulbs in this case?