dresden-elektronik/deconz-rest-plugin #3345: ConBee II not joining network, error 0xE8

Arquiteto, 2020-10-02:
## Describe the bug
ConBee II unable to join the network. 
I had some issues with ConBee on my QNAP NAS, that presented as ConBee changing USB device number in the system. After QNAP reboot it was assigned Bus 01 Device 04, and could go up to Device 100+ after few days. Each change caused it to be disconnected from virtual machine and I had to add it to VM manually again. Not sure if this is associated with my current problem, but it caused me to move my deCONZ to Docker container.

After setup of Marthoc deCONZ container I restored backup made on HA. It worked, but all the devices were offline and I couldn't add any new. I decided to upgrade firmware to latest, made it via docker script provided by marthoc. It wouldn't resolve the issue. I tried a fresh install without restoring backup but Phoscon app was now always saying "Firmware: not connected", deCONZ showed Conbee II connected as /dev/ttyACM0 but "Not in network". I tried to change USB port in QNAP, connect extension cable. Then I tried fresh install in Home Assistant again and finally in Windows - on every install issue seems the same.

As a last resort I downgraded firmware to 26580700 again - no change.

## Screenshots
[image]
[image]

## Environment

 - Host system: QNAP NAS and Windows
 - Running method: HA deCONZ Add-on in HassOS QNAP VM, Marthoc Docker container on QNAP, Windows 10
 - Firmware version: 26580700 and 26650700
 - deCONZ version: 2.05.81 on Windows, 2.05.83 on Marthoc Docker and 2.05.81 HA Add-on
 - Device: ConBee II
 - Do you use an USB extension cable: tried both

### deCONZ Logs
10:50:34:774 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26580700.bin.GCF
10:50:34:775 GW firmware version: 0x26580700
10:50:34:775 GW firmware version is up to date: 0x26580700
10:50:41:690 start reconnect to network
10:50:46:690 try to reconnect to network try=1
10:50:46:700 Device protocol version: 0x010B
10:50:46:769 Current channel 25
10:50:46:792 CTRL ANT_CTRL 0x03
10:50:47:214 Device protocol version: 0x010B
10:50:47:282 Current channel 25
10:50:47:303 CTRL ANT_CTRL 0x03
10:50:47:344 NET ZDO start network status 0xE8
10:50:51:689 try to reconnect to network try=2
... (goes until try=10 and starts again)

More log from restart of HA Add-on: https://pastebin.com/ecrbayJY

Smanar, 2020-10-02:
Make your try on windows as you have one.
Make a firwmare update on windows (the one you prefer) with the command "-t 60" > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

If you still have the "firwmare not connected" can you compare your deconz setting with this one > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help

Arquiteto, 2020-10-02:
@Smanar 
I tried on Windows, tried on Docker, tried on Raspbian - also with "-t 60". The firmware updates, verifies and says SUCCESS.

But my settings are different
[image]

When I try to change them they change back before I am able to write even one field and click save

SwoopX, 2020-10-02:
You need to leave the network before. Check the wiki, there's a larger article.

Arquiteto, 2020-10-02:
@SwoopX 
Problem is I can't leave it, because ConBee II fails to join the network and the button is always greyed-out. Is there any way I can stop it from trying to join so I can edit?

Smanar, 2020-10-02:
Perhaps same problem than him https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3154

Arquiteto, 2020-10-02:
@Smanar

Yes, that was it! After setting custom MAC address it connected to the network!
Thank you very very much.  I was about to declare my ConBee dead.

closed 2020-11-01 by stale[bot]

vjonas, 2023-04-05:
> @Smanar
> 
> Yes, that was it! After setting custom MAC address it connected to the network! Thank you very very much. I was about to declare my ConBee dead.
> 
> No idea why I didn't find it, I looked over other issues before commiting this one. Must have overlooked it.

How did you set a custom mac address to it?

Mimiix, 2023-04-05:
> How did you set a custom mac address to it?

This is something best asked on the forums instead of in an near 3 year old issue.

Locking this up.