dresden-elektronik/deconz-rest-plugin #3346: Bug report: Xiaomi Aqara dual channel relay - Group control doesn't work, direct control does

Rick-Jongbloed, 2020-10-02:
## Describe the bug

I've connected an Xiaomi Aqara dual channel relay and hooked it up in deconz. It works when using VNC. 
Then i've created a group and added the device to the group.

When I control the group by clicking on the light icon of the group, nothing happens. 
But when I click on the light icon in the device inside the group, the relay is controlled.

## Steps to reproduce the behavior
  If the problem is reproducable, list the steps here:

  1. Create a group
  2. Add the device (exposed as a light) to the group
  3. Turn on the group
  4. nothing happens
  5. Open the group and turn on the individual light by clicking on the light icon
  6. The relay is switched.

## Expected behavior
when the group is turned on or off, the relay is controlled.

## Screenshots
 I've created a Word document with the screenshots of all the clusters in VNC:
https://tools4ever-my.sharepoint.com/:w:/p/r_jongbloed/EXiTjQcVHodCivg4rrQdr1MBYXdKobNoISwLYtnpIvKwTQ?e=gqVxYx

## Environment
Home as
 - Host system: (Raspberry Pi 4B / arm32 )
 - Running method: (Home Assistent deCONZ Add-on )
 - Firmware version: (deCONZ_ConBeeII_0x26650700)
 - deCONZ version: (2.05.81)
 - Device: (ConBee II )
 - Do you use an USB extension cable: (yes ) 

### deCONZ Logs
If required, please me know. I will have to find out how I get usable logs out of HassIO.

## Additional context
Without really knowing whats going on, it seems like the group on/off/toggle functionality doesn't point to the right device control endpoints. Is there a default group endpoint which is missing on this device?

SwoopX, 2020-10-04:
It is not confirmed those devices support zigbee groups at all. To be more precise, the may allow you to assign it to a group, but it's questionable if they can use group casts unless somebody has successfully done that.

Rick-Jongbloed, 2020-10-05:
Thanks for your answer. Is there anyway I can help you guys to find out if group cast works for this device, or should I just leave it with this and be happy I can control the relays via Home Assistant? :-)

Rick-Jongbloed, 2020-10-21:
Closing issue as i'm not getting any response on my offer to help out :-)

closed 2020-10-21 by Rick-Jongbloed