dresden-elektronik/deconz-rest-plugin #3348: Zemismart Chain Roller Shades Driver M515EGB (TS0601)

fosterym, 2020-10-02:
Please add the correct support of Zemismart Chain Roller Shades Driver M515EGB.
At this time it defined as light and plug in Deconz App, and Thermostat in Home Assisstant integration

https://zigbee.blakadder.com/Zemismart_M515EGB.html

## Device

- Product name: Zemismart Chain Roller Shades Driver .
- Manufacturer: _TZE200_xuzcvlku
- Model identifier: TS0601
- Device type : cover

## Screenshots

[image]
[image]
[image]
[image]
[image]

Smanar, 2020-10-03:
Hello I have added the deive in the code, but I realy don't understand why it is reconized as thermostat ...

To test the new code, on an Unix machine (Hassio is not on Unix OS)

You have the procedure https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

Just replace the step 1 by
git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git
and you can skip the step
git checkout -b mybranch HEAD

The last step replace your actual file, so better to close deconz before, or deconz will be closed itself by the change. (you can make a backup if you want but in another folder, file is libde_rest_plugin.so)

After that you just need to re-include the device.

fosterym, 2020-10-03:
Hi!
Thanks for your fast reply!

I have raspbian os with hassio in docker.
Deconz installed as hassio plugin (in docker too).
Can you send to me instruction for correct replacement?

Smanar, 2020-10-04:
Sorry not possible with Hassio, you need a complete OS to make compilation.

Can make in blind mode, but will be usefull to to test it to be sure.

alekslyse, 2020-11-08:
Was this ever solved? I have the same device and it shows up as a light switch. 

Clicking on the buttons in HA gives the following in the log

19:12:15:059 Tuya debug 77: EP:  1
19:12:15:059 Send Tuya Request
19:42:11:376 Send Tuya Request
19:42:12:810 Send Tuya Request
19:42:14:058 Send Tuya Request

Here is the API output:

{
    "etag": "832117b63cf5eae7e34b23b854515c38",
    "hascolor": false,
    "lastannounced": "2020-11-08T18:25:35Z",
    "lastseen": "2020-11-08T18:33Z",
    "manufacturername": "_TZE200_xuzcvlku",
    "modelid": "TS0601",
    "name": "Window Lift",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "bc:33:ac:ff:fe:da:05:3a-01"
}

[image]
[image]
[image]

Smanar, 2020-11-08:
This device is in this fork, if you want to test it
`git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git`

alekslyse, 2020-11-08:
Is it an active pr for this or just test?

alekslyse, 2020-11-09:
> This device is in this fork, if you want to test it
> `git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git`

I tried your branch and the device comes up as a thermostat, but nothing new on 2.05.86 (with your rest plugin)

{
    "etag": "e2b08042748c141a6a5d079f6d4ef73e",
    "hascolor": false,
    "lastannounced": "2020-11-09T15:57:20Z",
    "lastseen": "2020-11-09T16:02Z",
    "manufacturername": "_TZE200_xuzcvlku",
    "modelid": "TS0601",
    "name": "Smart plug 2",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "bc:33:ac:ff:fe:da:05:3a-01"
}

[image]

Smanar, 2020-11-09:
Can you make another inclusion ? but without deleting the device ?

I m seing 2 way possibles for your device, reconized as switch but you will have 3 times the device or covering, realy strange you haven't one of them ....

alekslyse, 2020-11-09:
> Can you make another inclusion ? but without deleting the device ?
> 
> I m seing 2 way possibles for your device, reconized as switch but you will have 3 times the device or covering, realy strange you haven't one of them ....
> 
> You know the procedure to replace the lib file ?

I added it again, and waited 30 minutes, then it worked and came in HA as cover. When will this be integrated in the official API?

Smanar, 2020-11-09:
You will have a new stable in 4/5 days without it.
The code will be in the beta the 15th.

alekslyse, 2020-11-09:
Great. Tested now, though in home assistant up is down and down is up compared to the remote and buttons on the controller. Could up and down be switched in the code?

Smanar, 2020-11-09:
Yep, it s possible, but I m using same command for all devices, are you sure the problem is not from your device ? Wiring ? There is too an hidden command to chnage direction on some device

`/*! PUT, PATCH /api/<apikey>/lights/<id> with {"reverse":true}`

alekslyse, 2020-11-09:
Its no wiring but could be this model is opposite. Since its an rest command it dosent matter much

alekslyse, 2020-11-10:
> You will have a new stable in 4/5 days without it.
> The code will be in the beta the 15th.

Just to be clear its included in 2.05.88?

Mimiix, 2020-11-10:
@Smanar Please link the issue to your PR.

Smanar, 2020-11-10:
@alekslyse yes, if all is good, version 88.
Have you tried the "reverse" command (with Curl for exemple) ?

Mimiix, 2020-11-10:
@Smanar Not going to be 88. We are switching to another versioning. 

Stable is comming out first ;)

Smanar, 2020-11-10:
Lol, seriously ?
I have just read the version from the github homepage.

Mimiix, 2020-11-11:
I still need to change that one :(

alekslyse, 2020-11-14:
Its important to know if you run the blinds on .86 it will make stable 88 go crazy and set all entities unavailable every minute. Removing the blind entity (aka the light bulb) solves this issue until the tuya PR can be pulled

eric-zimmer, 2021-01-05:
I am thinking about buying this chain roller driver, is it support now in the latest beta or stable? Will it be supported in phoscon as well, I am not using HA. Only deconz and phoscon

Smanar, 2021-01-05:
On deconz yes, on phoscon, good question, IDK if phoscon handle window covering ...

Smanar, 2021-01-22:
Ok si it seem there is a new model for this device, but this one is not a router but an end device.

Probleme the code don't support it yet, no one have problem to include it ?

closed 2021-02-19 by Smanar

JasperMC, 2021-03-25:
Hello @Smanar,

I have this same device except battery powered (I see the original creator of this issue has it on "Main (Single phase)" and I'm unable to pair it through the Phoscon app. It shows up as a node in the deCONZ GUI connected to the Configuration tool, but is not available in the REST API when I test it with Postman. Is there any chance you could help me with this? I'm able to view the clusters and everything in deCONZ.

If this needs to be a separate thread that's fine - I thought it might fit here because it might be regarding the same model/device.

[image]

Smanar, 2021-03-25:
Yes, this issue is knowed, if you want to try https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4616/files
This PR create a battery sensor to help deconz to reconize it as powered device.