dresden-elektronik/deconz-rest-plugin #3349: deconz reset and 100% CPU usage after Update

Fruchuxs, 2020-10-03:
I use the Docker COntainer from marthoc. I upgraded today to marthoc/deconz:latest and I noticed a high CPU usage on one core. I restarted my pi and now all devices are lost. The folder where stored my settings is empty now.

Smanar, 2020-10-03:
Hello, have you a deconz backup (devices) to make try with it ?

Fruchuxs, 2020-10-03:
Nope, I didn't update the firmeware. Shall I try? Di you have a link to a doc / tutorial? I never updated the firmeware before. :p
I try to decompress an image from my sd card and grab the files. But the CPU usage is a problem.

Smanar, 2020-10-03:
If you haven't updated yourself the firwmare, there is few reason for it have stoped working.

The problem is I can say idf it s the 100% cpu use that make the firwmare not connected or the reverse, or perhaps there is no link.

For me the easier thing to test is the hardware, just install deconz on a PC and test it.

If you still have firmware not connected > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Fruchuxs, 2020-10-03:
Nah, its just a restart no boot loop.
Can't install deconz on a PC cuz the raspbee II is on the GPIOs from my rpi. The rpi has no gui installed.

Fruchuxs, 2020-10-03:
Rolled back to the latest deconz Image:

```
docker image ls
REPOSITORY                                 TAG                 IMAGE ID            CREATED             SIZE
marthoc/deconz                             latest              52e32f631bcf        2 days ago          256MB
marthoc/deconz                             <none>              f0d5b44e5960        2 weeks ago         242MB
```

I use the f0d5b44e5960 again and the firmeware is back again: 26520700

Mimiix, 2020-10-03:
@Fruchuxs Please use the issue format. I am really tempted to close this issue and ask you to create a new one.

Fruchuxs, 2020-10-03:
Sure, I can add the missing headlines. The only stuff I missed is the information that I use a raspbee II. But I added the information quickly.

Fruchuxs, 2020-10-03:
Hm, it wants an update to a conbee fw:
```
17:48:21:129 dev /dev/ttyAMA0
17:48:21:129 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26580700.bin.GCF
17:48:21:129 GW firmware version: 0x26520700
```

S474N, 2020-10-03:
Have same problem with Docker (but have ConBee II)

In log:
```
today at 20:50 [marthoc/deconz] Starting deCONZ...
today at 20:50 [marthoc/deconz] Current deCONZ version: 2.05.83
today at 20:50 [marthoc/deconz] Web UI port: 80
today at 20:50 [marthoc/deconz] Websockets port: 443
today at 20:50 [marthoc/deconz] VNC port: 5900
today at 20:50 /usr/bin/xauth:  file /root/.Xauthority does not exist
```

In htop is 100% CPU usage with process with Xtigervnc.

Have only actual image id 52e32f631bcf :(

S474N, 2020-10-03:
So now tested ConBee II on Windows 7 and upgraded firmware to 26650700 

```
GCFFlasher V3_14 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device COM24 (ConBee II)
deCONZ firmware version 26580700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
unlock!  (6242 ms)
flashing 164090 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
```

But when is ConBee II back in RPi ( `lsusb` can see it), problem is same.

S474N, 2020-10-04:
Here is temporary fix:
https://github.com/marthoc/docker-deconz/issues/271#issuecomment-702565796

Fruchuxs, 2020-10-04:
Thanks @S474N. After reading the issue I think the problem lies in the docker container and is not a deconz problem. Will track https://github.com/marthoc/docker-deconz/issues/271 for now - closing this issue.

closed 2020-10-04 by Fruchuxs