dresden-elektronik/deconz-rest-plugin #3350: Develco Siren SIRZB-110

jarlesb, 2020-10-03:
## Device

- Product name: Develco Smart Siren
- Manufacturer: Develoc Products A/S
- Model identifier: SIRZB-110
- Device type : 
  - Siren

## Screenshots

[Develco Siren SIRZB-110 - screenshot.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5322214/Develco.Siren.SIRZB-110.-.screenshot.pdf)
[Develco Siren SIRZB-110 - Tech manual.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5322216/Develco.Siren.SIRZB-110.-.Tech.manual.pdf)

jarlesb, 2020-10-03:
Please add device support for the Develco Smart Siren. Currently recognised as light and does not make any noise. See detailed specifications in the attachments.

Thanks!

closed 2020-10-05 by manup

tyr84, 2021-01-24:
So, I've added the siren, and it shows up as a light. However, I can only activate the siren through the deconz interface:
[image]

I've tried the same method as the develco smoke alarms, setting state-alert to select or lselect, both in the api, and through the deconz.configure service in home assistant, but there is no response from the siren.
[image]
[image]

Am I doing something wrong, also is it possible to reach the deconz functionality shown in the first screenshot through the api? (selecting warning mode etc)