dresden-elektronik/deconz-rest-plugin #3351: Develco Keypad / Alarm panel

jarlesb, 2020-10-03:
Please add support for Develco Keypad / Alarm panel. All details shoudl be given in the attachments.

Thanks!

## Device

- Product name: Develco Keypad
- Manufacturer: Develco Products A/S
- Model identifier: KEYZB-110
- Device type : 
  - Other: Keypad / Alarm panel

## Screenshots
[Develco Keypad KEYZB-110 - screenshot.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5322224/Develco.Keypad.KEYZB-110.-.screenshot.pdf)
[Develco Keypad KEYZB-110 - Tech manual.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5322226/Develco.Keypad.KEYZB-110.-.Tech.manual.pdf)

SwoopX, 2020-10-04:
You're probably not going to be happy with that one. In the end, it will only fire a couple of button evens...

jarlesb, 2020-10-04:
> You're probably not going to be happy with that one. In the end, it will only fire a couple of button evens...

Yes, it is just a few buttons, but good as part of an alarm system. In addition, it reads RFID, which I would also like to use for activating/deactivating. Will that not work?

SwoopX, 2020-10-04:
No, that's what I'm trying to express here. You could switch some lights or plugs on and off, nothing more.
You will also not be able to use the more sophisticated features of the siren without the Develco gateway. That stuff is closed source and must be reverse engineered with the gateway and devices at hand.

jarlesb, 2020-10-04:
> No, that's what I'm trying to express here. You could switch some lights or plugs on and off, nothing more.
> You will also not be able to use the more sophisticated features of the siren without the [...]

OK, that is then too bad... :( Any siren and keypad/alarm panel that is supported? Or something I could do to support the reverse engineering?

jarlesb, 2020-10-05:
At the moment, I am not even able to see the device in Home Assistant. So would appreciate if you could at least make it visible, then I may start with working with the buttons.

alekslyse, 2020-10-06:
I did a full wireshark sniff of that device on the z2m forum capturing all the buttons so if anyone is interested its a full dump posted there.

alekslyse, 2020-10-06:
Here is the full pcap file if anyone wants to have a go. the zigbee key is the z2m default for decrypt

[alarmpanel.pcapng (1).zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5335929/alarmpanel.pcapng.1.zip)

SwoopX, 2020-10-06:
Nice, I'll have a look when I got some time. You don't have their gateway, do you?

closed 2020-11-09 by stale[bot]

Smanar, 2021-10-09:
It's WIP https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5352