dresden-elektronik/deconz-rest-plugin #3353: Updated Docker - Nothing Works

Jimmy-SafeCash, 2020-10-03:
Hello all,

Have been running the docker image for about a month on v81 without issue. Upgraded today to :latest and everything has stopped working.

I ran docker-compose up -d to pull the latest version now I cannot connect to the phoscon web gui or the VNC. I receive the following from the logs:

`pi@raspberrypi:~/docker-compose/deconz $ docker logs deconz -f
[marthoc/deconz] Starting deCONZ...
[marthoc/deconz] Current deCONZ version: 2.05.83
[marthoc/deconz] Web UI port: 80
[marthoc/deconz] Websockets port: 443
[marthoc/deconz] VNC port: 5900
/usr/bin/xauth:  file /root/.Xauthority does not exist

New 'raspberrypi:0 (root)' desktop at :0 on machine raspberrypi

Starting applications specified in /etc/X11/Xvnc-session
Log file is /root/.vnc/raspberrypi:0.log

Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd raspberrypi:0 to connect to the VNC server.
`
Any ideas would be greatly appreciated.

haselchen, 2020-10-04:
Try Webport 8077 
You have to Change it at the variables. 
So you have to create a New Container.
After Change the Port Try IP from your Raspi:8077

Jimmy-SafeCash, 2020-10-04:
> Try Webport 8077
> You have to Change it at the variables.
> So you have to create a New Container.
> After Change the Port Try IP from your Raspi:8077
> I Use a Synology with Docker so the Result maybe not the Same.

Thanks. I will try this later. Why do you have to do this? I've never had to do this with docker before.

haselchen, 2020-10-04:
Info:

deCONZ is having a beta release at the 15th day of the month in which the beta of last month becomes stable. Pull requests done before the 10th of the month, are getting included in the next beta (if there are no issues ofcourse!). Stable release would be around the 1st - 5th of the month after latest beta. Current Beta: 2.05.82 Current Stable: 2.05.81

Next Beta: 2.05.85 expected at the 15th of October. Next Stable: 2.05.84 expected at 8th October (minor fixes and new firmware files)

Jimmy-SafeCash, 2020-10-04:
> Info:
> 
> deCONZ is having a beta release at the 15th day of the month in which the beta of last month becomes stable. Pull requests done before the 10th of the month, are getting included in the next beta (if there are no issues ofcourse!). Stable release would be around the 1st - 5th of the month after latest beta. Current Beta: 2.05.82 Current Stable: 2.05.81
> 
> Next Beta: 2.05.85 expected at the 15th of October. Next Stable: 2.05.84 expected at 8th October (minor fixes and new firmware files)
> 
> Edit: I think you took this Manual?(in german)
> 
> DockerUnterstützte HostsystemeLinux x86_64/amd64Linux armv7 (z. B. Raspberry Pi)Linux arm64
> 
> Um die Steuersoftware deCONZ in einem Docker Container zu betreiben, empfehlen wir das von der Community gepflegte Docker Image marthoc/deconz. Dieses ist auf amd64, arm64 und armhf Plattformen lauffähig.
> 
> https://hub.docker.com/r/marthoc/deconz
> 
> Installation
> 
> Im Folgendem wird beschrieben wie der deCONZ Container auf einem Linux Hostsystem gestartet wird.
> 
> USB-Zugriffsrechte für Nutzer setzen
> 
> sudo gpasswd -a $USER dialout
> 
> Hinweis: Die Zugriffsrechte werden erst nach dem Aus- und Einloggen bzw. nach einem Neustart aktiv.
> 
> Verzeichnis für persistente Konfiguration erstellen
> 
> mkdir -p ~/.local/share/dresden-elektronik/deCONZ
> 
> deCONZ Docker Container starten
> 
> docker run -d \ --name=deconz \ --net=host \ --restart=always \ -v /etc/localtime:/etc/localtime:ro \ -v ~/.local/share/dresden-elektronik/deCONZ:/root/.local/share/dresden-elektronik/deCONZ \ --device=/dev/ttyACM0 \ -e DECONZ_DEVICE=/dev/ttyACM0 \ marthoc/deconz
> 
> Mit dieser Konfiguration läuft deCONZ im Docker Container auf Port 80.
> 
> My Solution on a Synology
> [ [image] ](url)

Changing ports has had no effect. 

Nothing will load. I cannot even revert to the old version. Upgrading the container has completely broken everything.

tobsl, 2020-10-04:
Experiencing the same issue with headless docker on a raspberry pi 3.
Installation was running flawlessly with 2.05.80 for weeks.
Updating marthoc/deconz image to current :latest and nothing works anymore.
Phoscon UI page with gateway information shows "Firmware: not connected".
Switching back to old image (2.05.80) and everything is fine again.

**Logs**
Did set DEBUG_INFO=2 to enable more log output.
```
[marthoc/deconz] Starting deCONZ...
[marthoc/deconz] Current deCONZ version: 2.05.83
[marthoc/deconz] Web UI port: 80
[marthoc/deconz] Websockets port: 443
[marthoc/deconz] VNC port: 5900
/usr/bin/xauth:  file /root/.Xauthority does not exist

New 'raspi3deconz:0 (root)' desktop at :0 on machine raspi3deconz

Starting applications specified in /etc/X11/Xvnc-session
Log file is /root/.vnc/raspi3deconz:0.log

Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd raspi3deconz:0 to connect to the VNC server.

```
No more log output. :(
Also raspi `dmesg` shows nothing related to the conbee usb device.

**Tech spec**
USB conbee II with firmware 26650700
Raspberry Pi 3 Model B Rev 1.2

Any help or hint is appreciated.

Jimmy-SafeCash, 2020-10-04:
> Experiencing the same issue with headless docker on a raspberry pi 3.
> Installation was running flawlessly with 2.05.80 for weeks.
> Updating marthoc/deconz image to current :latest and nothing works anymore.
> Phoscon UI page with gateway information shows "Firmware: not connected".
> Switching back to old image (2.05.80) and everything is fine again.
> 
> **Logs**
> Did set DEBUG_INFO=2 to enable more log output.
> 
> ```
> [marthoc/deconz] Starting deCONZ...
> [marthoc/deconz] Current deCONZ version: 2.05.83
> [marthoc/deconz] Web UI port: 80
> [marthoc/deconz] Websockets port: 443
> [marthoc/deconz] VNC port: 5900
> /usr/bin/xauth:  file /root/.Xauthority does not exist
> 
> New 'raspi3deconz:0 (root)' desktop at :0 on machine raspi3deconz
> 
> Starting applications specified in /etc/X11/Xvnc-session
> Log file is /root/.vnc/raspi3deconz:0.log
> 
> Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd raspi3deconz:0 to connect to the VNC server.
> ```
> 
> No more log output. :(
> Also raspi `dmesg` shows nothing related to the conbee usb device.
> 
> **Tech spec**
> USB conbee II with firmware 26650700
> Raspberry Pi 3 Model B Rev 1.2
> 
> Any help or hint is appreciated.
> 
> Edit:
> Also tested current :stable (2.5.81). Exact same problem. Switching back to 2.5.80 and it runs again without issues.

Thanks for confirming @tobsl - I've been going made here thinking that it's only me!

I can confirm that 2.5.80 is the last version working on Pi4

Mimiix, 2020-10-05:
https://github.com/marthoc/docker-deconz/issues/271

I think your not alone ;)

crixyd, 2020-10-10:
Same issue here on 2.05.84... deCONZ is 100% unusable, with all of the same issues reported above. 2.05.80 was fine.

crixyd, 2020-10-10:
On further review, if I disable VNC in docker-compose.xml Phoscon UI will start, however all I get is firmware disconnected, and no ability to interface with the RaspBee II

malfroid, 2020-10-14:
For me, rolling back to armhf-2.05.82 and changing the VNC port to something other than 5900 (e.g. 5902) solved the issues.
Keeping the VNC port at 5900 caused 100% unusable deCONZ.

Example working docker-compose.yml file:
``` yml
version: '3'
services:
  deconz:
    container_name: deconz
    image: marthoc/deconz:armhf-2.05.82
    network_mode: host
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /opt/deconz:/root/.local/share/dresden-elektronik/deCONZ
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    environment:
      - TZ=Europe/Berlin
      - DECONZ_WEB_PORT=801
      - DECONZ_WS_PORT=445
      - DECONZ_VNC_MODE=1
      - DECONZ_VNC_PORT=5902
      - DECONZ_VNC_PASSWORD=<password>
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
```

Mimiix, 2020-10-14:
As i mentioned before, there's more users with this problem. I'd like to ask to keep the discussion in that issue: https://github.com/marthoc/docker-deconz/issues/271

closed 2020-10-14 by Mimiix