dresden-elektronik/deconz-rest-plugin #3354: New manufacter on market  Adeo/LeroyMerlin works almost correctly

tutur05, 2020-10-04:
Hi everybody :) 
I have a great news, Adeo/Leroy Merlin has new Zigbee Light. It's good quality and not expansive ! I try some one whith ConBee II and Jeedom . The pair works correcly, I can change color ! 
Just something I don't understand, I can't push light at 100% of brithness when I use software, only the original button can..
I beguining in Zigbee protocol, do you now what I miss ?
I try brightness at 255 but the light only 50% really..
sam whith saturation...

Exemple for E27 light RGB

> {
> 
>     "6": {
>         "ctmax": 65535,
>         "ctmin": 0,
>         "etag": "8629b0be34a4804ea65d499ea0309b79",
>         "hascolor": true,
>         "lastseen": "2020-10-04T07:31:49.505",
>         "manufacturername": "Adeo",
>         "modelid": "LXEK-1",
>         "name": "Extended color light 6",
>         "state": {
>             "alert": "none",
>             "bri": 149,
>             "colormode": "hs",
>             "ct": 370,
>             "effect": "none",
>             "hue": 6192,
>             "on": false,
>             "reachable": false,
>             "sat": 243,
>             "xy": [
>                 0.5102,
>                 0.4381
>             ]
>         },
>         "swversion": "2.4.0",
>         "type": "Extended color light",
>         "uniqueid": "60:a4:23:ff:fe:88:13:13-01"
>     }
> }

Any idee ? 
Thanks you for your jobs :) !

tutur05, 2020-10-04:
I progress a little.. 
It's seem there are 2 function
mode RGB (colormode = xy)
mode "light simple (colormode = ct)

If I'm in mode ct, i have 100% light.. 
but I can't change value colormode in Jeedom :/ 
I success change one whith phoscon App but i'm not sure how..

Smanar, 2020-10-04:
Hello, but it s a device that support RGB AND CT ?

On the Json we can see 
>"colormode": "ct"

This mode if for bulb that support only CT (color temperature), but I don't see what is the mode managed by devices that support both ...

> The current color mode of the light:
> 
>     hs - hue and saturation
>     xy - CIE xy values
>     ct - color temperature

It s possible jeedom look at this field to reconize the device type, and for him the device is not RGB but only CT

The problem can be from the bulb or from deconz (or perhaps jeedom if the CT colormode is right for this kind of device too)

But I m not sure what need to be the "colormode" for a device that support both ....

tutur05, 2020-10-04:
Hi ! 
Thanks for answer :) 
It's this model : https://www.leroymerlin.fr/v3/p/produits/kit-2-ampoules-connectee-led-standard-e27-telecommande-enki-lexman-e1506797854

I understand better , It's have the 3 mode you are correct when I tested, i see the 3 mode.

When i want send value I use this command in jeedom : 

[image]

So if I try this command : 
01.colormode::ct
It see: 
> Erreur lors de la requete : 127.0.0.1:8484/api/0329328FCF/lights/3/state(PUT), data : {"colormode":"hs"} erreur : 8 => parameter, colormode, is not modifiable.

01.bri::200
No problem. So maybee you right and Jeedom "block" command :/ 

What do you meen by GUI ? PhosconAPP ? yes

Thanks you :)

Smanar, 2020-10-04:
Nope the Deconz GUI, but you need an OS with desktop, and realy few jeedom user have it. But don't worry, not usefull ATM.

The "colormode" is read only, it s deconz that choose the value, so it can be guilty here.

I m asking more information to others ATM, I don't know weel how work light mode.

tutur05, 2020-10-04:
Thanks for all this information, I still search in time  I test with Windows, I found a lot information but it's too complicate for me ... ^^ 
I hope Jeedom find a solution ! It's possible it's the good news :) ! Just find how ^^ 
[image]

Smanar, 2020-10-04:
Don't worry, I don't think it s a big problem.

On your jeedom capture, there is 2 settings you are not using yet "hue" and "sat".
The user that know the more how works bulbs are away ATM, so need to wait for some time.

But if it works fine on phoscon, you can use the "developper tool" on your browser to see the phoscon request.

On my side IDK if there is a command for RGB+CT or if the 2 modes work separatly, "bri" is at max on the json, so if the bulb is not at 100%, perhaps need another command.

Mimiix, 2020-10-04:
@tutur05 Please use the proper format on requesting a device. (Issue template).

Closing.

closed 2020-10-04 by Mimiix