dresden-elektronik/deconz-rest-plugin #3357: Device can't associate to Conbee 2 because PAN is full

elbrini, 2020-10-04:
Hi,

I have a conbee2 that I use as coordinator:
Firmware version: 0x26650700
deCONZapp version:  2.05.81

When I try to perform a device association the response is always unsuccessful. The association response got status 0x01 which means the PAN is full.

I tried to reset the gateway among many things but I can't join the network.

Does anyone come across this issue and how can I fix it?

Thanks in advance

SwoopX, 2020-10-04:
> The association response got status 0x01 which means the PAN is full.

Where do you got that from?

Mimiix, 2020-10-05:
I think i've seen this before when people didn't use the extension cable. Information on your template is missing: Mainly if you use a extension cable or not.  

Next time, please use the template.

closed 2020-11-09 by stale[bot]