dresden-elektronik/deconz-rest-plugin #3365: Sylvania S+ 73693 missing tunable white settings

N5A, 2020-10-05:
I have two I've added through Phoscon, each only has a RGB color slider , clicking the Color Temperature selector below the color slider has no effect, they remain in RGB mode. 

OS: I have them connected to a ConbeeII running on a Pi4. 
Deconz Ver :Deconz installed a few days ago. running on ver 2.05.81
USB Ext.. Yes.
Consequently Home Assistant also only gives me RGB control.

What can I provide to assist in determining the issue I'm facing?

by template I don't see whats missing...

Mimiix, 2020-10-05:
Is is missing

N5A, 2020-10-05:
Fixed.

N5A, 2020-10-05:
May assist.. 
[image]
[image]
[image]

crazifyngers, 2020-10-12:
I am having a similar issue with four sylvania (osram/ledvance) smart+ rt5/6 rgbww downlights.  I moved from 2.05.79, where all channels worked to 2.0.5.84, where the CT channel is no longer working in phoscon and do not reflect changes in the api.  I am however able to change the CT value if I use the 2016 wireless light control interface.  This is how I am able to tell if the API is seeing any state changes.

guedeaux, 2020-10-12:
I too am having this issue on 2.05.84 with ledvance A19 RGBW.

N5A, 2020-10-12:
To note this did work on some version installed to my Home Assistant VM a year ago, however i've since moved Deconz to a centrally located Pi and off the Windows PC/HA VM due to unstable ConbeeII connection.

guedeaux, 2020-10-13:
Yes. These lights were working fine before, but now they have lost the color tuning and white level attributes in Phoscon. I am not sure when it occured, but I am on 2.05.84 now and they aren't working properly.

N5A, 2020-11-02:
*PUNT* Because the bot emailed me an order to do so.

guedeaux, 2020-11-03:
I have been hoping that this issue would resolve itself somehow, during a HA restart, host reboot, HA update, etc., but it is still present and super annoying through . I might try moving deconz to its own container instead of the HA add-on, not that doing this will help to resolve the underlying issue, though...

EricPalmquist, 2020-11-05:
I am having the same problems with the same lights.  ConBee (original) on Raspbee.  Version 2.05.78 firmware 26350500.

N5A, 2020-11-13:
Updated firmware to newest from LEDVANCE (2.2019). No immediate change.  THey were on older version I'll delete them from the network and readd them during the weekend.

guedeaux, 2020-11-14:
I deleted one, rebooted, re-added, rebooted... still no tunable white options

jasonsinclair5, 2020-11-16:
Anyone have any luck with this? I have a couple LEDVANCE BR30 RGBW bulbs that will not use color temp. The only white available is a bright white. I have just upgraded to 2.6.0 and firmware 26660700 on my Conbee II stick. I removed the light and re-added, still no luck. I just did an OTA upgrade to 00102428 on the bulbs, but still no luck with the color temp. I'm running deconz in a docker container, if that helps.

EricPalmquist, 2020-11-17:
I was able to hack the source code to get it to work for the RB30 RGBW and A19 RGBW.   I can share the source code changes if anyone would like.

Mimiix, 2020-11-17:
@EricPalmquist Feel free to make a PR from it! We might be able to integrate it then.

jasonsinclair5, 2020-11-17:
> I was able to hack the source code to get it to work for the RB30 RGBW and A19 RGBW. I can share the source code changes if anyone would like.

That would be awesome, but I'm running on docker, so I'm not sure I could re-compile the source.

EricPalmquist, 2020-11-21:
@Mimiix I think that I created a PR with my changes - this is my first one ever so not 100% that I did it correctly :)

N5A, 2020-11-23:
> I was able to hack the source code to get it to work for the RB30 RGBW and A19 RGBW. I can share the source code changes if anyone would like.

This is what Github and open source are for a million brains working on many problems. Anyone can pop in with the workable solution! I'd love to get the functions back! You rock!

Mimiix, 2020-11-23:
@EricPalmquist Looks good! I've changed the title ;)

closed 2020-11-30 by manup

N5A, 2020-12-05:
Thank you for the fix! Everything sees the color temp controls now.

Thomas-Vos, 2022-12-04:
Any reason this was set to `Color temperature light` in the PR instead of `Extended color light`? Color temperature light means no RGB support so that is wrong.

N5A, 2022-12-10:
Couldn't tell ya on that. Didn't have a hand in programming.