dresden-elektronik/deconz-rest-plugin #3366: Can a router lose requests ?

Smanar, 2020-10-05:
## Describe the question or issue you are having
Hello, it s something observed by another user, but if someone else have information about that, I m interested too.
So he have many ikea remote that use an ikea bulb as router
- if he use one remote fastly  (1 action by second) after some command, the remote stop working and all other too.
- after some time, all remotes work again.
- if he does 10 commands (1 by second) less than 10 are see by deconz.
- if he power off the bulbs, all works perfectly.

## Environment
- Ikea remote
- Ikea Bulb
- 11 routers
- 15 sensors

## Additional context
- He have tried with automation disabled
- He don't use group
- He don't use scene

Ok I can understand the problem, but BTW he have made a try with a simple network, can be possible the bulb was not able to mesh all requests from a simple remote ?

ebaauw, 2020-10-05:
Unfortunately, we’ve seen plenty of routing issues, so: yes.  I don’t think that’s the issue here, though.

Afaik, Ikea remotes use broadcasts (group commands) instead of unicasts, so their messages should reach the gateway, even if some routers are misbehaving.  Typically a Zigbee network can handle like one broadcast per second, so I would expect messages to get lost if you send 10 within a second.  By powering off the bulbs, you prevent the broadcast storm, as there are no routers to retransmit the message.

I had to change the symfonisk controller setup to send unicasts to the gateway instead of broadcasts, as it would send too many messages while turning the dial.  I suspect that’s probably why the original dimmer (with the turning dial) didn’t support group commands.

Smanar, 2020-10-06:
Ok, so it s realy a weakness.
On his try was not send 10 within a second, but 1 by second during 10s, not so terrific.

So it mean if we want to avoid that we can
- Use direct bind remote <> bulb ?
- Use wireless command that work with unicast ?

Zehir, 2020-10-11:
I think the Ilea stuff is too cheap, check this : https://github.com/home-assistant/core/issues/41475#issuecomment-706767971

MattWestb, 2020-10-12:
The broadcast limit is technically not in the zigbee layer its is in the underlying IEEE 802.15.4 that is limiting the bordast storms. And IKEA was having bad acting firmware that was blocking them self then getting to much commands but i think it was very old ones.

closed 2020-11-24 by stale[bot]