dresden-elektronik/deconz-rest-plugin #3368: Tuya TS0218 SOS Button

meremortals70, 2020-10-06:
Yet anther Tuya Device

[image]

[image]

## Device

- Product name: Tuya Pushbutton
- Manufacturer: _TYZB01_trqoesc6
- Model identifier: TS0218
- Device type : Remote

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

https://www.zigbee2mqtt.io/devices/TS0218.html

alexanderfitu, 2021-01-29:
I am looking for support for this to be added if possible. I can provide any logs if needed.

darking2539, 2021-07-07:
I have issues with this device. Detail in link: https://github.com/Koenkk/zigbee2mqtt/issues/7184. How to solve it? I see log (debug level) but noting at all show.

closed 2021-07-07 by close-issue-app[bot]

Mimiix, 2021-07-07:
@alexanderfitu Can you perhaps open an new issue ?  Follow this procedure: https://forum.phoscon.de/t/read-me-device-request-read-me/23

alexanderfitu, 2021-07-14:
> @alexanderfitu Can you perhaps open an new issue ? Follow this procedure: https://forum.phoscon.de/t/read-me-device-request-read-me/23

Will do!