dresden-elektronik/deconz-rest-plugin #3369: Tuya Gas/Water Valve TS0111 Enhancement Please

meremortals70, 2020-10-06:
This device registers in deconz as a light and has only one switch exposed.  There are additional sensors and can we make it something other than a light to reflect what it is?

[image]

## Device

- Product name: Tuya Smart Valve
- Manufacturer: _TYZB01_ymcdbl3u
- Model identifier: TS0111
- Device type :  Other: Valve

## Screenshots

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Smanar, 2020-10-06:
Your device is in use when you have take the capture ?

On the lasts capture "Simple metring" and "electrical mesurement" you have 0 in all values, can you use it to see if the value change.

But for me it useless, it s electrical mesurement and your device is used for fluid, so I don't think it will be relevant.

And you are lucky with a "light", for me this device need to be displayed as plug ^^.

meremortals70, 2020-10-06:
Thanks, 
The device was idle when I grabbed the captures.  I had just added it to the network.

I agree the additional sensors are probably not a great value to most people however I do track all my power usage as I have solar and battery with Iotawatt and currently plugs on all my high energy appliances to monitor and graph usage.

I also have automations which stop appliances like the dishwasher, washing machine and air conditioner being switched on when the solar and battery is low. Annoys the wife but she has an option to override manually but it does give her pause to think before she does it.

Smanar, 2020-10-07:
lol poor wife, take care at WAF.

meremortals70, 2020-10-08:
Not at all, need to change behaviour otherwise nothing changes

closed 2020-11-09 by stale[bot]

falaca, 2021-09-12:
@meremortals70 Aside from it showing up as a light, does this device work as expected with deCONZ?