dresden-elektronik/deconz-rest-plugin #3370: ADEO

tutur05, 2020-10-06:
## Device LEXMAN ENKI Lights

- Product name: The device name as shown on the product or package.
- Manufacturer: ADEO
- Model identifier: LXEK-1
- Device type : Lights warm/white AND RGB
https://www.leroymerlin.fr/v3/p/produits/ampoule-connectee-led-standard-e27-intensite-couleur-variables-enki-lexman-e1506793547

[image]

[image]

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

## 
Hi everybody :)
Sorry for bad last post ;) 
I have a great news, Adeo/Leroy Merlin has new Zigbee Light. It's good quality and not expansive ! I try some one whith ConBee II and Jeedom . The pair works correcly, I can change color !
Just something I don't understand, I can't push light at 100%
I beguining in Zigbee protocol, do you now what I miss ?
I try brightness at 255 but the light only 50% really..
same whith saturation...
I success to as 100% brightness whith phoscon app when I switch to color mode to light mode but Jeedom can't I dont no how Phocon change mode . 
Special thanks to @Smanar for help last week ;) 

> {
>     "3": {
>         "ctmax": 65535,
>         "ctmin": 0,
>         "etag": "0978a9409ae6f62b280912446ef10adc",
>         "hascolor": true,
>         "lastseen": "2020-10-04T08:07:40.721",
>         "manufacturername": "Adeo",
>         "modelid": "LXEK-1",
>         "name": "Extended color light 3",
>         "state": {
>             "alert": "none",
>             "bri": 254,
>             "colormode": "ct",
>             "ct": 156,
>             "effect": "none",
>             "hue": 33283,
>             "on": true,
>             "reachable": true,
>             "sat": 30,
>             "xy": [
>                 0.3141,
>                 0.3299
>             ]
>         },
>         "swversion": "2.4.0",
>         "type": "Extended color light",
>         "uniqueid": "60:a4:23:ff:fe:88:1e:7a-01"
>     }
> }

Smanar, 2020-10-06:
Hello again ^^.

Something strange you haven't the field "colorcapabilities" in the Json. On the cluster 0x0300 "color control" on the table on the bottom you haven't a field called "Color capabilities"  (not grayed ofc) ?

The value "colormode" is set by deconz according to value read in attribute 0x0008 and 0x4001 on the same cluster, if you can take a look in it (on the same table)

But if it work without problem in phoscon, I think the problem can be from jeedom, can you compare the command send by jeedom when you send a "bri" order (visible on jeedom log) and the command send by phoscon (will be harder, The easier way is enabling the browser debugging).

On deconz on the cluster 0x0008 "level control" you have the command for brightness "move to level" it take value from 0 to 255 (let 0 at transition time), I think this command will work.

I m not sure but I think the problem is from the field used in the same command.

ebaauw, 2020-10-06:
> Something strange you haven't the field "colorcapabilities" in the Json.

He’s running and old version of the API plugin, probably from before `colorcapabilities` was exposed.

tutur05, 2020-10-07:
Hi ! 
I try different way to reproduce, and you right both ! Jeedom send alway all Json request. So I duplicat Light in jeedom but I attribute only Bright and CT. After,  I try to change Ct and Jeedom send a json with only Bright and CT . Bingo, the light change to light mode !
@Smanar I'm sorry , I can't success find a way to see the request PhosconApp... I try with google chrome console but I don't find request json

So good news, It's officialy can supported at 100% by deconz :) ! Just adapted Jeedom in my case... DevTeam do you need more information for add to you supported list device ? 

Thanks a lot to you :)

Smanar, 2020-10-07:
>@Smanar I'm sorry , I can't success find a way to see the request PhosconApp... I try with google chrome console but I don't find request json

Don't worry, was in case you still have problem, but as you have find the solution yourself, it s useless now.

But for jeedom side, I realy don't know what to say, IDK why by defaut it is using both mode in same time.
I think too the field "colormode" is useless, the only mode available are "hs", "xy", "ct", so for bulb that support "xy" + "ct", what need to be this value ?

I think it s better to check wich one field are present in the "state" to check the bulb capabilities, but it seem it s not the capabilites detection that is problematic in jeedom.

closed 2020-11-09 by stale[bot]