dresden-elektronik/deconz-rest-plugin #3372: interlock: UNSUP_MANUF_GENERAL_COMMAND

RIGRIG1, 2020-10-06:
When I try to set interlock option I have next error: UNSUP_MANUF_GENERAL_COMMAND:
[image]

SwoopX, 2020-10-06:
Template not used.

closed 2020-10-06 by SwoopX

nasimon, 2021-01-01:
I have the same problem listed here. How can I solve it? Tried changing 0x103f mfgcode to 0x115f (like in "Xiaomi disconnect 1" attribute in cluster 0 that doesn't show the error when written or read.

Smanar, 2021-01-01:
What is the manaufacture number of your device ? visible in node info panel.
Are you sure your device support it ?

nasimon, 2021-01-01:
Yes, I could activate it using Mi Home (although the setting menu only appears sometimes). It's model LLKZMK11LM. In the node info it reports Manufacturer code 0x1037. generic.xml actually defines this field for 0x1037 code.
After retrying a little bit more, I changed 0x1037 in the field for 0x115f, added a comment and removed manufacturing code from attribute-set and then it started working:
            <attribute-set id="0xff00" description="Xiaomi Specific">
                <attribute id="0xff06" name="Interlock" type="bool" access="rw" required="o" mfcode="0x115f">
				    <description>Set to true to enforce only one switch is on at a time</description>
				</attribute>
            </attribute-set>

So, it doesn't make sense that it works with a different mfg code than the one reported but this is what I see. deconz 2.07.01, ConBee II with the latest FW 0x26680700.
Maybe is something special with that device?

nasimon, 2021-01-01:
```
            <attribute-set id="0xff00" description="Xiaomi Specific">
                <attribute id="0xff06" name="Interlock" type="bool" access="rw" required="o" mfcode="0x115f">
				    <description>Set to true to enforce only one switch is on at a time</description>
				</attribute>
            </attribute-set>

```