dresden-elektronik/deconz-rest-plugin #3374: OWON AC201 Thermostat (IR Blaster)

aryelevin, 2020-10-06:
## Device

- Product name: OWON AC201 Thermostat AC Control
- Manufacturer: OWON
- Model identifier: AC201
- Device type :  Thermostat

## Screenshots

[image]
[image]
[image]

This device is a thermostat which controls Air conditioners by emitting IR commands.
It features control on the mode and fan speed.
Using the deconz GUI its working perfectly, but it is not exposed to the REST API.
Also, it seems that the pairing process is unfinished since it not gets a name with the number of the sensor (its ID).

Mimiix, 2020-10-07:
Missing cluster screenshots.

aryelevin, 2020-10-07:
Which cluster is missing?
Attached the thermostat and fan control clusters:

[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-10-07:
How does the device work, can it heat and cool? Do you (need to) set the occupied cooling setpoint or do you just use the fan control cluster?

aryelevin, 2020-10-07:
Yes, I can set all three parameters:
Heating/cooling/auto/dry/etc mode (which is actually handled by the AC itself) -> Thermostat cluster
Target temperature (2 settings for cooling and heating each) -> Thermostat cluster
Fan mode (LOW/MED/HIGH/AUTO) -> Fan cluster

aryelevin, 2020-10-07:
The IR command which sent to the AC is always includes: Mode, Target Temp, Fan Speed.
So, any changes made to any of those values is immediately sends the up to date state to the AC via IR command.

SwoopX, 2020-10-07:
Ok, thanks. So that means we require 0x0011, 0x0012, 0x001B, 0x001C to be settable from thermostat cluster.

aryelevin, 2020-10-07:
I think that 1B has nothing to do with IR controlled Acs, but it is definitely good to allow it to be settable fro cases it might needed.
The fan control also needs to be settable in the Fan cluster.

SwoopX, 2020-10-07:
Right. Have excluded it as that was thermostat cluster related.

aryelevin, 2020-10-07:
Ah, OK. Didn't noticed it... 
Thank you very much.

aryelevin, 2020-10-12:
@SwoopX Just checking out if there is any news with it, and if its expected to be available with this upcoming beta at the 15th October?

Thanks

SwoopX, 2020-10-13:
That's gonna be though, not sure.

aryelevin, 2020-10-14:
Thanks a lot for the update...
It gives me a hope hahaha...

aryelevin, 2020-10-15:
@SwoopX  I see its not in this beta.
Do you have any progress on it which I can join? I'm a developer too (C/C++/JS/ObjC/Swift/Java/Go)?
In case not, do you have any guidelines to tell me where to start?

aryelevin, 2020-10-17:
Great! Thanks a lot!!!
Will give it a try and will report
Have a nice weekend

aryelevin, 2020-10-18:
@SwoopX OK, so, after testing a bit, the first problem is that the fanmode is not exists on the config object, I counlnt find the issue by myself, so you help is appreciated.
I've fixed small bugs, and continues to do so except this one, which I have no idea how to solve it , and after all solved, I'll create pull request of course.

Thanks a lot

aryelevin, 2020-10-18:
@SwoopX Also, After the device is added (which I need to manually read the Basic cluster manually by the GUI in order to trigger the finalization of the process), the On mode is always == "on", and the "mode" is always == null and never exits this state (the if statements is avoiding updating a null "mode", so it never gets out of this loop), and I didn't found where the default setup happens.
BTW, not only the fanmode is not available under config, there is no light resource created.

SwoopX, 2020-10-18:
@aryelevin Thanks for your feedback so far. It's really good to see that you can test the code and have the programming background to iron out the quirks. It's always hard to make some device inclusions remotely and not having the device at hand.

Gotta hund down a segfault first, but will chime in afterwards!

aryelevin, 2020-10-18:
I'm happy to assist and enjoy it a much.
I'm are simply don't know the structure well so getting lost quickly in the project.

aryelevin, 2020-10-18:
Hi @SwoopX ,

Here's my progress with getting it to somewhat working and also the fan is now exposed on the lights resources (still not working and needs adaptions to have the right data on it).
https://github.com/aryelevin/deconz-rest-plugin/commit/9afbf2c30c26fd6f69aedccffd3f0dd7ae4f776c

Please take a look and let me know

SwoopX, 2020-10-18:
@aryelevin  Can you please show me your current output from the REST API for the sensor? Having `NULL` values is kind of ok when the device has just recently been created. However, the current values should drip in after 10-15 mins as the device first needs to configure the attribute reporting and then wait for the reports to come in after the defined timeframes. The code is based on the assumption that reporting is supported for the interesting attributes. If it isn't, that could explain the values not changing and we need to switch over to polling them.

It would give us some further insights if you could manually set up the bindings and reporting for the newly added REST API values as explained [here](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...).

Also, I'm afraid your proposed changes could cause some unexpected behaviour for other devices, which is the reason I haven't taken that route. The idea was to have the fan mode included in the sensor resource and to no add an additional light resource just for that.

> the On mode is always == "on", and the "mode" is always == null and never exits this state (the if statements is avoiding updating a null "mode", so it never gets out of this loop), and I didn't found where the default setup happens.

Could you point me to that one?

aryelevin, 2020-10-18:
@SwoopX First of all, thank you very much.
Now, Here's the sensor object as received by the REST API:
{
    "config": {
      "coolsetpoint": null,
      "heatsetpoint": null,
      "mode": null,
      "offset": 0,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "87e0b281478062b1b22d4aac6976804c",
    "lastseen": "2020-10-18T17:30Z",
    "manufacturername": "OWON",
    "modelid": "AC201",
    "name": "AC201",
    "state": {
      "lastupdated": "none",
      "on": null,
      "temperature": null
    },
    "type": "ZHAThermostat",
    "uniqueid": "00:60:6e:ff:fe:97:bc:a7-01-0201"
  }
The device is much more than 15 minutes on the system.
I will look into you instructions in the link and will report back here.

Now, for the fan, I totally agree with you that it shouldn't add additional light resource, actually, I thought this is what you've trued to achieve, so I tried to replicate this.
I prefer it to work within the sensor object, but for weird reason the "fanmode" is dropped from the config object, do you have any idea why?

As for your last question, I will try to make it clearer here:
There is "on" key on the config object, and it is always = true, why? (I also referred to the issue with the NULL "mode", but leave this aside at the moment, since we have to check out whether its a attributes reporting issue or not).

aryelevin, 2020-10-18:
@SwoopX OK, I found the issue with the fanmode:
you defined it at a UInt8 value, but when adding It to the node, you specify it as a string, thus it rejected.
I'm going to revert my fan light Resource changes now.

SwoopX, 2020-10-18:
Thanks for coming back. I suppose an incorrect data type was the issue for `fanmode`. Already added a new commit for that to the PR.

> There is "on" key on the config object, and it is always = true, why? (I also referred to the issue with the NULL "mode", but leave this aside at the moment, since we have to check out whether its a attributes reporting issue or not).

Funny thing is, I asked that myself like 2hrs ago. I don't know for sure. It's either a relic or required for rules/scenes I'd say. Don't care about it too much.

aryelevin, 2020-10-18:
OK, so now it looks much better, but you have to merge this commit in order to make the "mode" working.
https://github.com/aryelevin/deconz-rest-plugin/commit/49d9c6b48048ee23d2a44957e19e92437443c632

Now will loointo the attributes reporting stuff.

Thanks

aryelevin, 2020-10-19:
@SwoopX OK, so after manually configuring the bindings for attributes reporting it seems to be working fine. (also configured the intervals etc.)
What does it means for now? issues with the default settings? or what?

SwoopX, 2020-10-19:
> OK, so now it looks much better, but you have to merge this commit in order to make the "mode" working.

Right, thanks!

> What does it means for now? issues with the default settings? or what?

Good question. So, when we're good fron the functionality perspective, I'd say you should try to reset and pair the device again. Please make sure to include the updated fixes PR into account. Usually, all the bindings and attribute reportings should be set up automagically. Even if missed, it should be retried every 1-2 hrs. Might also be a good idea to take logs from the join process and the time following. Given you use the GUI mode (ensure the services have been stopped before):
`/usr/bin/deCONZ --dbg-error=2 --dbg-info=2 > debug`

If you feel funny, we could also start exploring the unknown cluster. You need an additional plugin to discover the included attributes there, happy to guide you through.

aryelevin, 2020-10-19:
OK, I will have to check that out. Will do it later today.
As for the unknown cluster, I think I know what is it, its the IR configuration which allows you to configure the AC model you have and the type of the IR remote it uses.
I still want to expose that out, so lets go on on this haha..

SwoopX, 2020-10-19:
Alrighty. You need to compile this plugin https://github.com/ma-ca/deconz-cli-plugin, copy it to the deconz plugin folder and restart. How to connect is described in the repo. Then, issue the command below and share the output:

zclattrmanu YOURDEVICENWK 1 0xffac 0x113C 0C00000d
e.g. zclattrmanu 0x78c9 0xffac 0x113C 0C00000d

aryelevin, 2020-10-19:
@SwoopX I found some more attributes  in the thermostat cluster which I want to support (actually just the swing functions of the AC which is at 0x0045).
I created my general.xml with some of them.
My question is: Should I add all the attributes in this attributes set or I can just add the one I need for it to be merged here?

SwoopX, 2020-10-19:
Please add all, so we have them readily available

aryelevin, 2020-10-19:
OK, will do it and get back ASAP

aryelevin, 2020-10-19:
@SwoopX How to specify map32 data type in the xml?

aryelevin, 2020-10-19:
@SwoopX OK, never mind, found it: bmp32...
I've finished working on this, where is it in this repo?

aryelevin, 2020-10-19:
@SwoopX I've implemented the swing mode support in this commit, for me it seems to be working great.
https://github.com/aryelevin/deconz-rest-plugin/commit/3e31aaf50481ac506637af601ad081b4c15a401d

Please take a look and make sure it all fine, if so, just merge...

SwoopX, 2020-10-19:
I guess it's looking fine so far. You used s8 for the 8 Bit enumeration instead of u8, but that doesn't matter. Setting the attribute will probably not work though. Where have you discovered the attribute (which cluster), was it with my suggested command? Was that the only attribute?

aryelevin, 2020-10-19:
@SwoopX For me its working setting the attribute via the REST API (I'm using Homebridge for HomeKit integration), but I will look into what you're saying.
Its within the Thermostat cluster.
The way I found it was simple, I thought there must be an option to control the swing mode via Zigbee, so decided to look into Zigbee specs docs, and found something which for me was just seems related, so I gave it a try, and it worked.

SwoopX, 2020-10-20:
> so unless the number is not exceeding 127 it should work as expected, isn't?

Indeed, and therefore, it doesn't matter here.

> This attribute is within the Thermostat cluster, and there is some more attributes, but those aren't supported by my device and I don't need those functionalities (mostly info about the AC itself like the compressor type and refrigerant type etc).
The way I found it was simple, I thought there must be an option to control the swing mode via Zigbee, so decided to look into Zigbee specs docs, and found something which for me was just seems related, so I gave it a try, and it worked.

Ok, after checking the specs, I now know what you mean/did. I was assuming you did the discovery of the manufacturer specific attribues already and the attribute was from there.

aryelevin, 2020-10-20:
@SwoopX I will update the general xml.
To which file should I add it? the general xml in the root of this repo @ master?
Also, the plugin you instructed me to use is already installed in my system for a long time, since I used it to write the buttons configuration of the Ubisys D1 dimmer (This cannot change via the GUI for some reason, are this can be supported somehow?)

aryelevin, 2020-10-21:
@SwoopX Sorry to bother you... Just want to create the PR, and I want to make sure I'm doing it right, so again, my question is: which general xml file should I use to create the PR? the one in the root of this repo at the master branch?

aryelevin, 2020-10-21:
@SwoopX OK, So created my PR... Waiting for inputs about it...

Thank you very much for your support.

Now I will start looking into the secret cluster haha.

aryelevin, 2020-10-21:
@SwoopX OK, so, after trying your command (discovering the secret cluster's attributes), it seems it doesn't returns anything.
What could be the reason?

SwoopX, 2020-10-23:
Been busy lately.

> OK, so, after trying your command (discovering the secret cluster's attributes), it seems it doesn't returns anything.
What could be the reason?

Can you share the output? Eventuelly, the command didn't get recognized on the first time. It could also be possbile that the command is not implemented on the device.

Regarding the PR, we now have it in twice. Ok for you if I just take the necessary parts over in mine?

aryelevin, 2020-10-23:
> Been busy lately.
We're all the same... Don't worry... hahaha.. Thanks anyways.

> Can you share the output? Eventuelly, the command didn't get recognized on the first time. It could also be possible that the command is not implemented on the device.
I will do it again, I have much of noisy output duo to some plugs reporting their electrical consumption, so a lot of unrelated output.

> Regarding the PR, we now have it in twice. Ok for you if I just take the necessary parts over in mine?
Yes, sure, do it.

aryelevin, 2020-10-23:
@SwoopX Here's the output:
```
zclattrmanu 0xc506 1 0xffac 0x113C 0C00000d
 --> send OK
<-ZCL attribute report 0x0373 0x0B04 2 05 05 21 FFFFFF9A 59
<-ZCL discover attr 0xC506 for cluster 0xFFAC discoveryComplete = Yes
```
Please note the unrelated attribute report in the middle...
Thanks

aryelevin, 2020-10-26:
@SwoopX I've added another commit with Swing mode attribute reporting binding (forgot that thing).
(Had no chance to test actually, but just the same as the thermostat mode code, so should be fine).

aryelevin, 2020-11-01:
@SwoopX Just a ping.. ;-)

SwoopX, 2020-11-03:
Got your changes all ported to my PR ;)

aryelevin, 2020-11-12:
OK, so, after signed NDA to receive the secret cluster documentation, I got it, and as I guessed before, its related to the configuration of the thermostat itself (Like LED on/off and AC Remote type etc).
Cannot expose more than that duo to the NDA, so, sadly a big no no... hahaha.
Waiting for a final merge of this supported device as I'm using it now for a few weeks and it seems to be working great.

closed 2020-11-14 by manup

SwoopX, 2021-02-20:
@aryelevin Could you do me a favor and try setting the swing mode directly via API? Pretty sure it will not work since the code seems to be at the wrong place and therefore never called.

aryelevin, 2021-02-21:
@SwoopX You're right, now it is not working.
At the time we've worked on supporting it, it was working for sure, something changed it.
Can you point me to the issue? Or you working on a fix?

SwoopX, 2021-02-21:
I'll have that covered, thanks for checking. Accidentially stumbled accross that :)

aryelevin, 2021-02-21:
Great, thank you very much.
Regards

aryelevin, 2021-02-23:
@SwoopX I'm trying to implement the FFAC cluster commands, but I need to specify a manufacturer code for each command, how this can be achieved in the general.xml? 
I tried to use mfcode=0x113c but it not being included in the request (I sniff using zshark and Wireshark), so what can I do?

zhapwey, 2022-06-02:
Hi, I have seen this device and it seems very interesting, but I need to know if there is a way to configure the type of AACC so that it sends the correct ir commands, there is some way manually since from zigbee it does not seem to be implementing.

aryelevin, 2022-06-02:
@zhapwey I were able to configure it using a manufacturer command via the cli plugin and it worked perfectly.
If you need assistance on figuring out the command itself, let me know and I'll do my best to send you an example.

zhapwey, 2022-06-10:
hello:
I managed to sync the device with deconz without problems.
My problem is that I don't know how to send the code that you tell me, could you help me.

BR

aryelevin, 2022-06-10:
@zhapwey Here's my configuration I use.
You need to use it with the cli plugin on deconz.
You have to use the device's short ID (or whatever it called) etc.
I hope this helps you.

OWON AC201 set the remote code: 20 is the command and 4403 is the code which in this case 836 decimal.
For example: 836 which 344 in Hex:
zclcmdmanu 0xXXXX 1 0xffac 0x113c 204403
For example: 930 which 3a2 in Hex:
zclcmdmanu 0xXXXX 1 0xffac 0x113c 20a203
For example: 830 which 33e in Hex:
zclcmdmanu 0xXXXX 1 0xffac 0x113c 203e03
Gets the current code:
zclcmdmanu 0xXXXX 1 0xffac 0x113c 00

zhapwey, 2022-06-10:
thanks, Could you tell me how to access the cli plugin in deconz, I have no idea what this is?
Then I imagine that in that complement it will be necessary to write this line:

For example: 836 which 344 in Hex:
zclcmdmanu 0xXXXX 1 0xffac 0x113c 204403

zclcmdmanu -> command?
I see that the following parameters only change the last one? the rest are all the same?

BR, Thank you very much for your help it is much appreciated.

zhapwey, 2022-06-10:
Is it possible to use the cli on windows 10?