dresden-elektronik/deconz-rest-plugin #3375: SmartThings Motion Sensor (2018 model) report water no presence

martinezmp3, 2020-10-06:
deCONZ REST-API is reporting motion sensor (SmartThings Motion Sensor (2018 model) | IM6001-MTP01) as a water sensor 

## Describe the bug
the info received on the subscribe WS is the fallow:
when motion:
{"e":"changed","id":"19","r":"sensors","state":{"lastupdated":"2020-10-07T22:46:57.691","lowbattery":false,"tampered":false,"water":true},"t":"event","uniqueid":"28:6d:97:00:01:00:55:fd-01-0500"}
end of motion:
{"e":"changed","id":"19","r":"sensors","state":{"lastupdated":"2020-10-07T22:47:13.191","lowbattery":false,"tampered":false,"water":false},"t":"event","uniqueid":"28:6d:97:00:01:00:55:fd-01-0500"}

if sensor information is request "sensor/19"
this is the response: 
 {
  "config": {
    "battery": 94,
    "on": true,
    "pending": [],
    "reachable": true
  },
  "ep": 1,
  "etag": "68ed19dc930eb3eb38c3f0e72288053d",
  "lastseen": "2020-10-07T22:51:29.443",
  "manufacturername": "Samjin",
  "modelid": "motion",
  "name": "Motion Sensor",
  "state": {
    "lastupdated": "2020-10-07T22:47:13.960",
    "lowbattery": false,
    "tampered": false,
    "water": false
  },
  "type": "ZHAWater",
  "uniqueid": "28:6d:97:00:01:00:55:fd-01-0500"
}
## Steps to reproduce the behavior
none it start as soon you add the sensor 

## Expected behavior

state.water should be state.presence same value
## Screenshots

## Environment

 - Host system: (Raspberry Pi 4)
 - Running method: (Raspbian imagen download from https://phoscon.de/)
 - Firmware version: (26520700)
 - deCONZ version: (2.5.77)
 - Device: ( RaspBee II)
 - Do you use an USB extension cable: (N/A) -- only relevant for ConBee I/II

### deCONZ Logs

## Additional context

SwoopX, 2020-10-07:
Can you double check in deconz GUI as what the device exposes itself? Should be sufficient to read the attributes of IAS Zone cluster.

Could be that we should hardcode the sensor type.

martinezmp3, 2020-10-07:
attach screenshot temperature is working the way it should but i attach it to just in case you need it 

[image]
[image]
[image]
[image]

SwoopX, 2020-10-07:
Thanks. Are you sure you've read the attributes while the sensor was awake?

martinezmp3, 2020-10-07:
I'm sure, the sensor is reporting temperature ok and detecting the motion the problem is that it send status as water=true on motion and water=false on no motion

SwoopX, 2020-10-07:
Just taken another look at the screenshot. It was'n awake as the left dot is in red. You must wake the device (trigger it's function) and then immediately read the attributes to get anything meaningful at all.

martinezmp3, 2020-10-07:
[image]
this is the new one

SwoopX, 2020-10-08:
Hm, it's rare that a device uses the wrong type. Regardless, we can hardcode it in the next release, so should be fine then.

closed 2020-10-15 by manup