dresden-elektronik/deconz-rest-plugin #3377: New firmware versions online.

manup, 2020-10-07:
#### ConBee II

http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26660700.bin.GCF

#### RaspBee II

http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26660700.bin.GCF

#### RaspBee I and ConBee I

http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_Rpi_0x26390500.bin.GCF

* Fix frame counter not fresh enough after reboot or power-loss. This issue happened in some setups caused the network to be not reachable for a while (mostly 5–10 minutes sometimes hours).
* Fix ConBee I and RaspBee I PANID and Zigbee channel got scrambled under rare conditions.

octupole, 2020-10-12:
I tried to update directly from the web application on a ubuntu 18.04 LTS system (an intel Nuc). Unfortunately, the upgrade to 26660700 failed and made my Conbee II impossible to mount on the system on the serial port. I tried to use the GCFFlasher_internal command, but it still failed, giving an error by not completing the update numerous times before timing out. I resolved this by using a Windows 10 machine to do the firmware upgrade, see [here](https://flemmingss.com/how-to-update-conbee-conbee-ii-firmware-in-windows-10/). Was I the only one to have this problem?

Smanar, 2020-10-12:
So it have worked on windows but not Ubuntu (No VM or docker) ? You still have the command used ?

octupole, 2020-10-12:
Firmware upgrade has worked on Windows 10, not on Ubuntu 18.04 LTS. I didn't try the upgrade on docker or some VM system. Which commands? Now, everything works - the phoscon app - as before on my ubuntu box.

Smanar, 2020-10-12:
>Which commands?

The command used for update the firmware, with GCFFlasher_internal ?

octupole, 2020-10-12:
I just upgraded to the latest firmware version, there is no point for me to
try the GCFFlasher_internal command again. And most importantly, I don't
want to screw up the Conbee II by using that command again. Before I did
the upgrade on the Windows 10 machine I tried many times the command with
no success.

Claus1985, 2020-11-26:
What changes are included in the 26670700 release? 
Flashed this because I thought it was the newest stable release 🙈

SwoopX, 2020-11-26:
It resolved a rare situation where conbee II is stuck in permanent joining state, observable in deconz GUI.

Martinvdm, 2021-02-23:
As far as i can see, we are now on ; [26680700 ](http://deconz.dresden-elektronik.de/deconz-firmware/)
Any news from the release notes?

manup, 2022-10-07:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog

closed 2022-10-07 by manup