dresden-elektronik/deconz-rest-plugin #3379: After installing a new Raspberry OS, Lights, Switches and Sensors disappeared or not controlable.

Maddinthebrain, 2020-10-08:
## Describe the question or issue you are having

I had install a new raspbian as my SD-Card crashed. After installing the latest Deconz version, the lights, switches and sensors didn't work. I put on a older Backup which contains most of the devices. But only a few lights worked again. Most of them not. As for the switches and sensors. 

So I decided to reset the Gateway, which was a bad idea. Now it is completely broken. I put on again the backup, but not any light is working, nor a switch or sensor. Just 4 lights and one switch are appearing in Phoscon, but you cannot control them. No sensors or switches are appearing. In the deconz-gui this 4 lights are connected. The lights have a red blinking dot. One fitth light is there but with hexcode as name and without connection, it has no connection. The switch has no connection. 

Furthermore resetting devices for reconnection via phoscon does not work...

A update of the Conbee Sticks firmware doesn't work too. The update `"sudo GCFFlasher _internal -t 60 deCONZ_Rpi_0x263905000.bin.GCF"` tries to reboot the stick 10 times, but fails. 

What is the problem here? Any suggestions?

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi4B
 - Running method: (Raspbian Buster)
 - Firmware version: (26350500)
 - deCONZ version: (2.05.81)
 - Device: ConBee I 
 - Do you use an USB extension cable: yes, a USB-Hub (it worked before without issues)

### deCONZ Logs

How to make the log files?

## Additional context

About 10 Lights, 5 Switches, 10 Sensors

S474N, 2020-10-08:
First try update ConBee firmware, now is version 26390500 (file deCONZ_Rpi_0x26390500.bin.GCF):
http://deconz.dresden-elektronik.de/deconz-firmware/

Here is how to step by step:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

You must stop deCONZ service and try also ConBee directly to USB without hub.

Maddinthebrain, 2020-10-08:
OK, I have interesting news. For firmware flashing reasons I put the conbee stick into a windows 10 computer as it didn't work on the raspi. And suddently the know lights were working again. After reconnecting all devices everything is fine. The firmware is now the latest version and the conbee stick is in the windows PC. 

Then I made a backup of the configuration and put it back into the raspi after deinstall and reinstall deconz. Then I put the backup I made shortly before back on. And it didn't work. Hey that's not fun! What is the problem here?

I put it into my PC and it worked without any issue. What is interesting: The behaviour of building up the connections is different. On the PC its more a mesh. On the raspi it connects only to 4 or 5 devices directly, but the devices cannot be controled...

I guess there is an issue on the raspi. Maybe I have install a clean raspi a further time... Some other ideas? While installing deconz it tells me to remove something in /boot/cmline.txt (serial=enable???) and to add someting in /boot/config.txt (enable=uart???) which I made. The exakt words I don't remember. I followed the advices and made reboot. But without success with the connection issue.

Smanar, 2020-10-09:
All stuff about serial or UART are for the raspbee.

For the distance I don't think, as they are routeur, so they can't perturb each other for me.

The power supply ? you are using the usb hub on both ? SSD device ? (avoid usb3.0)

Try to wait some time, but BTW your first capture is realy bad, so much name are grayed.

You can try to run deconz in debug mode, to look for error message
`deCONZ --dbg-error=2 --dbg-info=2 --dbg-ota=0 --dbg-zdp=0 --dbg-zcl=0 --db-aps=0 --dbg-http=0`

closed 2020-11-09 by stale[bot]