dresden-elektronik/deconz-rest-plugin #3380: Device name

spunti, 2020-10-08:
## Device

- Product name: Single Switch Module T1 (with Neutral)
- Manufacturer: LUMI
- Model identifier: lumi.switch.n0agl1
- Device type : Please remove all unrelated device types. 
 - Switch
  

## Screenshots

[image]
[image]
[image]
[image]

[aqara-t1-single-switch-connecting-and-triggering.pcapng.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5350391/aqara-t1-single-switch-connecting-and-triggering.pcapng.zip)

[zigbee_pc_keys.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5350406/zigbee_pc_keys.txt)

Smanar, 2020-10-09:
Ha nice ^^
I have started a discussion yesterday with another user about this device, there is a problem ATM for this device, the cluster 0x0006 is invisible (but present).

Do you have a zigbee sniffer ?

Smanar, 2020-10-11:
Ok so have a take a look on the sniff with original gateway
- bad luck it don't ask for basic descriptor to compare (it s the device that ask for node descriptor)
- it is using a new attribute 0x00f7 an octet string, but nothing readable.

SwoopX, 2020-10-11:
That's attribute 0x00f7

```
64 10 00        Boolean                     false
03 28 17        Signed 8-bit integer        23
98 39 00000000  Single precision            0
95 39 00000000  Single precision            0
96 39 90391145  Single precision            2323.59766
97 39 00000000  Single precision            0
05 21 0100      Unsigned 16-bit integer     1
9a 20 00        Unsigned 8-bit integer      0
0b 20 00        Unsigned 8-bit integer      0
08 21 1402      Unsigned 16-bit integer     532
09 21 0001      Unsigned 16-bit integer     256
```

Smanar, 2020-10-12:
Ha yes lol, the magic attribute ^^.

Smanar, 2020-10-12:
Ok so bad luck, the device realy forget the cluster 0x0006

[image]

And it was totaly mute after the inclusion, I realy don't see a way to detect the cluster 0x0006.
It was more active on normal gateway than on deconz.

I think need to use hack based on device type "on/off light" and manufacture number 0x115f ?

spunti, 2020-10-15:
Any news? Or can I be helpful with something else?

Mimiix, 2020-10-15:
@spunti I've read about this in Discord. The devs had a internal discussion on this. Manup responded with the following:
[image]

Smanar, 2020-10-15:
Yep, for the moment we are waiting to know if manup want to make something clean in the core, or if I make an ugly hack in the api core ^^

spunti, 2020-11-03:
Any progress from manup with fixing this device? Would be nice to be finally able to control the device via HA :)

Smanar, 2020-11-05:
No yet sorry, but still in the "todo list"

manup, 2020-11-08:
Is there a sniffer log when the device sends on/off reports for endpoints other than the first (0x01)?

manup, 2020-11-08:
> `08 21 1402      Unsigned 16-bit integer     532` 

By looking at the special attribute 0x08 uint16, I think this represents a model identifier.

Smanar, 2020-11-08:
But the problem is to make the cluster 0x0006 "magically" appear in deconz, for the device can be usable in the API without hack.
Its something possible on deconz side with respecting the "zigbee rules" ?

flopp999, 2020-11-23:
I can also maybe help with something, just ask me and we will see what I can do

Smanar, 2020-11-23:
IDK if there is some improvement on deconz side or if we can do something ? (With using new descriptor engine ?)

labaland, 2020-11-26:
Any progress with fixing this device? Would be nice to control the device via HA

flopp999, 2020-12-09:
I have never used deConz GUI before, but today I had a look and I can see the WATT for this device.
I attached two pictures I don't know if that can help you.
[image]
[image]

Smanar, 2020-12-09:
I can see to the cluster 0006.
It was missing on @spunti capture, It s repared ?

The device works for you ?

flopp999, 2020-12-09:
I do get new values in 0x050B if I turn off/on the T1.
But I do not see any device in API.
This is as a Light, but my other Aqara Plug(different device) is as a sensor and then I get Consumption and Power

Aqara T1 build-in "box"
```
"manufacturername":"LUMI",
"modelid":"lumi.switch.n0agl1",
"type":"On/Off light"
```

Aqara Plug outlet
```
"manufacturername":"LUMI",
"modelid":"lumi.plug.maeu01",
"type":"ZHAConsumption"
```

Smanar, 2020-12-09:
>But I do not see any device in API.

Ha ? but it  s a ight, the support is native ....

flopp999, 2020-12-09:
The outlet plug is also under Lights
```
"manufacturername":"LUMI",
"modelid":"lumi.plug.maeu01",
"type":"Smart plug",
```

so what I am missing for T1 is to add it as a sensor, like Plug

flopp999, 2020-12-09:
Light section:
T1 On/Off
Plug On/Off

Sensor section:
Plug as Consumption
Plug as Power

Smanar, 2020-12-10:
So https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

```
git clone --branch Xiaomi https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

It s just a start, just make a test and tell me what is working or not, I will correct later.

flopp999, 2020-12-11:
Tried it now, but I do not see any new devices in API.
I have checked that I use the correct commit 
[image]

flopp999, 2020-12-11:
It is working now, I deleted the device and add it again, now I do get some new devices

Thank you :)

MiAutomations, 2020-12-15:
> It is working now, I deleted the device and add it again, now I do get some new devices
> 
> Thank you :)

I just deleted the mine and re-added using the last versions available,  and I can't get the report of the electricity values. 

Can you please share with me what is the version of your deconz and the fw version of your conbeeII

flopp999, 2020-12-15:
deconz GUI 2.07.01
phoszon GW 2.07.00
conbee FW 26660700

MiAutomations, 2020-12-16:
Thank you for your reply 

But I still have no report dada from the electricity... Do you install the  deCONZ development package present in this post? 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3380#issuecomment-742687126

flopp999, 2020-12-16:
> Thank you for your reply 
> 
> 
> 
> But I still have no report dada from the electricity... Do you install the  deCONZ development package present in this post? 
> 
> 
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3380#issuecomment-742687126

Yes I have

MiAutomations, 2020-12-16:
And if I'm using the hassio OS how can perform this installation ? 
I think I can't, once I don't have core linux available .... maybe somebody can help me in this point ...???

Smanar, 2020-12-16:
Hassio is not a real OS.
You need one to compile or install something.

MiAutomations, 2020-12-18:
> Hassio is not a real OS.
> You need one to compile or install something.

Ok thank you, and can you please let me  know how to proceed with this, in order to get this to work?

Thank you

Smanar, 2020-12-18:
@flopp999 Can you say me the sensor that are working and the one are not working ?

@MiAutomations depend of what you have as Unix machine (with complete OS). 
If you have one, just install deconz on it and after the dev stuff https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3380#issuecomment-742687126

flopp999, 2020-12-18:
> @flopp999 Can you say me the sensor that are working and the one are not working ?
>

Aqara wall Plug is working

Aqara T1, that you change rest-plugin for, is not working.
I had to comment-out to get it to update

Smanar, 2020-12-18:
Ha ? ok so I thought it was working because of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3380#issuecomment-743234187

You had to comment out ?  else you have error message ? wich one lines  ?

SwoopX, 2020-12-18:
I feel those devices have a comparable strange behaviour as the `lumi.plg.mmeu01`. Most of the time, they don't show the simple metering and electrical measurement clusters and then out of a sudden, they pop up simple descriptors due to whatever reason. This theory would be supported by the deviating screenshots here.

Never was able to find any pattern or reason for that...

spunti, 2020-12-22:
For me it would be ok, if the on/off operation is supported as first step :)

MiAutomations, 2020-12-27:
@SwoopX Mimiix Can you please let me know if you still need more info about the device in order to get this to work ? 

Thank you and best regards

spunti, 2020-12-29:
@MiAutomations I have checked your created issue about this switch. There you are writing, that on/off is already working? What have you done to get it working?

MiAutomations, 2020-12-30:
> here you are writing, that on/off is already working? W

Yes the relay on/off functions are working properly, I have a issue only in get the data from the electricity report for this device once this one support the power consuming report also, but is not available over the deconz/conbeII

spunti, 2020-12-30:
> > here you are writing, that on/off is already working? W
> 
> Yes the relay on/off functions are working properly, I have a issue only in get the data from the electricity report for this device once this one support the power consuming report also, but is not available over the deconz/conbeII

Please explain what you have done to get it working. Are you using the dev build from @Smanar?

spunti, 2021-01-05:
@MiAutomations could you please give information about your deconz version? And please let me know, how you have connected the t1 switch to deconz. For me the switch only appears in the software, not at the web interface.

Mimiix, 2021-01-05:
@SwoopX @Smanar What do we need for measurement? Or is this simply a _lie_ of the devices?

spunti, 2021-01-05:
As first step it would be cool to have the on/off operation working. It seems to work for one user, but not for me. The switch is not visible in Phoscon (I assume that it should appear under the light section). 
deCONZ Version: 2.07.1

ebaauw, 2021-01-07:
> but some user have the cluster 0x0006 visible and other not.

I think deCONZ should show the cluster when it sees a message from it, even if it isn't returned on the _Simple Descriptor_.  You probably need to turn the relay on through a wired in-wall swich before it sends an attribute report for the _OnOff_ attribute.

> What do we need for measurement? Or is this simply a lie of the devices?

That's probably reported through the _Analog Input_ clusters.  The second endpoint will be (current) power, the third (life-time) consumption, if they follow the logic of other Aqara wired devices.  The _Multistate Input_ on the fourth endpoint likely reports button actions of the wired switch.  You might want to check if the switch input can be detached from the relay.

spunti, 2021-01-08:
> I think deCONZ should show the cluster when it sees a message from it, even if it isn't returned on the _Simple Descriptor_. You probably need to turn the relay on through a wired in-wall swich [...]

I'm turning it on and off multiples times every day. Or what do you mean exactly? For me the 0x0006 cluster is not visible (see screenshots from initial post).

Smanar, 2021-01-08:
You are using a switch to turn it on or off ?

spunti, 2021-01-08:
> You are using a switch to turn it on or off ?

Yes, it is connected to a switch.

Smanar, 2021-01-08:
@ebaauw do you think if I use a hack (one more) to force the light entry be created according to the model id, the API code will work ?

MiAutomations, 2021-01-08:
> you please give information about your deconz version? An

Helo, the version is now Current version: 6.6.2, and in my case appears on the web interface deconz when i search for new device and also in the Home Assistant entities but obly as a switch not the information about the electricity data

SwoopX, 2021-01-08:
Ok, I've taken another look at the sniffer log initially provided. The switch definititely seems to send attribute reports from the on/off cluster and reacts on those commands...

[image]

Question now is why deconz doesn't pick this up or doesn't represent that cluster in the GUI respectively.

Smanar, 2021-01-08:
IDk, realy strange, manup have said me tha same thing than ebaauw, and was correctly recently in core from my memory (have same problem for invisible tuya cluster).
From manup if the cluster is know (in the general.xml file) it can be added itself when deconz see it.

SwoopX, 2021-01-22:
Can anybody please read the attributes from the lumi specific cluster in deconz GUI and share a screenshot? I might have stumbled upon something accidentally I'd like to verify.

Smanar, 2021-01-23:
Oups, I totaly forget this issue.
I m trying a hack for the invisible cluster https://github.com/Smanar/deconz-rest-plugin/commit/5f6acaf24350e65ac807d32ee4d4dbdd1447bc9b#diff-d0aaee66bc2f60fdd243c0e5b86aebca01fa3eb36c7ec39578d5beeca47078ab

Do you think it can work ?

SwoopX, 2021-01-23:
No need to do any magic if my theory is right. It is probably related to attribute 0x0009 on that cluster. When it is set to 0, it seems to supress a significant amount of reporting messages, including on/off

Smanar, 2021-01-23:
Ha ? can be realy nice information.

spunti, 2021-01-25:
> Can anybody please read the attributes from the lumi specific cluster in deconz GUI and share a screenshot? I might have stumbled upon something accidentally I'd like to verify.

What do you mean with "lumi specific cluster"?

SwoopX, 2021-01-25:
When viewing the node in deconz GUI, there is a cluster called like that. I need a screenshot with the attributes being read

spunti, 2021-01-25:
@SwoopX 
[image]

SwoopX, 2021-01-25:
Yeees :)

Now double click on the value of  attribute 0x0009 and set it to 1 and cross fingers to let some magic happen. You might need to discinnect the device from power for it to work afterwards.

[26 more replies]

closed 2021-06-16 by github-actions[bot]