dresden-elektronik/deconz-rest-plugin #3382: Add the ability to adjust cooling setpoints of zigbee thermostats

andynbaker, 2020-10-08:
## Feature request type

Extend the REST-API to allow setting of the cooling setpoint of zigbee thermostats, specifically the Zen-01.
## Description

Full disclosure, I have no idea what I'm doing here. I have a conbee II and run the deCONZ Home Assistant add-on and integration. I also have three Zen-01 thermostats that control traditional combined central heating and cooling HVAC systems, (so both HEAT and COOL modes are supported). I have the thermostats successfully paired, and by [tweaking the deCONZ home assistant integration,](https://github.com/home-assistant/core/issues/41182#issuecomment-703255600) I can successfully change the mode between heating and cooling.

However, it appears the deCONZ REST-API (and therefore the deCONZ HA integration) cannot control the cooling setpoint. In cooling mode, I cannot change the setpoint. However, in heating mode I can change the setpoint. As I posted [here](https://github.com/home-assistant/core/issues/41182#issuecomment-705759416) in the HA integration, it appears the REST-API does not currently support adjusting the cooling setpoint?

## Considered alternatives

I have run the ZHA Home Assistant integration instead of deCONZ, using the conbee II, and was able to successfully pair the thermostats and control the cooling setpoints. However, I'd prefer to use deCONZ instead of ZHA and would love to get this feature  carried over.
## Additional context

SwoopX, 2020-10-08:
I got this already waiting locally to be pushed. I expect this to be available with the beta next week.

Does the zen has a fan control cluster btw?

andynbaker, 2020-10-08:
Yay! And yes, I think it does have a fan control cluster. At least, using the ZHA integration in Home Assistant I was able to change the fan from auto to on (and it was a really nice feature and enabled me to set up circulation routines when the HVAC system was idle for an extended period of time).

Does this help?

[image]
help?

SwoopX, 2020-10-09:
Yeah, that helps. Can you please double check if you got a light resource for steering the fan? If not, try searching for lights and change the fan mode.

andynbaker, 2020-10-09:
Just checked again. I do not have any extra lights representing the fan. I just removed and repaired one to be sure and no lights popped up.

SwoopX, 2020-10-24:
Just a quick update: Haven't forgotten about that, but I need to wait for other PRs to be merged first before I can bring this one in.

andynbaker, 2020-10-24:
You da man. Thanks for taking care of it. As soon as you guys push this out, I can work with the guys who manage the home assistant integration to add the function there.

andynbaker, 2020-11-28:
Hey @SwoopX! I pulled the latest deconz docker image and read through the rest API docs so that I could test this out. As of now, I can both change the thermostat mode to cool and the fanmode to on/auto without an issue with this message body and endpoint:

```
http://[MY_API_IP_AND_PORT]/api/[MY_USERNAME]/sensors/[MY_THERMOSTAT]/config
{"mode":"cool"}
```
or
```
http://[MY_API_IP_AND_PORT]/api/[MY_USERNAME]/sensors/[MY_THERMOSTAT]/config
{"fanmode":"on"}
```
But when I try to change the coolsetpoint I always get an error response:

```
http://[MY_API_IP_AND_PORT]/api/[MY_USERNAME]/sensors/[MY_THERMOSTAT]/config
{"coolsetpoint":"2350"}

[{"error":{"address":"/sensors/2/config/coolsetpoint","description":"invalid value, 2350, for parameter coolsetpoint","type":7}}]
```
Am I doing something wrong here?

CC @Kane610

SwoopX, 2020-11-29:
Found the issue. I added a type check for that parameter which expects the value to be numeric. However, it seens to arrive as string. Not too sure if I can find the why on short notice before the next release in some hours. I doubt, I remove the check for the time being.

closed 2020-11-30 by manup

SwoopX, 2020-11-30:
@andynbaker Hey, could you please let me know how you've send the cool point temperature value? I just made a test with setheatpoint and the data type is, as expected, double when I send it.

andynbaker, 2020-11-30:
I'm not sure I understand the question exactly. Maybe this helps: I'm using node-red to make a simple http request. Per the documentation here, I'm attempting to adjust the cool set point with the coolsetpoint config attribute:

[image]

Here's my PUT request endpoint: `http://[MY_DECONZ_IP_AND_PORT]/api/[MY_USERNAME]/sensors/[MY_DEVICE]/config` and here's my message body: `{"coolsetpoint":"2350"}`. This is the response I get: 

```
address: "/sensors/2/config/coolsetpoint"
description: "invalid value, 2350, for parameter coolsetpoint"
type: 7
```
Is this what you're after?

SwoopX, 2020-11-30:
I guess so. Please try without "" around the temperature value

andynbaker, 2020-11-30:
That was it! I'm so used to wrapping everything in node-red that I didn't realize I could even do it without. OK so I can confirm that I can in fact change the mode and the coolsetpoint per the documentation. You're the man.

SwoopX, 2020-11-30:
Great, riddle solved. Please note, I will re-enable the type check with the next release in 2 weeks.

solvisit, 2021-02-27:
It is possible to add the same coolingsetpoint to Bitron 902010/32, it has the ablity to switch mode from "heat" to "cool".
For enabling both modes it is necessary to set 0x001B (Control Sequence of Operation) of thermostat cluster to 0x04  (Cooling and Heating 4-pipes) before changing 0x001C (System Mode) to 0x02 (Cool)

solvisit, 2021-02-28:
I will open a new feature request, maybe is the best way to resolve the issue.