dresden-elektronik/deconz-rest-plugin #3383: Adurosmart ERIA Smart Wireless Dimming Switch Remote 81825

stevenwfoley, 2020-10-08:
## Device
- Product name: ERIA Smart Wireless Dimming Switch Remote
- Manufacturer: ADUROLIGHT
- Model identifier: Adurolight_NCC
- Device type: Switch
https://adurosmart.com/product/eria-smart-wireless-dimming-switch-remote/

## Screenshots
[image]
[image]

mdolnik, 2020-10-11:
FYI there is an closed / stale issue for this remote (#1730) which never got resolved.

I'm willing to make changes to the source code and test with my Eria remote if someone with more expertise can chime in and suggest what would need to be changed.

Mimiix, 2020-10-12:
Would like to continue in #1730 ! :)

closed 2020-10-12 by Mimiix