dresden-elektronik/deconz-rest-plugin #3389: Is it possible to change a/the Device ID

Gammenon, 2020-10-10:
Hi everybody,

I unintentionally deleted a sensor which drove me crazy cause of reoccuring pairing problems.
For now I got it reconnected but it has a new ID.
Former 34, now 36.
Is there any way to change the id by hand (over the api) to get the sensor back to its former place?

Otherwise I have to modify all iobroker scripts (and grafana dashboards) usings this sensor.

Thanks in advance ;)

ebaauw, 2020-10-10:
It’s bad practice to base your scripts on the resource IDs of Zigbee devices.  Better use the mac address from the `uniqueid`.

Gammenon, 2020-10-10:
Unfortunatly the deconz iobroker adapter labels the pulled info by the id, so no choice here ;)
Is there any documentation for the way you suggest?

Gammenon, 2020-10-10:
oh, I figured out how to connect to the deconz DB and noticed, the "old" device is still in there with a "deleted" state. Would it be possible just to switch it back to "normal" and delete the "false" new one instead.
And just to have things clean, remove this then deleted item sid=36? Or is there an autoincrement?

Mimiix, 2020-10-12:
Hi! 

As you are not using the template, i am closing this issue. Feel free to open a new one.

closed 2020-10-12 by Mimiix