dresden-elektronik/deconz-rest-plugin #3390: Successful updates to deCONZ_ConBeeII_0x26660700.bin.GCF ?? Anyone?

Sjonnie2018, 2020-10-10:
I am wondering if there are (any?) positive/successful upgrades to firmware deCONZ_ConBeeII_0x26660700.bin.GCF? All I read over here are bugs and problems. Much of these related to specific combinations of hard- and software.

I have a habit of upgrading soft- and firmware for my devices and equipment but would like to know what I get into.  So I would like to know who has made a successful upgrade to the latest firmware and how you went about.

Greetz,

Sjonnie

Smanar, 2020-10-10:
Yes, there is some success ^^.

I can't say if this one is better for you, and to update it just avoid virtual machine, and you can rollback if you have problem with it.

And BTW try to use the template, even for short question, moderator are trying to clean the github.

Sjonnie2018, 2020-10-10:
Hi Smanar/Thorgal,

Thanks for your reply. I would have liked to stick to the template but couldn't phrase my question in the correct way. Sorry!

I am eager to learn of the experiences of others in this matter!

Greetz,

Sjonnie

Smanar, 2020-10-10:
For short question try the discord > https://discord.gg/QFhTxqN
Will be faster for you.

Sjonnie2018, 2020-10-10:
Hi Smanar,

Yet another place? Another registration, another password and another place I have to monitor to get information?

Registered of course but I kinda would have liked it if it was possible to share information here on github...

Regards,

Sjonnie

SwoopX, 2020-10-10:
Uodated 3 devices, no issues (and never had any).

Btw, your choice of choosing the channel. Discord is faster and more interactive. Github is for development and unfortunately often misused as a forum 🤷‍♂️

Sjonnie2018, 2020-10-11:
OK, I will close this issue. Thanks for your replies.

closed 2020-10-11 by Sjonnie2018