dresden-elektronik/deconz-rest-plugin #3391: light group switches off and on again

pep4n, 2020-10-10:
Hi deconz Team,

please see problem below. 

How would you troubleshoot this? 

Thanks,
p

## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: NAS Synology DS918+
 - Running method: Marthoc Docker container
 - Firmware version: (26580700)
 - deCONZ version: (2.05.82)
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs

## Additional context

Smanar, 2020-10-10:
Hello, something you can check when the bulb have unwanted state, it's if the state is the same on phoscon

ioBroker no have log at all, or no log during the problem ? if you send order the command is visible on log ?

pep4n, 2020-10-10:
Hi thanks for getting back @Smanar !  

> Hello, something you can check when the bulb have unwanted state, it's if the state is the same on phoscon

ok will check that. 

> ioBroker no have log at all, or no log during the problem ? if you send order the command is visible on log ?

ioBroker has a log but no item during the problem. 
no, normal switching is not logged. but scripts show up there when executed.

pep4n, 2020-10-10:
I checked phoscon while this occurred: yes they show off in phoscon UI also.

Smanar, 2020-10-10:
I think the logs can be realy usefull in your situation ...

It s possible tu run deconz with debugging enabled only for http, but idk if it s possible on docker.
And I don't know at all ioBroker.

We need to check if there is not something that trigger a command.

Do you have scene in phoscon ? a remote in the group ? Can you make a try without ioBroker  ?
If a bulb is outside the group, it s not impacted ? You have the problem only for this group ?

SwoopX, 2020-10-10:
You could check the REST API for the devices in question. If the timestamp for `lastannounced` fits the on/off action experienced, that might be an indication for the device got hung and reboot. I recall we had such a case.

pep4n, 2020-10-10:
Thanks guys! I'd like to provide logs. Only it seems not that easy to pull. 
The possibilities are extensive: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-switches 

What command should I put to receive sufficient output? And where will the file be saved? 

@Smanar I have scenes and remote in that group. 
Bulbs outside the group are not affected. Its only this group. 3 Ikea tradfri bulbs.

Smanar, 2020-10-11:
I think the usefull witch is "dbg-http" 

The file will be saved at the path you are making the command.

But for me the easier way is finding the log in ioBroker.

An easier thing you can do too, is using a websocket client in your browser (there is some plugin for that), with that you will see all websocket send by deconz. You will not see the command send to deconz by ioBroker (if there is one) but you can see if a scene is triggered.

On my side Im using on firefox "Browser Websocket Client" with the url "ws://192.168.1.1:8088"

pep4n, 2020-10-11:
I'll try to figure out the logs this evening. 

In ioBroker there is nothing in the log re deconz when this appears. 

regarding the websocket I don't get any connection. 
I tried ws://192.168.20.3:8088 (deconz), ws://192.168.20.2:8088 (ioBroker) and ws://192.168.20.1:8088 (gateway) 
i guess this is due to my noobness ,)

Smanar, 2020-10-11:
It s the same ip you are using for Phoscon and the websocket port is probably configured somewhere in iobroker.

pep4n, 2020-10-11:
If bridge port is WS port it should be ws://192.168.20.3:8080, right? 

[image]

Smanar, 2020-10-11:
Sure for the IP, but for the port ....
Deconz use 2 port, one for REST request and 1 for websocket.

The first one is visible here 
https://dresden-light.appspot.com/discover

pep4n, 2020-10-11:
is 8080. 
dont get a connection. 
anyway to find out the other port?

pep4n, 2020-10-11:
> 
> 
> You could check the REST API for the devices in question. If the timestamp for `lastannounced` fits the on/off action experienced, that might be an indication for the device got hung and reboot. I recall we had such a case.

thanks for the hint. 
How would I check this? 
Thanks,
p

Smanar, 2020-10-11:
Same answer for both ^^ https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

To see the websocket port > `http://IP:PORT/api/API_KEY/config`
To see the light information > `http://IP:PORT/api/API_KEY/lights`

But I m sure you already have all information in IoBroker plugin
In version 2.0.3 > "added websocket port info to configuration"

The Json is visible too on log.

pep4n, 2020-10-11:
first link shows this: 
[image]

second link seems not possible:
[image]

pep4n, 2020-10-11:
first link has the websocket port: 8443

second link gives this output: 
the lights in question have a KUE_ in the beginning. 

`{"1":{"etag":"e9092158f9709a3b130a2d2f7999bc06","hascolor":false,"lastannounced":null,"lastseen":"2020-10-11T17:54Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x26580700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:ea:ea-01"},"2":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"etag":"fff40c100460434ca7e24fb9b04029c3","hascolor":true,"lastannounced":"2020-10-05T17:17:50Z","lastseen":"2020-10-11T17:55Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WS opal 1000lm","name":"WHZ_vorne","state":{"alert":"none","bri":253,"colormode":"ct","ct":454,"on":true,"reachable":true},"swversion":"2.0.022","type":"Color temperature light","uniqueid":"58:8e:81:ff:fe:17:3c:84-01"},"3":{"etag":"d5b1c44e68e46da9ba27727e80b128fb","hascolor":false,"lastannounced":"2020-10-10T22:10:10Z","lastseen":"2020-10-11T17:56Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW 806lm","name":"KUE_Fenster","state":{"alert":"none","bri":254,"on":true,"reachable":true},"swversion":"2.1.022","type":"Dimmable light","uniqueid":"cc:cc:cc:ff:fe:90:f5:6f-01"},"4":{"etag":"fe03320478020a9e7c8b701386d9b398","hascolor":false,"lastannounced":"2020-10-05T17:10:08Z","lastseen":"2020-10-11T17:57Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW clear 250lm","name":"WHZ_oldschool","state":{"alert":"none","bri":254,"on":true,"reachable":true},"swversion":"2.1.022","type":"Dimmable light","uniqueid":"cc:cc:cc:ff:fe:89:c9:42-01"},"5":{"etag":"e868605dcfa939c4e0494cf8dfeb0ef3","hascolor":false,"lastannounced":"2020-10-10T22:10:10Z","lastseen":"2020-10-11T17:53Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW clear 250lm","name":"KUE_mitte_oldschool","state":{"alert":"none","bri":254,"on":true,"reachable":true},"swversion":"2.1.022","type":"Dimmable light","uniqueid":"14:b4:57:ff:fe:49:e0:61-01"},"6":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"etag":"f64eeec4fe613b59bcaf3ae2d4d2bf00","hascolor":true,"lastannounced":"2020-10-05T21:36:05Z","lastseen":"2020-10-11T17:53Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E14 WS opal 600lm","name":"WHZ_hinten","state":{"alert":"none","bri":253,"colormode":"ct","ct":370,"on":true,"reachable":true},"swversion":"2.0.022","type":"Color temperature light","uniqueid":"84:2e:14:ff:fe:75:eb:69-01"},"7":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"etag":"5b2dac0c52ca82080e16f80e8b747d7e","hascolor":true,"lastannounced":"2020-10-09T16:31:03Z","lastseen":"2020-10-11T17:54Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WS opal 1000lm","name":"KUE_Wand","state":{"alert":"none","bri":254,"colormode":"ct","ct":454,"on":true,"reachable":true},"swversion":"2.0.022","type":"Color temperature light","uniqueid":"bc:33:ac:ff:fe:85:98:14-01"}}`

SwoopX, 2020-10-11:
One of the lights has rebootet today, the remaining 2 yesterday, based on the `lastannounced` timestamps. Does that correspond with your experience?

pep4n, 2020-10-11:
> One of the lights has rebootet today, the remaining 2 yesterday, based on the `lastannounced` timestamps. Does that correspond with your experience?

hmm not really happens much more often. 

It even gets more weird now as one (only one) bulb from another group seems to "hang" as well. off and then on again. 

I left the websocket open since it was connected approx. 2hrs ago. 
See log as follows:

`{"attr":{"lastannounced":null,"lastseen":"2020-10-11T18:03Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","swversion":"0x26580700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:ea:ea-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:05:ea:ea-01"}

{"attr":{"lastannounced":null,"lastseen":"2020-10-11T18:04Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","swversion":"0x26580700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:ea:ea-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:05:ea:ea-01"}

{"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"lastannounced":"2020-10-05T17:17:50Z","lastseen":"2020-10-11T18:04Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WS opal 1000lm","name":"WHZ_vorne","swversion":"2.0.022","type":"Color temperature light","uniqueid":"58:8e:81:ff:fe:17:3c:84-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"58:8e:81:ff:fe:17:3c:84-01"}

{"attr":{"lastannounced":"2020-10-10T22:10:10Z","lastseen":"2020-10-11T18:05Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW 806lm","name":"KUE_Fenster","swversion":"2.1.022","type":"Dimmable light","uniqueid":"cc:cc:cc:ff:fe:90:f5:6f-01"},"e":"changed","id":"3","r":"lights","t":"event","uniqueid":"cc:cc:cc:ff:fe:90:f5:6f-01"}

{"attr":{"lastannounced":"2020-10-05T17:10:08Z","lastseen":"2020-10-11T18:05Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW clear 250lm","name":"WHZ_oldschool","swversion":"2.1.022","type":"Dimmable light","uniqueid":"cc:cc:cc:ff:fe:89:c9:42-01"},"e":"changed","id":"4","r":"lights","t":"event","uniqueid":"cc:cc:cc:ff:fe:89:c9:42-01"}

{"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"lastannounced":"2020-10-05T21:36:05Z","lastseen":"2020-10-11T18:06Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E14 WS opal 600lm","name":"WHZ_hinten","swversion":"2.0.022","type":"Color temperature light","uniqueid":"84:2e:14:ff:fe:75:eb:69-01"},"e":"changed","id":"6","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:75:eb:69-01"}

{"attr":{"lastannounced":"2020-10-10T22:10:10Z","lastseen":"2020-10-11T18:06Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW clear 250lm","name":"KUE_mitte_oldschool","swversion":"2.1.022","type":"Dimmable light","uniqueid":"14:b4:57:ff:fe:49:e0:61-01"},"e":"changed","id":"5","r":"lights","t":"event","uniqueid":"14:b4:57:ff:fe:49:e0:61-01"}

{"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"lastannounced":"2020-10-05T21:36:05Z","lastseen":"2020-10-11T18:07Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E14 WS opal 600lm","name":"WHZ_hinten","swversion":"2.0.022","type":"Color temperature light","uniqueid":"84:2e:14:ff:fe:75:eb:69-01"},"e":"changed","id":"6","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:75:eb:69-01"}

{"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"lastannounced":"2020-10-09T16:31:03Z","lastseen":"2020-10-11T18:07Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WS opal 1000lm","name":"KUE_Wand","swversion":"2.0.022","type":"Color temperature light","uniqueid":"bc:33:ac:ff:fe:85:98:14-01"},"e":"changed","id":"7","r":"lights","t":"event","uniqueid":"bc:33:ac:ff:fe:85:98:14-01"}

{"attr":{"lastannounced":null,"lastseen":"2020-10-11T18:07Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","swversion":"0x26580700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:05:ea:ea-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:05:ea:ea-01"}

{"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"lastannounced":"2020-10-05T17:17:50Z","lastseen":"2020-10-11T18:08Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WS opal 1000lm","name":"WHZ_vorne","swversion":"2.0.022","type":"Color temperature light","uniqueid":"58:8e:81:ff:fe:17:3c:84-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"58:8e:81:ff:fe:17:3c:84-01"}

{"attr":{"lastannounced":"2020-10-10T22:10:10Z","lastseen":"2020-10-11T18:09Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW 806lm","name":"KUE_Fenster","swversion":"2.1.022","type":"Dimmable light","uniqueid":"cc:cc:cc:ff:fe:90:f5:6f-01"},"e":"changed","id":"3","r":"lights","t":"event","uniqueid":"cc:cc:cc:ff:fe:90:f5:6f-01"}

{"attr":{"lastannounced":"2020-10-05T17:10:08Z","lastseen":"2020-10-11T18:09Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW clear 250lm","name":"WHZ_oldschool","swversion":"2.1.022","type":"Dimmable light","uniqueid":"cc:cc:cc:ff:fe:89:c9:42-01"},"e":"changed","id":"4","r":"lights","t":"event","uniqueid":"cc:cc:cc:ff:fe:89:c9:42-01"}

{"attr":{"lastannounced":"2020-10-10T22:10:10Z","lastseen":"2020-10-11T18:10Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WW clear 250lm","name":"KUE_mitte_oldschool","swversion":"2.1.022","type":"Dimmable light","uniqueid":"14:b4 [... cut ...] },"e":"changed","id":"6","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:75:eb:69-01"}

{"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"lastannounced":"2020-10-09T16:31:03Z","lastseen":"2020-10-11T20:01Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WS opal 1000lm","name":"KUE_Wand","swversion":"2.0.022","type":"Color temperature light","uniqueid":"bc:33:ac:ff:fe:85:98:14-01"},"e":"changed","id":"7","r":"lights","t":"event","uniqueid":"bc:33:ac:ff:fe:85:98:14-01"}`

MattWestb, 2020-10-12:
I you have some IKEA bulbs its likely the OTA updater have updating the bulb firmware and then the device is rebooting = normal then device have upgraded the firmware. The problem is for the moment is that some (8) new firmware is broken on the servers so its the OTA server can repeating it in innisfil => sporadic random reboot of those devices.
I have not looking thru your logs only reading of the behaviour.

SwoopX, 2020-10-12:
@pep4n I don't see anything unusual in addition in the logs provided. However, regarding the device reboot, I was a day ahead already in my previous comment, so shit that one day back.

pep4n, 2020-10-12:
Thank you for the comment!

In this group I have two bulbs with FW 2.1.022 (TRADFRI bulb E27 WW clear 250lm and TRADFRI bulb E27 WW 806lm) and one with 2.0.022 (TRADFRI bulb E27 WS opal 1000lm)

pep4n, 2020-10-12:
> 
> 
> @pep4n I don't see anything unusual in addition in the logs provided. However, regarding the device reboot, I was a day ahead already in my previous comment, so shit that one day back.

good morning SwoopX, I'm afraid I don't get your last sentence :)

pep4n, 2020-10-12:
It even gets more weird now. 
Since last night I have a single (!) Bulb from another group switching on/off. Just one bulb.
It's a TRADFRI bulb E27 WW clear 250lm. 

I'm really not happy about all that funky behaviour. And don't ask my wife about it...!

Mimiix, 2020-10-12:
We really have nothing to do with your wife 😅 . Anyway, you mentioned iobroker? Their dev mentioned yesterday that iobroker 2 (or something? Beta?) is having issues with it.
What version are you on?

MattWestb, 2020-10-12:
@pep4n Do you have activating / running the IKEA OTA script for updating firmware the last month ?
If yes renaming the "/home/pi/otau" directory to something else (/home/pi/Xotau) and restart deCONZ and the IKEA OTA is disabled and cant sending updates to your devices => No device reboot because of  OTA updates.

I agree with Dennis about your wife but if your system is not "wife friendly" you have REAL problems  ;-)))

pep4n, 2020-10-12:
haha the wife-comment was rather to make clear what @MattWestb mentioned: Slowly running into real problems here ,) 

to answer your questions: 
@Mimiix No beta in place. I'm running Node.js: v12.18.4 and NPM: 6.14.6. 
@MattWestb  no OTA scripts runnning. No updates done. Just freshly bought and installed.

pep4n, 2020-10-12:
@Mimiix so you say this is not a bug? 
Just for me to understand: How do you come to that conclusion?

Mimiix, 2020-10-12:
Because if it was a bug, i had users flooding here :) Seems to be local to your own system, hence i made it a user question.

pep4n, 2020-10-12:
that makes sense. 
Still I really think the problem is with deconz! 
The system is relatively new. 
Only 9 lights are connected (6/9 with deconz). Almost no scripts in place. Nothing else.

Jey-Cee, 2020-10-12:
Simple test, stop the adapter and see whats happening. 
If the behavior is still the same, no problem with the adapter.
Otherwise: 
- What is your adapter version? 
- Set the log level for the instance to debug, then you can see all websocket messages and API calls. 
- Provide the log of the timeframe where the unexpected switching is happen.

Mimiix, 2020-10-12:
> that makes sense.
> Still I really think the problem is with deconz!
> The system is relatively new.
> Only 9 lights are connected (6/9 with deconz). Almost no scripts in place. Nothing else.

Could be, But it is rare then. There's currently 65k users of deCONZ. If this was a broad issue, i had more complaints than one at this point.

pep4n, 2020-10-12:
Just now I had the one bulb switching off. 
here is the log for the 2 minutes

```

deconz.0 | 2020-10-12 16:55:10.453 | debug | (28531) Code 200: Request succeded set group state 6: [{"success":{"/groups/6/action/on":true}}]
-- | -- | -- | --
deconz.0 | 2020-10-12 16:55:10.445 | debug | (28531) dp: on; id:Groups.6 tmp: deconz,0,Groups,6
deconz.0 | 2020-10-12 16:55:10.445 | debug | (28531) stateChange Groups.6  {"val":true,"ack":false,"ts":1602514510443,"q":0,"from":"system.adapter.javascript.0","user":"system.user.admin","lc":1602514480355}
deconz.0 | 2020-10-12 16:54:51.927 | debug | (28531) Event has attr-Tag
deconz.0 | 2020-10-12 16:54:51.926 | debug | (28531) Websocket message:  {"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"lastannounced":"2020-10-05T21:36:05Z","lastseen":"2020-10-12T14:54Z","manufacturername":"IKEA  of Sweden","modelid":"
deconz.0 | 2020-10-12 16:54:40.403 | debug | (28531) Code 200: Request succeded get group attributes 6:  {"action":{"alert":"none","bri":254,"colormode":"ct","ct":500,"effect":"none","hue":0,"on":true,"sat":128,"scene":null,"xy":[0,0]},"deviceme
deconz.0 | 2020-10-12 16:54:40.398 | debug | (28531) Code 200: Request succeded get group attributes 1:  {"action":{"alert":"none","bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":false,"sat":127,"scene":"3","xy":[0,0]},"devicememb
deconz.0 | 2020-10-12 16:54:40.389 | debug | (28531) reachable: true
deconz.0 | 2020-10-12 16:54:40.388 | debug | (28531) on: true
deconz.0 | 2020-10-12 16:54:40.388 | debug | (28531) ct: 454
deconz.0 | 2020-10-12 16:54:40.387 | debug | (28531) colormode: ct
deconz.0 | 2020-10-12 16:54:40.387 | debug | (28531) bri: 254
deconz.0 | 2020-10-12 16:54:40.386 | debug | (28531) alert: null
deconz.0 | 2020-10-12 16:54:40.381 | debug | (28531) reachable: true
deconz.0 | 2020-10-12 16:54:40.381 | debug | (28531) on: true
deconz.0 | 2020-10-12 16:54:40.380 | debug | (28531) bri: 254
deconz.0 | 2020-10-12 16:54:40.379 | debug | (28531) alert: null
deconz.0 | 2020-10-12 16:54:40.377 | debug | (28531) Websocket message: {"e":"changed","id":"6","r":"groups","state":{"all_on":true,"any_on":true},"t":"event"}
deconz.0 | 2020-10-12 16:54:40.376 | debug | (28531) Websocket message: {"e":"changed","id":"1","r":"groups","state":{"all_on":false,"any_on":true},"t":"event"}
deconz.0 | 2020-10-12 16:54:40.369 | debug | (28531) Event has state-tag
deconz.0 | 2020-10-12 16:54:40.368 | debug | (28531) Websocket message:  {"e":"changed","id":"7","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":454,"on":true,"reachable":true},"t":"event","uniqueid":"bc:33:ac:ff:fe:85:98:14-0
deconz.0 | 2020-10-12 16:54:40.366 | debug | (28531) Event has state-tag
deconz.0 | 2020-10-12 16:54:40.366 | debug | (28531) Websocket message:  {"e":"changed","id":"4","r":"lights","state":{"alert":null,"bri":254,"on":true,"reachable":true},"t":"event","uniqueid":"cc:cc:cc:ff:fe:89:c9:42-01"}
deconz.0 | 2020-10-12 16:54:40.364 | debug | (28531) Code 200: Request succeded set group state 6: [{"success":{"/groups/6/action/on":true}}]
deconz.0 | 2020-10-12 16:54:40.357 | debug | (28531) dp: on; id:Groups.6 tmp: deconz,0,Groups,6
deconz.0 | 2020-10-12 16:54:40.357 | debug | (28531) stateChange Groups.6  {"val":true,"ack":false,"ts":1602514480355,"q":0,"from":"system.adapter.javascript.0","user":"system.user.admin","lc":1602514480355}
deconz.0 | 2020-10-12 16:54:21.219 | debug | (28531) Event has attr-Tag
deconz.0 | 2020-10-12 16:54:21.219 | debug | (28531) Websocket message:  {"attr":{"lastannounced":"2020-10-12T13:53:03Z","lastseen":"2020-10-12T14:54Z","manufacturername":"IKEA  of Sweden","modelid":"TRADFRI bulb E27 WW clear 250lm","name":"KUE_m
deconz.0 | 2020-10-12 16:53:53.611 | debug | (28531) Event has attr-Tag
deconz.0 | 2020-10-12 16:53:53.610 | debug | (28531) Websocket message:  {"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"lastannounced":"2020-10-09T16:31:03Z","lastseen":"2020-10-12T14:53Z","manufacturername":"IKEA  of Sweden","modelid":"
deconz.0 | 2020-10-12 16:53:53.551 | debug | (28531) Code 200: Request succeded get group attributes 6:  {"action":{"alert":"none","bri":254,"colormode":"ct","ct":500,"effect":"none","hue":0,"on":false,"sat":128,"scene":null,"xy":[0,0]},"devicem
deconz.0 | 2020-10-12 16:53:53.540 | debug | (28531) Code 200: Request succeded get group attributes 1:  {"action":{"alert":"none","bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":false,"sat":127,"scene":"3","xy":[0,0]},"devicememb
deconz.0 | 2020-10-12 16:53:53.534 | debug | (28531) Websocket message: {"e":"changed","id":"6","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}
deconz.0 | 2020-10-12 16:53:53.532 | debug | (28531) reachable: true
deconz.0 | 2020-10-12 16:53:53.532 | debug | (28531) on: false
deconz.0 | 2020-10-12 16:53:53.531 | debug | (28531) ct: 454
deconz.0 | 2020-10-12 16:53:53.531 | debug | (28531) colormode: ct
deconz.0 | 2020-10-12 16:53:53.530 | debug | (28531) bri: 254
deconz.0 | 2020-10-12 16:53:53.529 | debug | (28531) alert: null
deconz.0 | 2020-10-12 16:53:53.527 | debug | (28531) reachable: true
deconz.0 | 2020-10-12 16:53:53.526 | debug | (28531) on: false
deconz.0 | 2020-10-12 16:53:53.525 | debug | (28531) bri: 254
deconz.0 | 2020-10-12 16:53:53.525 | debug | (28531) alert: null
deconz.0 | 2020-10-12 16:53:53.522 | debug | (28531) Websocket message: {"e":"changed","id":"1","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}
[... cut ...]
deconz.0 | 2020-10-12 16:53:02.753 | debug | (28531) Websocket message:  {"attr":{"lastannounced":"2020-10-12T13:53:03Z","lastseen":"2020-10-12T14:53Z","manufacturername":"IKEA  of Sweden","modelid":"TRADFRI bulb E27 WW clear 250lm","name":"KUE_m
deconz.0 | 2020-10-12 16:53:01.407 | debug | (28531) Code 200: Request succeded set group state 6: [{"success":{"/groups/6/action/on":true}}]
deconz.0 | 2020-10-12 16:53:01.401 | debug | (28531) dp: on; id:Groups.6 tmp: deconz,0,Groups,6
deconz.0 | 2020-10-12 16:53:01.400 | debug | (28531) stateChange Groups.6  {"val":true,"ack":false,"ts":1602514381399,"q":0,"from":"system.adapter.javascript.0","user":"system.user.admin","lc":1602513569177}

```

Smanar, 2020-10-12:
>deconz.0 | 2020-10-12 16:53:53.510 | debug | (28531) Code 200: Request succeded set group state 6: [{"success":{"/groups/6/action/on":false}}]
>deconz.0 | 2020-10-12 16:53:01.407 | debug | (28531) Code 200: Request succeded set group state 6: [{"success":{"/groups/6/action/on":true}}]

You have send yourself an "on" then an "off" order ?

pep4n, 2020-10-12:
no I didn't  :( 
could we investigate where this originates?

pep4n, 2020-10-12:
here is an example of the first group (is only 1 bulb atm.) 

```

deconz.0 | 2020-10-12 18:35:31.005 | debug | (28531) Event has attr-Tag
-- | -- | -- | --
deconz.0 | 2020-10-12 18:35:31.005 | debug | (28531)  Websocket message:  {"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"lastannounced":"2020-10-05T17:17:50Z","lastseen":"2020-10-12T16:35Z","manufacturername":"IKEA  of Sweden","modelid":"T
deconz.0 | 2020-10-12 18:35:30.646 | debug | (28531) Code 200: Request succeded set group state 6: [{"success":{"/groups/6/action/on":true}}]
deconz.0 | 2020-10-12 18:35:30.639 | debug | (28531) dp: on; id:Groups.6 tmp: deconz,0,Groups,6
deconz.0 | 2020-10-12 18:35:30.639 | debug | (28531)  stateChange Groups.6  {"val":true,"ack":false,"ts":1602520530638,"q":0,"from":"system.adapter.javascript.0","user":"system.user.admin","lc":1602520410371}
deconz.0 | 2020-10-12 18:35:29.013 | debug | (28531) Subscribed to updates...
deconz.0 | 2020-10-12 18:35:29.008 | debug | (28531) Websocket connection closed
deconz.0 | 2020-10-12 18:35:00.528 | debug | (28531) Code 200: Request succeded set group state 6: [{"success":{"/groups/6/action/on":true}}]
deconz.0 | 2020-10-12 18:35:00.521 | debug | (28531) dp: on; id:Groups.6 tmp: deconz,0,Groups,6
deconz.0 | 2020-10-12 18:35:00.520 | debug | (28531)  stateChange Groups.6  {"val":true,"ack":false,"ts":1602520500519,"q":0,"from":"system.adapter.javascript.0","user":"system.user.admin","lc":1602520410371}
deconz.0 | 2020-10-12 18:34:30.473 | debug | (28531) Code 200: Request succeded set group state 6: [{"success":{"/groups/6/action/on":true}}]
deconz.0 | 2020-10-12 18:34:30.466 | debug | (28531) dp: on; id:Groups.6 tmp: deconz,0,Groups,6
deconz.0 | 2020-10-12 18:34:30.466 | debug | (28531)  stateChange Groups.6  {"val":true,"ack":false,"ts":1602520470465,"q":0,"from":"system.adapter.javascript.0","user":"system.user.admin","lc":1602520410371}
deconz.0 | 2020-10-12 18:34:29.007 | debug | (28531) Event has attr-Tag
deconz.0 | 2020-10-12 18:34:29.007 | debug | (28531)  Websocket message:  {"attr":{"lastannounced":null,"lastseen":"2020-10-12T16:34Z","manufacturername":"dresden  elektronik","modelid":"ConBee II","name":"Configuration tool  1","swversion":"0x26580
deconz.0 | 2020-10-12 18:34:00.458 | debug | (28531) Code 200: Request succeded set group state 6: [{"success":{"/groups/6/action/on":true}}]
deconz.0 | 2020-10-12 18:34:00.452 | debug | (28531) dp: on; id:Groups.6 tmp: deconz,0,Groups,6
deconz.0 | 2020-10-12 18:34:00.451 | debug | (28531)  stateChange Groups.6  {"val":true,"ack":false,"ts":1602520440450,"q":0,"from":"system.adapter.javascript.0","user":"system.user.admin","lc":1602520410371}
deconz.0 | 2020-10-12 18:33:57.400 | debug | (28531) Event has attr-Tag
deconz.0 | 2020-10-12 18:33:57.400 | debug | (28531)  Websocket message:  {"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"lastannounced":"2020-10-09T16:31:03Z","lastseen":"2020-10-12T16:33Z","manufacturername":"IKEA  of Sweden","modelid":"T
deconz.0 | 2020-10-12 18:33:30.477 | debug | (28531) Event has attr-Tag
deconz.0 | 2020-10-12 18:33:30.476 | debug | (28531)  Websocket message:  {"attr":{"lastannounced":"2020-10-05T17:10:08Z","lastseen":"2020-10-12T16:33Z","manufacturername":"IKEA  of Sweden","modelid":"TRADFRI bulb E27 WW clear 250lm","name":"WHZ_ol
deconz.0 | 2020-10-12 18:33:30.414 | debug | (28531)  Code 200: Request succeded get group attributes 6:  {"action":{"alert":"none","bri":254,"colormode":"ct","ct":500,"effect":"none","hue":0,"on":true,"sat":128,"scene":null,"xy":[0,0]},"devicemem
deconz.0 | 2020-10-12 18:33:30.407 | debug | (28531)  Code 200: Request succeded get group attributes 1:  {"action":{"alert":"none","bri":127,"colormode":"hs","ct":0,"effect":"none","hue":0,"on":false,"sat":127,"scene":"3","xy":[0,0]},"devicemembe
deconz.0 | 2020-10-12 18:33:30.401 | debug | (28531) Websocket message: {"e":"changed","id":"6","r":"groups","state":{"all_on":true,"any_on":true},"t":"event"}
deconz.0 | 2020-10-12 18:33:30.400 | debug | (28531) Websocket message: {"e":"changed","id":"1","r":"groups","state":{"all_on":false,"any_on":true},"t":"event"}
deconz.0 | 2020-10-12 18:33:30.400 | debug | (28531) reachable: true
deconz.0 | 2020-10-12 18:33:30.399 | debug | (28531) on: true
deconz.0 | 2020-10-12 18:33:30.399 | debug | (28531) ct: 454
deconz.0 | 2020-10-12 18:33:30.398 | debug | (28531) colormode: ct
deconz.0 | 2020-10-12 18:33:30.397 | debug | (28531) bri: 254
deconz.0 | 2020-10-12 18:33:30.397 | debug | (28531) alert: null
deconz.0 | 2020-10-12 18:33:30.395 | debug | (28531) reachable: true
deconz.0 | 2020-10-12 18:33:30.395 | debug | (28531) on: true
deconz.0 | 2020-10-12 18:33:30.394 | debug | (28531) bri: 254
deconz.0 | 2020-10-12 18:33:30.393 | debug | (28531) alert: null
deconz.0 | 2020-10-12 18:33:30.383 | debug | (28531) Event has state-tag
deconz.0 | 2020-10-12 18:33:30.382 | debug | (28531)  Websocket message:  {"e":"changed","id":"7","r":"lights","state":{"alert":null,"bri":254,"colormode":"ct","ct":454,"on":true,"reachable":true},"t":"event","uniqueid":"bc:33:ac:ff:fe:85:98:14-01
deconz.0 | 2020-10-12 18:33:30.381 | debug | (28531) Event has state-tag
deconz.0 | 2020-10-12 18:33:30.381 | debug | (28531)  Websocket message:  {"e":"changed","id":"4","r":"lights","state":{"alert":null,"bri":254,"on":true,"reachable":true},"t":"event","uniqueid":"cc:cc:cc:ff:fe:89:c9:42-01"}
[... cut ...]
deconz.0 | 2020-10-12 18:30:22.582 | debug | (28531)  stateChange Groups.6  {"val":true,"ack":false,"ts":1602520222581,"q":0,"from":"system.adapter.javascript.0","user":"system.user.admin","lc":1602519436567}
deconz.0 | 2020-10-12 18:30:21.220 | debug | (28531) Event has attr-Tag
deconz.0 | 2020-10-12 18:30:21.220 | debug | (28531)  Websocket message:  {"attr":{"lastannounced":null,"lastseen":"2020-10-12T16:30Z","manufacturername":"dresden  elektronik","modelid":"ConBee II","name":"Configuration tool  1","swversion":"0x26580

```

Smanar, 2020-10-12:
>deconz.0 | 2020-10-12 16:53:53.510 | debug | (28531) Code 200: Request succeded set group state 6: [{"success":{"/groups/6/action/on":false}}]

Thoses line are caused by a request to deconz, something have send an order, and deconz have answered "[{"success":{"/groups/6/action/on":false}" 
So I think you have something on your system that trigger this order.

pep4n, 2020-10-12:
hmm ok thats so strange! 
also why only one bulb of the group of 3?

Smanar, 2020-10-12:
Still with `[{"success":{"/groups/6/action/on":false}}]` ?

closed 2020-11-24 by stale[bot]