dresden-elektronik/deconz-rest-plugin #3392: Error incorrect sRGB profile and Update

clipse2004, 2020-10-10:
Hey Guys,
I updated to the newest Firmware and DeConz Software for my Raspberry and Conbee 2 Stick.
When I reboot my device I always get these error:
Okt 11 00:34:45 phoscon deCONZ30861: libpng warning: iCCP: known incorrect sRGB profile

What can I do? 

And the other Problem, my Phoscon app don’t show me newer Version/Updates.
I was on 2665700 firmware before and it didn’t tells me there is a newer one :/

Smanar, 2020-10-11:
Hello,

So first please use the template for issues.
This error is normal, don't take care, I think you have too "This plugin does not support propagateSizeHints()"

The lastest firmware are not in all deconz version, if you have a real machine you can just wait for a new version to use this feature with the webapp, or make the update manualy https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Mimiix, 2020-10-12:
As @Smanar Mentiones. 

However, the Firmware update is whatever is packaged with the deconz version you have.
 
Closing this issue as it has been answered and most importantly: Lacking issue template.

closed 2020-10-12 by Mimiix