dresden-elektronik/deconz-rest-plugin #3393: Bosch  ISW-ZPR1-WP13 pir sensor missing features like tamper ans battery percentage

mamrai1, 2020-10-11:
## Device

- Product name: Bosch  ISW-ZPR1-WP13
- Manufacturer: Bosch
- Model identifier:  ISW-ZPR1-WP13
- Device type : Please remove all unrelated device types. 
  - Sensor

  - Other: 

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mamrai1, 2020-10-11:
1) motion sensor is triggered for 1 second and then sleeps for 3 minutes..
2) temperature is reported only when there is a change over 0.5 degrees celcius...( as i can see from home assistant history)
3) pairing was done near the conbee ii before moving the sensor to position..(initial pairing on position was not ok_through ikea light)

mamrai1, 2020-10-11:
api response after opening sensor case shows that tamper works (http://192.168.1.100:40850/api/7A79971660/sensors/17):

{"config":{"battery":null,"duration":60,"on":true,"pending":[],"reachable":true},"ep":5,"etag":"948eaa7befa5835d7fd7a38f6c3c8b1a","lastseen":"2020-10-11T14:43Z","manufacturername":"Bosch","modelid":"ISW-ZPR1-WP13","name":"Kitchen Motion sensor","state":{"lastupdated":"2020-10-11T14:43:57.887","lowbattery":false,"presence":false,"tampered":true},"swversion":"20140929","type":"ZHAPresence","uniqueid":"00:15:5f:00:b4:4d:1c:f2-05-0500"}

but also 2nd id of same sensor (http://192.168.1.100:40850/api/7A79971660/sensors/18):

{"config":{"battery":null,"offset":0,"on":true,"reachable":true},"ep":5,"etag":"cf48ac696715c56e5e79368c96c4c2c2","lastseen":"2020-10-11T14:43Z","manufacturername":"Bosch","modelid":"ISW-ZPR1-WP13","name":"Kitchen Motion sensor","state":{"lastupdated":"2020-10-11T14:43:46.295","temperature":2614},"swversion":"20140929","type":"ZHATemperature","uniqueid":"00:15:5f:00:b4:4d:1c:f2-05-0402"}

Is this normal?

SwoopX, 2020-10-11:
Erm, I don't understand the question here. In case it's if it is normal to have 2 sensors for the device, then yes.

Thanks for the provided info. That should be sufficient to allow us making the necessary changes for the battery readings.

Regarding the threshold for the temperature change reports, that can be changed at will via deconz GUI. Check out https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-configure-attribute-reporting and choose attribute `Measured value` from Temperature Measurement cluster.

mamrai1, 2020-10-11:
Thanks. How can I manually expose battery state and tamper on home assistant? Is there a guide?

SwoopX, 2020-10-12:
No clue, that's what you should ask in the HA universe.

closed 2020-10-15 by manup

mamrai1, 2020-10-23:
One more question: Why 'tampered' attribute is not shown in phoscon?