dresden-elektronik/deconz-rest-plugin #3394: how invoke OffWaitTime by using REST API?

muchasuerte, 2020-10-11:
## Describe the question or issue you are having
By using Deconz UI I'm able to use 2 function delayed start and delayed off, for the first case I know how to invoke the first with REST API `curl -H 'Content-Type: application/json' -X PUT -d '{"on": true, "ontime": 600}' http://192.168.1....`
but I did not found any info how to invoke the off with time by REST API.  

OnTime `{"on": true, "ontime": 600}`
OffWaitTime `?`

## Screenshots
[image]

## Environment
Zigbee device that I would like use to manage: TS0121

 - Host system: Raspberry Pi
 - Running method: Raspbian 
 - Firmware version: (26580700)
 - deCONZ version: (2.05.78)
 - Device: ConBee II
 - Do you use an USB extension cable: yes

ebaauw, 2020-10-11:
It’s not exposed by the API.

muchasuerte, 2020-10-11:
Thank you.
is there no way to use it from HA?
is there some other way to manage it other then REST?

ebaauw, 2020-10-11:
There's nothing to manage; it's a parameter to a Zigbee command.  You can include it when sending the command from the GUI or from the CLI plugin.

muchasuerte, 2020-10-11:
ok, yes I have found it `const quint16 offWaitTime = 0;` on DeRestPluginPrivate::addTaskSetOnOff maybe I can try to add it if I'm not in wrong the REST API is the interface used by several integration extension home assistant included , I would like use it to manage long task where I want to be sure that the service go off after x time

ebaauw, 2020-10-11:
> I would like use it to manage long task where I want to be sure that the service go off after x time

That's what `ontime` does.  See the ZCL spec.

muchasuerte, 2020-10-12:
So maybe is my TS0121 which is bugged, because when I use that command it switch ON after X seconds while by using the OffWaitTime from on it switch to off after X seconds.

muchasuerte, 2020-10-26:
@ebaauw might be that the problem on deconz firmware who is not managing properly my MS104Z Zigbee? I can see this device rebranded with several names another option if the issue is not on plugin could be the firmware bugged?

Smanar, 2020-10-26:
I think there is more chnace for the problem was from the device ^^.

muchasuerte, 2020-10-29:
I sent an email to info@samotech.co.uk on their site they report deconz compatibility but ontime do not works 

curl -H 'Content-Type: application/json' -X PUT -d '{"on": true, "ontime": 10}' http://<deconz-server-ip>/api/<api-key>/lights/1/state

by sending OffWaitTime command from deconz UI this device manage the command like ontime command

closed 2020-12-19 by stale[bot]