dresden-elektronik/deconz-rest-plugin #3395: haozee smoke sensor - it's possible to include?

Jfcanadell, 2020-10-11:
- Product name:   wireless smoke detector.
- Manufacturer:    _TZYB01_dsjszp0x.
- Model identifier:  TS0205.
- Device type :  Sensor

[image]

[image]

[image]

[image]

[image]

[image]

accelle17, 2020-11-02:
Here's another screenshot for the device clusters and node info:

  - Endpoints and clusters of the node
[image]

  - Node Info panel
[image]

  - Basic Cluster attributes in the Cluster Info panel.
[image]
[image]

0001 POWER CONFIGURATION CLUSTER:
[image]
[image]

0003 IDENTIFY CLUSTER:
[image]

0500 IAS ZONE:
[image]
[image]

0003 IDENTIFY CLUSTER (2):
[image]

0019 OTAU:
[image]

accelle17, 2020-11-02:
There's already support for TS0204 which is a gas sensor, so it might just need to be added with the same code. Also, this device can act as a controllable siren I believe.

closed 2020-11-14 by manup

saxn-paule, 2020-12-08:
> There's already support for TS0204 which is a gas sensor, so it might just need to be added with the same code. Also, this device can act as a controllable siren I believe.

Exactly this was my intent when I bought it. How could I use it as a siren?

SwoopX, 2020-12-09:
Unless it hasn't got a IAS Warning cluster, you can't.