dresden-elektronik/deconz-rest-plugin #3397: Missing button map for ZGRC-KEY-009

rvanmoorsel, 2020-10-11:
## Describe the bug

I've got two wall switches that appear to be quite common in Europe. They fit in a normal wall frame and can replace traditional switches. They're marketed under several names, but deCONZ recognizes them as Sunricher ZGRC-KEY-XXX. In a old version of deCONZ I could use the 'deconz_event' service listener in HA, but now they aren't sending the switch presses anymore. 
They are recognized correctly in deCONZ (and obviously HA), but the button map seems off. 
I've posted this issue in another thread (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1286) and the user SwoopX advised me to file a bug report.

## Steps to reproduce the behavior

  If the problem is reproducable, list the steps here:

  1. I press a button and it isn't registered.
  2. The logfile produces an [INFO] - No button map for... entry.

## Expected behavior

  I'd like to see a button pressed event in HA / deCONZ

## Screenshots

  Not applicable.

## Environment

  deCONZ version: 2.05.84 / 14-9-2020
  Firmware version: 26660700

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: (2.05.84)
 - Device: ConBee II
 - Do you use an USB extension cable: yes -- only relevant for ConBee I/II

### deCONZ Logs

  There are two rows with each four buttons.
  When I press the 'ON' (Top left) button, this debug log is written:
    [INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: ONOFF (0x0006) command: ON (0x01) payload[0]: 000
  When I press the 'OFF' (Top right) button, this debug log is written:
    [INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: ONOFF (0x0006) command: OFF (0x00) payload[0]: 000
  When I press the second button from the top on the left (Brighter), this debug log is written:
    [INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP_WITH_ON_OFF (0x06) payload[0]: 000
  When I press the second button from the top on the right (Dim), this debug log is written:
    [INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP_WITH_ON_OFF (0x06) payload[0]: 001
  When I press the third button from the top on the left (White), this debug log is written:
    [INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0B9
  When I press the third button from the top on the right (Color control), this debug log is written:
    [INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR (0x07) payload[0]: 0CE
  When I press the bootom button from on the left (S1), this debug log is written:
    [INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: SCENES (0x0005) command: RECALL_SCENE (0x05) payload[0]: 000
  When I press the bootom button from on the right (S2), this debug log is written:
    [INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: SCENES (0x0005) command: RECALL_SCENE (0x05) payload[0]: 000

## Additional context

  It's an eight-button switch. I've also got a two-button version, and I've posted a separate bug report (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3396) for this one.
  I've used ZHA for a few weeks, but it was so unstable, opposed to the 1 1/2 year + stability of deCONZ, I quickly built everything    back. ZHA recognized the button actions correctly, so I hope it can be fixed in deCONZ too.
  If any more info is needed, please let me know.

Smanar, 2020-10-11:
Hello, wich one deconz version have you ?

Do you have the json file "button_maps.json" ?

rvanmoorsel, 2020-10-11:
Hi Smanar,
I'm running deCONZ version 2.05.84 as an addon in HomeAssistant (HassIO 5.2 x64). I have installed SSH, but when I log in, there is no /usr/share/deCONZ directory. I assume that this is because deCONZ runs a separate docker container.
Is there any way to see if the file is there? A quick google doesn't help me finding it.

Smanar, 2020-10-11:
I don't know at all how work docker.
But yes on version 84, the file is already here, and editable with a simple tex editor, so easy to add your devie model ID id in the file

SwoopX, 2020-10-11:
You really gotta need to find the file `button_maps.json`. On a bare metal install, the code searches for that file in the following directories in the given order:

- `$HOME/.local/share/dresden-elektronik/deCONZ/devices`
- `/usr/share/deCONZ/devices/`

As the first option is also the directory where your database resides and therefore, should be persistent, you "just" gotta find out where that is in your case.

The debug output above looks very promising. Once you found the file, we can guide you for the amendment.

rvanmoorsel, 2020-10-12:
It took me a little tinkering, but I've got the button_maps.json file and opened it in my notepad++ editor. I can read it a bit, and I'm guessing I need to add the switches and use the debug code. But a little guidance would be nice :)
Thanks for trying to help me out here!

Smanar, 2020-10-12:
There is a line 
>            "modelids": ["ICZB-RM", "ZGRC-KEY-012", "ZGRC-KEY-013", "RGBgenie ZB-5001"],

add your model id inside.

rvanmoorsel, 2020-10-12:
I downloaded the file and edited it. But a big disadvantage for me is that I'm running deCONZ as an addon in HomeAssistant. Their gituhub is: https://github.com/home-assistant/hassio-addons/tree/master/deconz
But after I figured out how to enter the docker container, I found the button_maps.json file. When I compare this file, with the github version, there are a few differences. The most noticeable difference is this:

Original version:
``
"icasaRemoteMap": {
            "modelids": ["ICZB-RM", "ZGRC-KEY-007", "ZGRC-KEY-009", "ZGRC-KEY-012", "ZGRC-KEY-013", "RGBgenie ZB-5001"],
            "map": [
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_1", 
``

Compared to the app version:
`"icasaRemoteMap": [
		[1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
		[1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
		[1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "1", "S_BUTTON_1",
`

The rest of the file is the same. I used a compare tool, and the only difference are the missing lines "modelids": and "map":. But to test it all, I placed 'your' github version of the button_maps.json file inside my docker, but that doesn't seem to work When I restart my docker, all files are back to their original version. Right now I'm trying to figure out if there's a way to just restart the deCONZ service, but didn't find a way for that yet. But I did want to post a follow-up.

I was under the impression that you guys work closely with the deCONZ-addon for HA - guys, but apparently, that's a wrong assumption. 
Is the 'modelids'-bit really the only link there is to my hardware? Because if that's the case, I'm guessing a whole lot of other stuff should go wrong with the button_maps.json file currently present in the app-version.

I'm curious if you guys can still assist me now, or if I should take this matter to the people at https://github.com/home-assistant/hassio-addons/tree/master/deconz and make a case there?

Thanks a lot for all the help so far!

Smanar, 2020-10-12:
Ok so bad luck, they are 1 version late.
I don't think it  s possible to change the model id on this one.

SwoopX, 2020-10-13:
Ok, so from what I see, we would need a new button map for this one. Based on the information so far, it should look like this:

```
"sunricherMap2": {
            "modelids": ["ZGRC-KEY-009"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Step up (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "Move up (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF", "1", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Step down (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_4", "S_BUTTON_ACTION_HOLD", "Move down (with on/off)"],
                [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "Stop_ (with on/off)"],
                [1, "0x01", "COLOR_CONTROL", "MOVE_COLOR_TEMPERATURE", "185", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "Move color temperature"],
                [1, "0x01", "COLOR_CONTROL", "MOVE_TO_COLOR", "206", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "Move to color"],
                [1, "0x01", "SCENES", "RECALL_SCENE", "0", "S_BUTTON_7", "S_BUTTON_ACTION_SHORT_RELEASED", "Recall scene 1"],
                [1, "0x01", "SCENES", "STORE_SCENE", "0", "S_BUTTON_7", "S_BUTTON_ACTION_LONG_RELEASED", "Store scene 1"],
                [1, "0x01", "SCENES", "RECALL_SCENE", "0", "S_BUTTON_8", "S_BUTTON_ACTION_SHORT_RELEASED", "Recall scene 2"],
                [1, "0x01", "SCENES", "STORE_SCENE", "0", "S_BUTTON_8", "S_BUTTON_ACTION_LONG_RELEASED", "Store scene 2"]
            ]
        },
```

However, I added some assumptions in there. Can it be that you only used short presses? You should also check long and double presses. I added some anticipated commands for your 2nd button row, but that needs to be verified and eventually missing stuff added for the other buttons.

rvanmoorsel, 2020-10-13:
You are right. I only did short presses on the buttons. On the ZGRC-KEY-007 switch, I did tried a double press, but that resulted in two registered single presses. So my assumption is that there isn't a double-press option.
Tonight or tomorrow (depending on my time), I'll try long press for both buttons of the ZGRC-KEY-007, and long press for both at the same time.
With the ZGRC-KEY-009 I'll try double press on all buttons. If that's the same result as with the ZGRC-KEY-007, I guess there isn't a double-press function on these switches. I'll also do a long-press on all buttons. I'll post the results asap. 
Do you want me to post everything here, or the ZGRC-KEY-007 results in the bug report I created for that switch?

And knowing missing out on the latest version being the reason for not working correctly, Isn't that bad. Just testing my patience. Seeing how you guys pick up this request for a new switch, makes me happy and confident about the choice I made for Zigbee with deCONZ.

rvanmoorsel, 2020-10-14:
Hi,
I took my ZGRC-KEY-009 and pressed about all the buttons (long, short, double) and this is what I see in the logs:
- Long press ON 
No registered log
- Long press OFF
No registered log
- Long press Brighter (Second button from the top on the left)
`[INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_WITH_ON_OFF (0x05) payload[0]: 000`
Release long press Brighter (Second button from the top on the left)
`[INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STOP_WITH_ON_OFF (0x07) payload[0]: 000`
- Long press Dim (Second button from the top on the right)
`[INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_WITH_ON_OFF (0x05) payload[0]: 001`
- Release long press Dim (Second button from the top on the right)
`[INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STOP_WITH_ON_OFF (0x07) payload[0]: 000`
- Long press W (Third button from the top on the left)
`[INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_COLOR_TEMPERATURE (0x4B) payload[0]: 001`
- Release long press W (third button from the top on the left)
`[INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: STOP_MOVE_STEP (0x47) payload[0]: 000`
- Long press Color (Third button from the top on the right)
`[INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_HUE (0x01) payload[0]: 003`
- Release long press Color (Third button from the top on the right)
`[INFO] - No button map for: ZGRC-KEY-009 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_HUE (0x01) payload[0]: 000`
- Long press S1 (fourth button on the left)
No registered log
- Long press S2 (fourth button on the right)
No registered log

None of the buttons report logging when I double-press them. I've posted ll single press logs in earlier posts, but if it's more convenient for you guys, I can make one big post, containing all logs.

To be complete, here's the log from the ZGRC-KEY-007. I'll also update the specific post on that switch.
- Long press ON
`[INFO] - No button map for: ZGRC-KEY-007 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_WITH_ON_OFF (0x05) payload[0]: 001`
- Long press ON released:
`[INFO] - No button map for: ZGRC-KEY-007 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STOP_WITH_ON_OFF (0x07) payload[0]: 000`
- Long press OFF
`[INFO] - No button map for: ZGRC-KEY-007 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_WITH_ON_OFF (0x05) payload[0]: 000`
- Long press OFF released:
`[INFO] - No button map for: ZGRC-KEY-007 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STOP_WITH_ON_OFF (0x07) payload[0]: 000`

If there's any more info you guys need, or if I need to do some testing, please let me know!

SwoopX, 2020-10-15:
On the first glimpse, it contains the missing parts. Thanks! Will have a deeper look later and let you know.

closed 2020-10-15 by manup

rvanmoorsel, 2020-11-14:
Hi,
Just now, the deCONZ plugin was updated by the guys at HomeAssistant. I was looking forward to this update, because of the ZGRC-KEY update. After the update, I checked my wall switches, and they work! So thank you for that!
There is just one thing, I didn't notice earlier. There was a mix-up between the two ZGRC-KEY items. 
This means that the text that's under ZGRC-KEY-007 needs to be under ZGRC-KEY-009 and the other way around.
How can I best address this issue, so it will be fixed at the source? Of course I can change the file manually (although this is hard in a docker container), but if it can be ficed at the source, it'll be way better.

Thanks in advance!

SwoopX, 2020-11-14:
You're welcome. For the time being, you need to swap in your local file (button_maps.json).

rvanmoorsel, 2020-12-14:
I'm so sorry to bug you great guys again! But with the latest deCONZ update in my HomeAssistant  the ZRG-007 / ZRG-009 swap was implemented. There unfortunattely is still one problem. Three buttons don't work yet. The error in the debug log is different. The first time it was 
`No button map for XX`
Now it's:
`No button handler for:` 

The three buttons with errors are 5, 7 and 8.

Button5: (5002)
In the logfile:
`[INFO] - No button handler for: ZGRC-KEY-009 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0E6`

Button7: (7002)
In the logfile:
`[INFO] - No button handler for: ZGRC-KEY-009 endpoint: 0x01 cluster: SCENES (0x0005) command: RECALL_SCENE (0x05) payload[0]: 000`

Button8: (8002)
In the logfile:
`[INFO] - No button handler for: ZGRC-KEY-009 endpoint: 0x01 cluster: SCENES (0x0005) command: RECALL_SCENE (0x05) payload[0]: 000`

I notice the last logs are the same.

Is there anything I can test / check? Maybe you need more information?

Thanks again for all the help!

SwoopX, 2020-12-14:
There was indeed a typo easily overlooked for button event 5002, already fixed.

For the scenes buttons, I don't really get what the payload of those is. You could give it another go and add `--dbg-aps=2`, so the full aps data becomes visible. Eventually, that gives us a better understanding. I still don't get why there's no store scene command...

rvanmoorsel, 2021-01-14:
My apologies it took so long. But I just restarted deCONZ with the `--dbg-apps=2` option. I don't see a change in the outcome, but maybe you will?
S1 button:
`21:22:07:088 	asdu: 015a05000001
21:22:07:090 [INFO] - No button handler for: ZGRC-KEY-009 endpoint: 0x01 cluster: SCENES (0x0005) command: RECALL_SCENE (0x05) payload[0]: 000
21:22:07:646 APS-DATA.indication srcAddr: 0xbec0, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 215, rssi: -67
`
The S2 button:
`21:26:25:966 	asdu: 015e05000002
21:26:25:969 [INFO] - No button handler for: ZGRC-KEY-009 endpoint: 0x01 cluster: SCENES (0x0005) command: RECALL_SCENE (0x05) payload[0]: 000
21:26:26:360 APS-DATA.indication srcAddr: 0xbec0, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 239, rssi: -64
`
If there is anything more you need, just let me know. I'll also start digging around with the ZHA HomeAssistant plugin. During the week and a half I ran that software, both Scenes buttons worked. So maybe they've got the payload right.

SwoopX, 2021-02-07:
Hm, I gotte need to see the full ZCL payload for those events to fully understand what's going on and what might be required in addition. A PR of mine was just recently merged to exactly provide that in future. So I guess we need to postpone this a bit and wait for HA to adopt to the upcoming version. Then, we can give it another shot with hopefully a clearer picture.