dresden-elektronik/deconz-rest-plugin #3400: Expose min_temp/max_temp

PhilippMundhenk, 2020-10-12:
## Feature request type

When using Home Assistant (HA) with a Eurotronic Spirit Zigbee thermostat, I noticed that the default values for available temperature range are reported (7..35°). However, the Eurotronic only supports 7..30 degrees, which is also correctly shown via VNC unter 0201 Thermostat > 0x0015/0x0016 Min/Max Heat Setpoint Limit. I could not find these limits exposed on the REST API. I think it would make sense to add them, so that thermostats in HA can show the correct limits.

This would also need to be connected to the min_temp/max_temp attributes in HA by the deconz integration.

## Description

Expose 0201 > 0x0015/0x0016 as min_temp/max_temp via REST API and integrate into deconz HA component.

## Considered alternatives

## Additional context

PhilippMundhenk, 2020-10-12:
Current workaround: Manually customize in configuration.yaml:
```yaml
customize:
  climate.thermostat_32 :
    max_temp: 30
    min_temp: 7
```

Mimiix, 2020-10-12:
@SwoopX  @kane610 Who needs to fix this?

Kane610, 2020-10-12:
@Mimiix deconz rest api needs to expose these values for api consumers to use