dresden-elektronik/deconz-rest-plugin #3406: Tradfri Remotes suddenly stop working

ChrisPrefect, 2020-10-13:
## Describe the bug
Some Tradfri 5-Button remotes suddenly stop working. I had several Tradfri-remotes that just had no function anymore. Changing the battery did not help. The remotes have been added to a light group and worked for several weeks or month before just stopping to control the light group.

I tried to just reset the remotes by pressing the button 5 times, but that did not change anything.

I then did the same but with Phoscon searching for a new remote and then the remote got paired right away.

The old remote was still shown with black text (not grayed out) in the list of remotes, even after the battery has been out of the remote for several hours.

Interestingly, the old entry in the list of switches then disappeared and a new remote with the default name was shown. After changing the name back to the original name of the remote, it worked again and was already paired to the light groups with full functionality restored.

This happened now about six times with four different remotes.

## Steps to reproduce the behavior
Have about 20 Tradfri-Remotes and 80 Tradfri Lights and use the network for a few months. Some remotes might just stop working.

## Expected behavior
The remotes should not loose connection or the binding.

## Screenshots ## Environment

[image]

[image]

[image]

 - Host system: VM with VirtualBox under Windows
 - Running method: Raspbian beta image, upgraded to the newest version
 - Firmware version: 26580700
 - deCONZ version: 2.05.83
 - Device: ConBee II
 - Do you use an USB extension cable: yes

ChrisPrefect, 2020-10-13:
Oh wait, the VM is for ioBroker. Deconz and Phoscon run actually on a Raspberri Pi 4. I updated the post.

> Be adviced: Version 84 is out on Windows. that might fix it too ;)

I updated so many times but nothing ever got fixed for me :(   And nothing in the release notes here refers to the problems I experience: https://github.com/dresden-elektronik/deconz-rest-plugin/releases

svippe, 2020-10-13:
@ChrisPrefect I've experienced the same issue but with the bulbs as well...

They just become unavailable all of a sudden and it's driving me insane. I thought this was a faulty device so I returned mine but I've come down to two problems:

1. Later deconz versions (from 2.05.80 and onwards) are buggy and should not have been released as stable in my view.
2. The range limits on the deconz device/routing capabilities of my network seems off. I can send commands to and from my ikea on/off remote but my ikea tradfri GU10 will not receive the commands from the mesh network, even though I'm only 4 meters away from another bulb (E27 Ikea in another room 4 m away from the GU10).

If these issues continues to the same extent I will not continue with Zigbee and move in to wifi bulbs. I'm glad I can investigate during day times otherwise the next 3 weekends would have been solely troubleshooting this...

Mimiix, 2020-10-14:
>Why is this now a user question and not a bug report? Remotes should not just stop working. That's not a question.

@ChrisPrefect Because if it was a bug, i had a lot more users complaining about this. There is a issue on Ikea stuff https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261 and that is in progress. So it might be that you are experiencing that.

@svippe : Can you open a own issue? Because there might be other stuff going on.

ChrisPrefect, 2020-10-14:
Wait, what?? Mimiix, you are saying that this is not a bug because I am the first to report it? SERIOUSLY??

This is not the first time you do this kind of stuff. You also try to fend of user problems in the discord channel, instead of helping. Why are you so hostile? Why are you defending deconz instead of recognizing all the many, many problems it has and try to get them fixed by the devs as soon as possible? 

Marking comprehensive bug reports as "user question" really takes the cake...

And your only excuse is that no one else has reported it so far? Wow...

How many users have 100+ Tradfri nodes in a big area network with 5 Tradfri repeaters and dozens of remotes and use this over a period of several month? I bet very few users. And how many will report it here if a handful of remotes suddenly fail and have to be reset and bound again? No one will ever report this, except me. You should be thankful that I took the time to investigate and write a complete bug report with screenshots.

#1261 has nothing to do with this bug. And did you check the date of this bug? 14 Feb 2019 (!!!!!) This issue is soon TWO YEARS OLD! And it is still nowhere near fixed! Svippe confirmed this with his report.

This is really stressful. So many hours and days are wasted in troubleshooting Deconz/Conbee/Phoscon. This system just does not work properly in a big Tradfri-Network. And no one is doing anything to fix this. Except trying to remove bug reports and hide them as "user comment". This is not helpful. 

I will call dresden elektronik for the fourth time tomorrow and discuss this directly with them.

Mimiix, 2020-10-15:
@ChrisPrefect Hey, that is not my intention here! I really am trying to help users where i can. I am not hostile, i simply try to keep things clean. Please accept my apologies, as this is not what i am intending at all. 

In terms of the problems: I am forwarding the issues to @manup whenever i feel there is a need to. In my time here, i went trough all issues and hardly any issue is the same. I do agree there is a lot to be done and that's why i initially i picked up this position: I was frustrated. Just frustrated that deCONZ was a bit of a mess. I messaged Robban and asked if i could make the Discord more active.  Reached out to Manup and fast forward , here i am. I cleaned the old issues and try to prioritze stuff that needs fixing. Ikea is one of the last. But hey, i can't change a mess of 2 years in simply 3 months. 

Frankly, i know a few users , in person, with 100+ nodes in the network. They have no issues at all. So it is not as generic as you think it might be.  There are over 65k installations of deCONZ. DE has their own lab. So yes, i am thankful for your report and you are trying everything you can. But there are so many factors. See the new bug on the Opple switches: I had a few reports on discord and here. After that, i can see they are related and confirm the bug.  It's just a label, don't worry.

If you really feel that nobody is doing anything to fix this: Please see the discussion in #1261 . There's a few there and i am shaking the tree whenever i can.  I'm happy to invite you over to start moderating with me and help users.

ChrisPrefect, 2020-10-18:
Hello Mimiix

OK, thank you for your explanation.

It is really frustrating and hopeless as a user. Since many, many months there are massive problems and nothing gets fixed so far.

How can we go forward? I REALLY want to pay for fixing this issues. I would spend 1'000€ if my Zigbee network would run stable (reliably react to button presses and Alexa-commands) and scenes would work properly. Can it be this hard?

You tell me that others run big Tradfri-Networks without issues. This gives me hope, but at the same time shows that there really needs to be some kind of paid support to find issues in my particular Zigbee network. If it works for others, it should be easily fixable without any code changes in deconz or phoscon, right?

Sometimes the lights work when I press the button on a Tradfri remote. Sometimes not. This should be debuggable somehow, right? But how? When I turn on a group of lights with Alexa, almost always only a few lights react. So turning on a group with Alexa or a remote somehow works completely different. Why? Who can support this? Is there a third party electrician/IT company that can support such an installation? Why doesn't Dresden Electronic offer paid support like most other open source companies?

What is the path for me to get usable lights in my home in the foreseeable future?

Thanks for your help!

Mimiix, 2020-10-19:
Hi Chris,

In going forward, let's start with these steps. This will provide me some details:)

Are you able to give me the following information?
- Full system specs of the host machine/it is running on. 
 Host OS,
RPI: What else is on there?
What SD are you using?
Running method (VM?Native? Docker? HA Addon? Hoops?)
Any other WiFi stuff going around?
Any other USB devices to the RPI?
What PSU are you using for the RPI?

In addition, can you add:
- Deconz Logs (Full) of 1 day of running. Using the full debug flags. Let's see if i can find some Errorcodes.

I also just noticed: You are on a older version of the Firmware of the conbee. Please use this guide: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually and update to the latest versions. As far as i know, there have been fixes for ikea devices. 

Alexa seems to be buggy, but i am not sure on that. Alexa uses the rest-api, and so does Phoscon. I've heard more users on this, but I have no clue on how it works with deCONZ. I've asked this to Manup. I'll come back to you on that.

Looking forward to your reply!

ChrisPrefect, 2020-10-19:
Hosts: Currently Raspberry Pi 4, before Raspi Pi 3B, but with same problems

Host: Currently downloaded beta deconz desktop image burned with etcher and updated manually. Before command line version, with same problems. Also tried on an Intel i5 Windows machine but with same problems.

Original Raspi 4 USB-C power supply. I even have a big fan on top of the Raspi 4 to cool it. It now only reaches 42°C

There is nothing else on the Raspberry. It's the original deconz image.

Tried different SDs Sandisk, Kingston, currently OV 32GB

No other USB devices on the Raspi. Conbee II is currently plugged in with a 2 m extension on the lower USB2.0 Port.

Raspi is connected with Ethernet to a Switch 30 cm away and that switch to a 24 port Unifi PoE Switch, connected to a Dream Machine Pro. I tried different USB extensions and USB ports and also bought a second Conbee II, but with the same issues.

There are 6 Unifi accesspoints around the house. I moved as many devices as possible to 5G8. I set the APs all to 2G4 WiFi channel 1. Not ideal for the WiFi, but farthest away from the Zigbee channel 25. I also tried unplugging the AP that is in the same closet as the Conbee II, but that did not change anything. There are no other WiFi around from neighbors.

There is however a separate Tradfri Zigbee network in the basement with its own Tradfri gateway. There are about 37 Tradfri lights (FLOALTS, GU10) and 8 remotes and 4 motion detectors on this network. I wanted to move all the devices from this network to the deconz-Network too, but I worry that this would make the deconz network even slower.

I updated deconz to 2.05.86 now. And Conbee II to 26660700. It's a problem that they are not numbered sequentially. I saw the "0700" at the end and thought  that this was the same version I already have...

What command do I need to run to get the logs for a day?

I read in the other thread that deconz can't handle commands that are sent simultaneously and needs several seconds between commands. If I use Alexa to turn on a group with 10 GU10 spots, Alexa will send 10 commands simultaneously to switch on each light individually. This always fails and only 40-60% of the lights actually get turned on.

Here is a video of the Alexa problem (watch with sound): https://www.youtube.com/watch?v=GPDQK2BT19Y

This is a BIG bug in the deconz network. All commands need to be executed reliably, no matter how many commands are sent simultaneuously.

Sometimes it takes literally 15-30 seconds for a simple button press on a Tradfri remote to turn on a group of two lights. What happens during this time? Is deconz sending millions of commands every second till the lights turn on? Or is the signal from the remote too weak (needs to travel several hops) to reach the coordinator, but the remote only retries every few seconds or so?

This is a Phoscon bug, I also added a video: #3265 
This issue is also annoying and you forwarded it 6 weeks ago. Was there any reaction or progress? #3135 

Thanks for your help!! 

Some (older) screenshots:

[image]

[image]

[image]

[image]

[image]

Mimiix, 2020-10-20:
Thanks for the detailed report! I'll start with the alexa part before going into your message. 

**The alexa part:** 
I asked manup on the alexa integration, that one is very basic and thus explaining what you mean:

_The Alexa "support" is simply using the fact that the Echo build int integration for Philips Hue (but not the Hue Skill) is using the same API REST-API. Alexa sees deCONZ just as an Philips Hue gateway._ 

This is something i heard more often with users. I doubt if we can fix this, as alexa is sending the commands and deconz is acting on them. I think we should make some specific alexa integration on this to fix this. deCONZ can send a lot of commands at the same time: I used to have some groups in HA to control 5+ lights in deCONZ. That went well (maybe 1 second between the first and last light?).  I think the bug is within Alexa, not deCONZ here. But, that could be logged too! **Todo Dennis: I'll get this with Manuel. I can't give a ETA on this.**

**Hardware**
The reason i asked for detailed hardware is 1: I wanted to see what happened on your systems, 2: the SD. In the past, i had a bad experience with a slow SD card which caused slow light transitions. I had only 5 - 8 devices in deconz, but the SD card slowed that madly like you experienced. Could you get me a serial/product ID of your SD? Just to make sure it is fast. **Todo Chris: Get serial of SD/SD Details**.

The WiFi is a bit busy, but i am mostly interested in the channel 25 of Zigbee. I have asked the devs if that could be a thing. I do know for a fact that Konke devices only like channel 15. Maybe that could be the case with ikea aswell. Worth a try in any case!  The network in the basement _shouldn't_ be a issue. But that would be visible from the logs too. **Todo Dennis: Get info if ch25 is a issue.**

As you are running native linux, you can enable logs by adding the following flags to your running script :
--dbg-info=2 --dbg-aps=1 --dbg-error=2 > debug.txt

That would create a file called 'debug.txt' in the place the deconz script is started. For more info on this, please check [this page](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags)  **Todo Chris: Enable logs and provide logs**

On the 15 - 30 seconds: I have no clue at all.  That should not happen.  However, i think it could be related to anything of the above, thus lets wait for the logs and see from there. The logs should provide [error codes](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log) which we can use to debug. 

On this one: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3135 I missed the ball. Sorry, been busy lately:( I'll check with manup for a reply from him. **Todo Dennis: Get Manup to reply to 3135** 

Let's see what happens from here :)

MattWestb, 2020-10-20:
For Zigbee channel 25 its one of the primary Light link channels (11,15, 20 and 25) =  all LL certified devices like old IKEA suld and is working on it and its also one demand in Zigbee 3 certification = all new IKEA devices.
Have 3 Zigbee network running on ch 15 (deCONZ), 20 (IKEA) and 25 (ZHA) without problems with IKEA devices (Wifi is blocking ch 11 for my) but not so extended as Chris.
Konke have only one device Zigbee certified (one new GW / HUB) = the old devices is not one real zigbee device.

ChrisPrefect, 2020-10-21:
Thanks for the input.

There is a difference if deCONZ sends a single command to a group or Alexa sends one command per light in a group. Alexa has no concept of Zigbee groups. It sees each light individually. You can group them together in the Alexa app, but that just means that Alexa will send the same command to all of these individual lights in this "virtual" group. But that fails with deCONZ because Alexa then sends 10-15 commands at the exact same time to the API and deCONZ can't handle this and the commands only reach a few of the lights and others are just lost.

I will clone the SD card to another card and see if that helps. But I already used several different cards with the RPI3 and RPI4 and when switching from deCONZ headless to the desktop/VNC image. How can I get the serial info from the card? I don't normally use Linux, sorry :-)

I also did a scan of the Wi-Fi / Zigbee band with an RF explorer and I could not see any persistent signals. The noise floor was at the same level as before and after the Wi-Fi band. When I pressed the buttons of a remote, I could see spikes around Zigbee channel 25. So the signal-to-noise ration is at least not zero and there are no other signals on this frequency.

I will try to enable logging (have to google a bit to learn about the running script).

Here is some small bug I just experienced when trying to pair a new Aqara temperature sensor:
 
[Phoscon App - Google Chrome 2020-10-22 00-26-22.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5419000/Phoscon.App.-.Google.Chrome.2020-10-22.00-26-22.zip)

Mimiix, 2020-10-22:
> I will clone the SD card to another card and see if that helps. But I already used several different cards with the RPI3 and RPI4 and when switching from deCONZ headless to the desktop/VNC image. How can I get the serial info from the card? I don't normally use Linux, sorry :-)

It's on the SD itself. I think on the back. But it should also say something about the class. 

> I will try to enable logging, but I had no luck so far. Where exactly do I need to set the logging options?

You need to edit the service file afaik. The ExecStart line. I am not too familiar with that myself. https://www.linode.com/docs/guides/introduction-to-systemctl/ refer to that. _I'm learning things too here_ 😄

ChrisPrefect, 2020-10-25:
I found out how to edit the service script and add the logging. But I don't see any log files being written. Is this normal?

Shouldn't there be a simple and clear how-to article available from Dresden Elektronik on how to provide these logs? It seems quite essential to debug bugs that customers report.

[image]

MattWestb, 2020-10-25:
@ChrisPrefect I think manup last finding can being one of the reasons for your problems https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-716002473  ;-(((

ChrisPrefect, 2020-10-25:
Interesting bug, and I think I also ran into this one maybe 6 times during the last year. But power cycling the light does fix it. Yes, it is very annoying if this happens, but at least there is a clear fix for it and it happens rarely.

But not being able to switch a group of lights reliably with Alexa every time is more annoying. Or the broken scenes and color temperature control with Tradfri FLOALTs. There are so many other bugs.

Also, Phoscon seems not to be actively developed anymore? Where are the on/off buttons for groups and individual lights? Moving the slider to turn them on surely is just a first draft, right?

MattWestb, 2020-10-25:
I wondering if the new firmware with source routing is working well with the IKEA bulbs or the coordinator is losing the routers / end devices and how its handling the routering then one one end device is changing its parent.
ZHA have doing great work with the routing and tracking where the end devices is coming from so not need doing broadcast finding them if they don't awaring then sleeping (With EZSP NCP).

I finding how Alexa is switching "groups of light" very strange. Not using zigbee light groupes (that is one broadcast / command) and sending individual commands to etch devices (1 command X devices) is not how you is working with lights in the zigbee world and abusing the system resources.

Then working with lights i using the "old web app" then is only using real light grupes and doing real zigbee binding that working that deCONZ is offline (Updating or have killing the SD card).
Phoscon is one big mess for implanting HA sensors and switches that is not working with binding lights (offline = not working) so i have moving all my sensors to ZHA with EZSP coordinator for eliminating the "Xiaomi problems" and only using "nice" routers for avoiding problems.

One question: Then one remote is offline in deCONZ is it sending group commands to the bulbs and is the bulbs reacting of the command ? 
Its interesting then group commands is broadcast and and not being routed routed thru the coordinator. Unicast (direct commands to one devices) is being routed thru the mesh and is the mesh having problems its failing then not finding the right route to the device.

ChrisPrefect, 2020-10-25:
Alexa can't use Zigbee groups because you can have one light in many different groups.

And it should really not be a problem to send two commands in total to a group with two lights. But even this fails regularly with Alexa. deconz somehow can't handle sending out commands in parallel.

MattWestb, 2020-10-25:
Then Alexa is little stone age equipment in the Xrocket age.
And Philips HUE is steering the lights around after wath is displaying on the TV (with no zigbee standard way for getting the speed) and the mesh is not collapsing / blocking and still having pretty good compatibility with other certified devices.

My experience is the that much of the problem is blocking / unsynced communication that is making pairing and other thing failing more than often (I don't believe that the mesh is blocking reading all readable attribute in one bulbs basic cluster that is 2 meter away (normally the sw version is not being red) and 20 seconds later i can reading the missing attribute manually and getting the response, i have testing it more times and it's the same).

Zigbee have one very bad limiting thing and its that the underlying network layer (IEEE 802.15.4) is limiting broadcast to around 1.2 / second. Unicast is not limited (only the "wire speed") and should not being any problems (like OTA file transfer). Then looking with wireshark is IKEAs GW is mostly using broadcast between all devices in the mesh (Lighting standard) and its working also then doing OTA updating of devices. 

Power on / of 2 lights is zero load for the mesh. 
One Zigpy dev have spamming the mesh by toggling 30 lights / second without killing the network (with one new TI coordinator). 
In ZHA i am using one IKEA ICC-1-A modul (its used of nearly all ikea devices) as coordinator and its have the same or better HW and RF performance as second gen Rasp/CornBee and is using sirlabs standard EZSP firmware (the coordinator firmware for the SOC) and one 7€ IKEA bulb have the same performance in the mesh if not the application is crashing in the firmware.

Mimiix, 2020-10-26:
> I found out how to edit the service script and add the logging. But I don't see any log files being written. Is this normal?
>
> Shouldn't there be a simple and clear how-to article available from [...]

I agree on that simple how-to article.  I think you need to add the file to the string. Here is a community created explanation [link](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-command-line-parameters#available-debug-parameters)  That logs are getting easier: There's going to be a event window implemented next beta.

ChrisPrefect, 2020-10-26:
It does not say how and where to add these parameters :-( How do I set up logging? Thanks!

Mimiix, 2020-10-27:
@SwoopX Are you able to explain a bit here?

SwoopX, 2020-10-30:
@Mimiix Erm, explain what? Copy&Paste?

See here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3406#issuecomment-713916340 The issue in the screenshot was not using the full path to the command + using the wrong command name. `/usr/bin/deCONZ`

ChrisPrefect, 2020-10-30:
Uhm, what? I don’t understand.

Are you referring to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3406#issuecomment-716073261 ?

The command and path where not changed. This is the original file as found in the deconz gui image.

Only the logging parameters where added.

But I don’t get any logfiles.

How do we enable logging, so that we can help you guys debugging?

SwoopX, 2020-10-30:
```
sudo systemctl stop deconz
sudo systemctl stop deconz-gui
/usr/bin/deCONZ --dbg-info=2 --dbg-aps=1 --dbg-error=2 > debug.txt
```

ChrisPrefect, 2020-10-30:
Doesn't work :-( 
[image]

SwoopX, 2020-10-30:
So, no gui...
`/usr/bin/deCONZ -platform minimal --dbg-info=2 --dbg-error=2 > debug.txt`

ChrisPrefect, 2020-10-31:
No, I use the GUI image.

Only "sudo systemctl stop deconz-gui" really stopps the service.

The new command doesn't work either: 
[image]

SwoopX, 2020-10-31:
Why do you think that? It's doing alright as far as I can tell...

ChrisPrefect, 2020-11-09:
Is there any update on this?

I just lost again four Xiaomi Aqara Sensors. They where paired, but now one is not responding, two have lost their names and defaulted back to just "Temperatursensor" or "Vibrationssensor" and one is completely missing from the sensors list.

Can it be that these sensors don't update the path in the mesh? When I pair them near the Conbee2 stick they work, but when I bring them to their dedicated room, which has a Tradfri repeater nearby, they are no longer connected.

I found another bug in the phoscon interface. It won't let me set the sensor to the name it had once, even tough it now has again its default name. I can't see why the old name is still blocked. No device shows the old name.

[Phoscon App - Google Chrome 2020-11-09 22-47-04.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5513477/Phoscon.App.-.Google.Chrome.2020-11-09.22-47-04.zip)

MattWestb, 2020-11-10:
@ChrisPrefect Mapnu have doing some commitments that hopefully is coming in the next beta. Hi trying getting down the reporting rate for groups and other unnecessary status that is broadcasted to groups and may cause broadcast storms and blocking the mesh. https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3614#issue-517376635
I hope hi is managen getting it right then getting broadcast storms the underlying networks is going in blocking state and being irresponsible.

Mimiix, 2020-11-10:
@ChrisPrefect I am waiting on your logs 😅 

But as we are close: On the next Beta there should be a dedicated Log window within deconz. Easier to get logs by then.

andordavoti, 2020-11-10:
Facing the same issue. My remotes used to show up as the "Basement Floor Lamp" in the screenshot, this way I could directly bind the remote to lights. Now they don't show up anymore. Using Deconz as an add-on in Home Assitant.
[image]

Mimiix, 2020-11-11:
@andordavoti You don't have the same issue, i'm for sure 😅 . To fix yours, use the old WebUI. On discord it shows in the #faq.

ChrisPrefect, 2020-11-22:
Another bug: Tradfri Wall-Sockets in a group in Alexa. Turning them off with Alexa works, turning them on works too, but Alexa always says "Weihnachten does not reply. Check network connection and power of the device".

Is this already reported in another issue?

[image]

Mimiix, 2020-11-25:
@ChrisPrefect Should report in the Phoscon repo.

However, how are we on the logs?

ChrisPrefect, 2020-11-25:
Is this really a Phoscon bug? Just because I postes a screenshot from Phoscon does not mean that it has anything to do with Phoscon. I strongly suggest that this is a REST API bug. The devices are found individually by Alexa and I add them to a new group within the Alexa app. Nothing here is using Phoscon. I just used a screenshot from Phoscon to show the devices. Maybe I should change it to not make it confusing.

[image]

ChrisPrefect, 2020-11-25:
Hilft das irgendwie weiter? 

[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5599593/debug.txt)

Mimiix, 2020-11-26:
@ChrisPrefect Is that a log from a situation where the remotes don't do anything anymore? There seems to be no issues with that log.

Indeed, no Phoscon issue on the alexa. Either search this git for a report on Alexa yet or create a new one ^^.

ChrisPrefect, 2020-11-26:
There are far too many open and closed/unresolved issues to search for and check if it might be related. I will document all bugs I personally have here and you can assign the reports to other issues or create new ones if that is necessary. There are so many small and big bugs, I don't think it makes sense to dissect them into sub parts as they all seem connected and related.

That's the log that was started a few weeks ago. But I guess it gets overwritten when deconz starts again? We disconnected the power yesterday to hook up a solar array. There are no dates in the log file?

Mimiix, 2020-11-26:
> I will document all bugs I personally have here and you can assign the reports to other issues or create new ones if that is necessary.  

This is not something we are going to facilitate. You have a search button. If you can't find a issue with the related issue, create a new one. Nobody is going to stick to this topic to be your help here for your own bugs/issues, that's undoable. I hope you understand that.

> But I guess it gets overwritten when deconz starts again?  

I am not sure of your timezone(you always seem to reply in my night hours so i guess you are not in EU), but i think the log starts whenether you open the log window.   The times in the log indicate that too, as your log started at 01:06:12:080 and ended at 01:06:12:080 .  So i'd suggest opening the log whenever you have the issue / remotes not working.

ChrisPrefect, 2020-11-26:
I am an end-customer. I bought a commercial product which is faulty. I used ALL the required support-channels to get the product fixed. But it is still not fixed. I don't get paid to research and document bugs here. The employees of Dresden Electronics get paid to do that. All I can offer is to point out bugs and dump all the information I have here. Everything else is the responsibility of Dresden Electronics.

I already invested maybe 60 hours in bug fixing and documenting this system. And end-user should never have to do this.

I also tried opening separate tickets for separate bugs, but they went nowhere. They did not get fixed, they did not get recognized, and they all seem like a complete waste of time. How many employees from Dresden Electronics are currently developing this system? It feels like it's just supported by 2-3 volunteer programmers?

Again, I don't get paid to bug fix this system, so during the day I have other jobs and only at night, in my spare time, I am able to respond to the questions here and provide more free work to help you guys fixing this broken system.

So even the logging isn't working at all. There should be about 4 weeks of logging in that file. Why is all this data missing? When will the "user friendly" logging be available?

Mimiix, 2020-11-26:
Alright, then i'll leave you to it. As I'm afraid I can't help you. 

I'll suggest you email or Call DE and solve it that way.

**I will keep monitoring this issue, like all others, on a moderation note.**

MattWestb, 2020-11-26:
I only saying you are not alone in EU that have the customer rights but dont getting it from DE @ChrisPrefect 
Many of them is moving on to . . .

Mimiix, 2020-11-26:
**@MattWestb Please stick on-topic.**

ChrisPrefect, 2020-11-26:
> I'll suggest you email or Call DE and solve it that way.

I ALREADY DID!!!! I CALLED THEM MANY TIMES AND TALKED FOR AT LEAST TWO HOURS IN TOTAL! AND I SENT THEM DOZENS OF MAILS WITH A LOT OF INFORMATION AND NOTHING HAPPENED! NOT EVEN AN ANSWER AT ALL!!!

This is all fucked up...

WHAT DO I HAVE TO DO TO GET THIS PRODUCT WORKING CORRECTLY???

ChrisPrefect, 2020-11-26:
> **@MattWestb Please stick on-topic.**

he IS sticking to the topic! The topic of this whole issue is that there are many, many bugs and nobody is fixing them!

Mimiix, 2020-11-26:
> And people like Mimiix are actively driving off people that seek help and are willing to provide FREE labor and support fixing the bugs.

 **And now you crossed my line. I am locking this issue until tomorrow. ** 

I am probably the one that invested the most private time into this project, you throw a lot at me. I took all of it, but this, this feels just ungrateful and just hurts.

Issue locked.

manup, 2020-11-26:
> many bugs and nobody is fixing them!

I completely understand the problems you see are annoying and wanting everything stable and fixed is sure a goal for everyone here.

A lot of things *get* fixed often with the help of this great community.
Bug fixes are listed in each release note which are published every month, please don't ignore them and say nothing is been done.

With the wide range of products and different Zigbee setups naturally there can problems, which most people never see but some get bitten hard (like in this case). Currently there are >95 K gateways out there, it's not possible to get every issue to be fixed fast, some are more difficult to fix some are easier, same for questions. There is always progress, unfortunately not everything can be addressed in a timely manner.

The story with Ikea lights is a long and complex one. In some setups, mostly larger ones bugs do occur which are not easy to fix (if they can be fixed at all from our side) Please note the same problems also happen with the original Ikea gateway.

The investigation in the Ikea issues (like the other 2 years long Github issue) eats heaps of time and so much has been tried in the progress, in the hope to get routing in line, so far the source routing approach looks the most promising.

The upcoming release gets improvements in that regard as well which helps with lights which can be brought back by pressing `0`. But these are just steps, nobody has found the Ikea silver bullet so far. And there are also bugs with Ikea which we just can't fix, for example some lights tend to crash somehow in largish intervals, they are still visible in the sniffer to don't communicate with other devices. Such firmware bugs can only be fixed by Ikea.

For example in my kitchen I've a Ikea E27 light which is always powered and crashes every few months for unknown reasons. When this happens not only the light isn't controllable anymore but also the Aqara motion sensor which happens to be paired to it won't get the messages through to the coordinator anymore. Currently the only way to fix this is to power-cycle the light, I tried all kinds of weird Zigbee tricks in countless nights to get a light in that state to come back but so far it's a dead end. It looks like the lights incoming NWK Layer message queue is stuck.

Non the less tests with Ikea are made regularly to see if the situation can be improved to get them as reliable as Philips Hue bulbs. Next in line will be automatically created source routes based on the data of the neighbor tables, to basically sideline device mesh routing.

And of course I also hope that new Ikea OTA updates get released to help in that regard.

-----

But back to this issue with the remotes, usually they work very well in moderate setups. By default the round Ikea remotes send group casts out which the gateway picks up and translates them into button events.

What would be helpful is a log with Info debug level, where a remote which is seemingly dead is triggered by button press:
We also need the MAC address of the remote to see which device is it in the log.

Here is an example for my round Ikea remote with MAC address 0x00158D0002A1E952 when I press the up button.
The last line is the incoming group cast.

```
20:34:07:478 [INFO] - Button 2002 Step up (with on/off) TRADFRI remote control
20:34:07:527 ZCL attribute report 0x00158D0002A1E952 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
```

If that doesn't occur in your log when you press a button it would be an indication that something lower level isn't working, could be routing, but usually group casts work well as long as the parent node of the remote forwards them.

Mimiix, 2020-11-27:
**As mentioned: I'll unlock today. If any other comment goes outside of our scope: Fixing this issue, i'll lock it again.**

ChrisPrefect, 2020-11-27:
Thank you manup!

I see the problems. I see how frustrating this can be. Thanks for your work.

But: It is even more frustrating for end users. DE sells a product. This product is in many ways faulty and frustrating to work with. These problems are not communicated clearly. End-users like me have to deal with these bugs EVERY DAY. And we also have to defend the system against our wives. A button that works 80% of the time is broken and needs to be fixed. And it needs to be fixed within days or weeks, not months or years or not at all.

Power-cycling bulbs, shouting at Alexa, going to a desktop computer (because phoscon on mobile does not work) just to switch a stuck light back to the correct color temperature or re-binding a sensor EVERY DAY is very, very tiring...

Mimiix: The Issue IS NOT some single bugs but the whole bug- and support situation in general. So the scope here is exactly what we are discussing here. Or should we open another ticket for that? No problem. Please provide a ticket where we can discuss the whole situation of the project and the state of support from DE.

As DE does not offer any other support, does not react to e-mails and has no support forum, THIS is the official place for support. DE even told me that on the phone. All support is handled trough Github issues.

You repeatedly tell people who try to report bugs that they did report it somehow wrong. Don't do that. If you think the information provided is in the wrong form, you are free to clean up. But don't put people off like that who are trying to get help.

You told me several times that some bug I reported is a phoscon bug, just because you saw a screenshot from phoscon. In all cases, it was NOT a phoscon bug. But we lost time and energy just because you discussed this technicality (and where in fact wrong about it).

Locking threads is very cowardly and just makes everyone involved more angry. This stuff has to be discussed. And there are many emotions involved. YOU HAVE TO ACCEPT THIS! If you lock threads here, we can also take this to the several home automation Facebook groups, where you have no control over it.

Again: This is not an open source github project. This is the official end user support page for DE. As long as there is no other support from DE, you have to accept this too.

MattWestb, 2020-11-27:
I agree with @ChrisPrefect  and its hurting my very very much that the official support channel is isaying "ist not one deconz-rest-plugin issue its Phoscon" and vise versa and then ignoring helping the customer (**NOT USER THEY HAVE PAYING FOR THE PRODUCT)** and some time only closing it with some comments that one company dont daring saying to one customer.

Back to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3465.
I like to having one official statement from the company about it !
The customer have trying very hard getting his system working for 5 weeks and hi still dont having one working system but the real support not really helping and if its one hardware problem or software have not being cleared but both is the sellers problem not the custom !
Hi is using devices that is recommended by the company (Philips HUE, IKEA and Xiaomi) and no DIY or strange no name devices.

I have professional experience of purchasing large communication systems and have also throwing some system out completely and my the organisation was taking the full loss of time (2 years) and the economical (was not possible getting some money from the company because they  wasn't not having so much left that we have paying and didn't like putting them in bankruptcy thru making one juristick process) But it was only some millions € wasted.

The company have great knowledge and the products suld being premium clase, but somthing is missing in the end that all private customers in EU have right to getting for the byed product from one company, support !

If not the company cant supplying support for the product they is selling then then it can being very expensive not only economical but also the reputation is getting very bad damage.

As long Manuel P or some other of his competent colleagues not have enough time to giving the customer technical support its still one large internal orginasion problem for the company.

Only doing the support without one ticket system i have not word for (have ITIL cert and have working under ISO 9001 for many years, if some is knowing what its standing for).

And the most sadly thing is that all users is needing Manuel P for getting the real deep problem solved here but its looks that hi is not possible to being cloned and cant spending enough time for doing it.

If the official represents from the company in this github like continuing threating and dont like hearing some truth i can living with being permanent blocked in this "github / support portal". In the end its one company policy how treating there paying customers (as long they is having some ?)

I like the products and functionality and I dont like closing doors but if its necessarily its they is other doors that is open that can handle large zigbee system and its working in real.

LG

manup, 2020-11-27:
Hey guys I've written a few lines about the support situation in
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3765

Believe me I understand the frustration when things don't work. We are all more than busy and eager on fixing things to improve that. Please note issues are very often not systematic and hard to reproduce, I would *love* to have a very buggy setup at home or in the company to quickly investigate and fix things, but for most part my setup is rock stable as it is for many people. Because of this we've started to build a larger mixed "real world" test setup in the company, but here the Ikea routing can't be seen so far, we do extend this network further to hopefully trigger more issues.

Note that bugs are investigated a lot, I can't guarantee you this particular issue is going to be fixed in time X, that's simply not possible. I can only provide my help here to figure stuff out and hopefully fix the issue, but here I need your help (like the requested logs in my last comment).

If that's not a viable way, I would recommend to return the product and get a refund, we do whatever we can here but everything takes time.

I suggest in this issue we focus on the tech from now on.

ChrisPrefect, 2020-11-27:
I would love to give you full access to my setup. You can get a VPN with SSH and VNC/teamviewer access and even live cams to see the lights react if this is necessary :-) 

There is no easy way to "return it and get a refund". It's not about the 40€ or so. In fact, I even bought multiple Conbee-Sticks just to make sure I didn't get a faulty one. I actually need this setup. I have several Tradfri-Hubs, Hue-Hubs, cc2531 sticks and also the new dev-board from Ti lying around. But they all have their downsides, like having to script each and every function manually in ioBroker for example. This is where Phoscon provides all the basic functionality. And switching to another setup is straight up three days of 100% work. So not really something I would try over the weekend...

Is it really this hard to fix some basic problems, like tradfri-lamps that should be turned off in a scene, or Alexa not being able to reach all lamps in a group, or the color temperature problem with tradfri, or the "something went wrong" Alexa response when changing color temperature even though the command was successful etc. I would imagine that if after one week of debugging and coding these bugs should be gone forever, right?

SwoopX, 2020-11-28:
I might have found how this manifests and therefore, why devices get unresponsive and require a 0 press in the middle of the night together with manup. He mentioned toe corresponding code is complex, so give him some time. Potentially the silver bullet to this.

walthowd, 2020-11-28:
Just wanted to pop in and confirm I see this same issue with IKEA outlets well as Ecosmart bulbs, all of which I believe are on the EFR32MG1P platform. 

Issue seems to be exacerbated by a group of six to twelve devices simultaneously joining the network / and announcing. 

Devices will partially crash, broadcast link status frames with zero links, not respond to unicast/aps frames but respond and react to group/broadcast commands (on/off). 

Child end devices of crashed routers will never find a new parent ( they seem to think crashed parent is working ). Crashed router will not forward all frames from end devices. For example, a aqara motion sensor will send an occupancy attribute report which is lost but the lux attribute report is successfully delivered. 

I'm a Home Assistant ZHA user and have reproduced this behavior with both a SiLabs coordinator and a TI ZStack 3 coordinator coordinator. 

As this affects both IKEA and other EFR32 platforms I assume the underlying root cause is a SiLabs Emberznet bug in the NCP firmware.

MattWestb, 2020-11-28:
I still finding it interesting if the device is communicating with no or some device.
Some findings that manpu have done looks like the security frame counter is routing of sync in the router.
For some days i was restarting my HA installation for updating the implanting of the tuya TRV and testing them. HA was having problem starting some integrations and the EZSP was going in falur state some times. After I was having 2 IKEA controlled outlets and one CWS bulb that was not forwarding communications from there children. The controlled outlets was working after repowering. The bulb was reacting from the first on groupe command from the remote bounded to the groupe and then nada. Repower it was working. This looks 110% ad tasmota was having then resetting the security frame counters on restart of the coordinator. First broadcast command is working OK, then the bulb is sending the attribute report with unicast to the coordinator and the feme counter is bad and blocking all traffic until being repowered.

The Xiaomi Aqara is interesting how the devices is sending both type of message, think unicast and to which address in the network (normally 0x0000 to the coordinator).
Perhaps the stack is partly alive and still can routing some types of frames.

My feeling is that its some strange in the silabs network layer / stack that can messing the security frame counter in the device.

walthowd, 2020-11-29:
Does this seem like a likely culprit? 

[image]

From:
https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.7.4.0.pdf

MattWestb, 2020-11-29:
@walthowd I think you have both right and wrong in the findings.
Its looks ECC is used CBKE and with Zigbee Smart Energy (ZSE) Security that we is normally not using here. [Reference AN1233: Zigbee Security](https://www.silabs.com/documents/public/application-notes/an1233-zigbee-security.pdf)
But silabs is not writing that is only then using CBKE only "such as CBKE".
Assuming its being used the the TC is issuing one pare application link key and the devices is using the ECC algorithm for en/decrypting the key exchange for both devices. I dont knowing if deCONZ is issuing application link keys but if. The application link key is used then 2 devices is exchaching privat information that is encrypted only known of the 2 devices.

The second scenario is then one zigbee 3 devices is Requesting a New Link Key after Joining.
But it should only happens then joining one device so not very likely happen after the system is running and not adding new devices.

I have finding @manup scenario then moving one float panele and the network was going of sync / braking down.

In the paper a bow its also describing the security mechanism and the "2.1.2 Hop-by-Hop Security" and "2.1.6 The Network Security Frame Counter" that is very essential for the functionality for one zigbee 3 network.

If one device all security frame counter is unsynced then its completely dead and cant communicating. But what is happening then one link is going of sync in one direction (the counter is not one its on in and one out) so cant communicating thru the way / route / link but all other links / routes is working. Then broadcasts is working (becos some links is still working) until some device is trying  sending  somthing back to the source of the broadcasting device unicast address and this not working in that direction (its like having one serial com port that that can sending but not transmitting or vise versa). 
My example from yesterday: Groupe command (broadcast) 0x0006 on/off on. The light is turning on and sending one attribute repost of its changed state that is sending in unicast to the coordinator 0x0000. If the unicast route is broken in one direction because frame counter is of sync between 2 devices in the route / link the command is being missed or the ack is being missed. The mesh is flagging the link as not working and all command must finding on new working route to the devices if its possible. If finding one working route then its OK but if it was the last working route / link that was working and the link is flagged as not working its being one completely break down of the communication to the device (and for devices that must using it for communicating). Its little complicating getting all functions working together and understanding the how its working then somthing is going wrong.

One problem is that the security frame counter should not being resorted to zero then one device is being re powered but perhaps Silabs is accepting one larger diff after repower and syncing it with its neighbours then the devices is sending one device acunsment. 
The tasmota security frame counter problem was always going away then repowe one of the zigbee 3 devices and it was only affecting zigbee 3 devices and no not zigbee 3 devices like Xiaomi sensors and old LL firmware IKEA bulbs. But if one LL device is behind one or more zigbee 3 it can being "isolated". then the last link is flagged bad.

Its only brainstorming and its little deep but perhaps its getting some clue to finding or going around the problem without doing like some user is doing (repower the network every midnight).

Its definitely one zigbee 3 problem and perhaps one silabs one but not exclusive IKEA devices sa we have seen the last time. More is coming the Sonoff and tuya is rolling out more and more new devices with EFR32 NVPs every day.

One PM is that IKEA is not working in centralised mode (with trust center) they is working in distributed security mode and its like small groups of lights is connected thru one groupe that one controlling device is controlling and other groups is more or less autonome but is using the same network. 
And they cant building large networks like some of our users is doing so they dont have our problems.

walthowd, 2020-12-01:
Perhaps a post on the SiLabs community forums could help us determine which bug might be applicable? Obviously directly contacting the correct person in charge of zigbee firmware at IKEA would be most helpful, but finding them seems like a challenge. If they were contacted, it would also be interesting to see if they are aware they are publishing broken firmware for several of their devices.

MattWestb, 2020-12-01:
I have asking for adding config group support for the signal repeater (that is only floating around) and the remote so the scene changing can being bonded to other groupe then 0Xff09 for getting default them working better with no touchlink systems (Group bindings is being setted with touchlink command then pariting) but still no answer.

IKEA can only winning if helping solving our problems and can selling more of their products around the globe.

Some devs is having Silabs support with the dev kits perhaps some can posting the question in the support forum.

[16 more replies]

closed 2021-04-10 by github-actions[bot]