dresden-elektronik/deconz-rest-plugin #3407: Tuya TS0601 thermostat : Wrong time syncronization

helmuteke, 2020-10-13:
## Describe the bug

At this thermostat there is a intern clock
When we use a tuya gateway the thermostat is syncronize correct , when using the conbee gateway there is always a wrong time.
But seems to be a random time.

## Steps to reproduce the behavior
Not reproducable , it happens always.
Used one thermostat on Conbee gateway , wrong time
Other on Tuya at same moment , perfect time

## Expected behavior
Syncronize a good time

## Screenshots

## Environment

Commercial version 2.05.81
Firmware version 0x26580700

 - Host system:(Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: 26580700
 - deCONZ version: 2.05.81
 - Device: ConBee II
 - Do you use an USB extension cable: no

### deCONZ Logs

## Additional context

closed 2020-11-24 by stale[bot]

Smanar, 2020-12-12:
@helmuteke Hello, do you have an unix machine to compile a code to make some tries ?

neonal, 2021-01-07:
@Smanar I have the exact same problem. When TRV is connected to deCONZ it gets a completely wrong time from the gateway. The time isn't skewed like couple of hours, it's completely wrong. Wrong in the sense of it takes a day for half an hour to pass on the TRV. Let me know if I can help, my RaspBee is connected to RPi so I think I can compile.

Smanar, 2021-01-07:
We are on this problem on this issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3304
BTW whar are your device ? he is using the tuya cluster, the Time cluster or both ?