dresden-elektronik/deconz-rest-plugin #3410: Conbee II disappearing

cweber9843, 2020-10-13:
## Describe the question or issue you are having

I experience similar behavior mentioned in other issues: After a while of stable operation, the Conbee II stick became unstable and drops out of the system, only to reappear some seconds later on a new USB Bus location (see the lsusb log, note that `Bus 002 Device XXX` keeps counting up). I think it never lives long enough to get a new ttyACM* assigned. 

- I tried with and without USB cable, and different USB ports. 
- In contrast to others, I have deCONZ installed directly from apt, not in a docker container. No other sticks (e.g. Z-Wave) are present and Bluetooth is turned off. 
- I also tried binding the Conbee II as suggested by @SwoopX in his [reply to issue #3341](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3341#issuecomment-703501616).

I noticed one point: the deCONZ log notes api version 2.05.81, whereas `dpkg -s deconz` insists that version 2.05.84-ubuntu-bionic-stable is installed. 

And to finally make the issue a question: Any hints on how to fix this? 

## Screenshots

## Environment

 - Host system: Dell Ultrabook Latitude E7450 
 - Running method: Both Ubuntu 18.04 and 20.04, as system service 
 - Firmware version: 26660700 (from log)
 - deCONZ version: 2.05.81 (from log), but apt says 2.05.84 is installed. 
 - Device: ConBee II 
 - Do you use an USB extension cable: yes (tried with and without)

### deCONZ Logs

`lsusb:` [lsusb_log.log](https://www.jottacloud.com/s/0563eeadd617bba470ba934d2aeade215ff)
`deCONZ --dbg-error=2 --dbg-info=2 -platform_minimal:` [deconz_log_20201012.log](https://www.jottacloud.com/s/0560ca8afd16250403a991a7881a099bbef)

## Additional context

1xIKEA Trådfri remote
1xIKEA Trådfri LED driver
1xIKEA Trådfri motion sensor

SwoopX, 2020-10-14:
You have not yet pinned deconz to the tty alias yes. You can do so by adding `--dev=/dev/YOURTTYALIAS` to the command line call.

cweber9843, 2020-10-14:
Indeed, I am not sure how to do this. I edited the `conbee.rules`:
```
# ConBee II disbale modem manager and mtp probing
ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{MTP_NO_PROBE}="1"
KERNEL=="ttyACM?", SUBSYSTEMS=="usb", DRIVERS=="usb", ACTION=="add", ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", SYMLINK+="
ttyConBee2", GROUP="dialout", MODE="0660"
```
Reloaded the configuration with `sudo udevadm control --reload-rules ` and now created a [new log](https://www.jottacloud.com/s/056e95bde00d7b34e678ed6cac5a1f86214) with:
`deCONZ --dbg-error=2 --dbg-info=2 --dev=/dev/ttyConBee2 -platform minimal`

Looks to me like the new symlink is not used, am I missing something more here? 

And, as you can observe in the new log, ConBeeII connected to ttyACM0 for a short while, before dropping out again.

cweber9843, 2020-10-15:
I now connected the ConBeeII to a PC with Windows 10 and experience the same behavior: The stick disappears and reappears randomly, see screenshot below. The DeCONZ software can't connect.
[image]

SwoopX, 2020-10-15:
I've asked @manup to have a look at that. Looks pretty odd.

manup, 2020-10-16:
If the firmware doesn't manage to connect at all, can you please try to start the flash process for the latest firmware version again as described in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-windows

With the -t 60 parameter it should pick up the update.

If the issue still persists, I'm afraid the USB connector of the stick might be too lose, in this case please request and replacement. Did you order via Amazon?

cweber9843, 2020-10-18:
Thanks for the hint, I now flashed 0x26580700 on a windows system, successfully ran deCONZ there, then connected the ConBeeII to my linux system and was able to connect to start deCONZ and connect to the network again! 
Maybe a firmware downgrade will also help others with the connection issues (#3430, #3341).
I noticed that the phoscon app wants to install a raspbee firmware - can this be the reason for the connection problems, when deCONZ is running on a PC system?

cweber9843, 2020-10-18:
...and here is the confirmation, after updating to current firmware through the phoscon app, the stick is gone again on ubuntu.

closed 2020-10-18 by cweber9843

codyzu, 2021-05-24:
A note for anyone else reading this issue. I had the same problem after trying to update my ConBee II from the phoscon app. It would appear and disappear endlessly on my raspberry pi. I could observe this by running `lsusb` and `sudo ls -la /dev/serial/by-id/`. Trying to flash it with `GCFFlasher_internal` on the pi wouldn't work either. Running `sudo GCFFlasher_internal -l` it would appear as a `RaspBee` (maybe that is just default on the pi, since it would be a serial device).

I connected it to my Ubuntu PC and could observe the same behavior. Notably, when it was working, it would correctly be shown as a `ConBee II` with `GCFFlasher_internal -l`. After several tries with GCFFlasher on my PC, it finally flashed with success and became stable (would not disappear). It then worked on the raspberry pi also.

ankostis, 2021-05-30:
@codyzu With which firmware version have you achieved stability?