dresden-elektronik/deconz-rest-plugin #3411: New device - water sensor

stappel, 2020-10-14:
Please add the following device as water sensor:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2020-10-14:
Closing this issue. Not using template.

closed 2020-10-14 by Mimiix

SwoopX, 2020-10-14:
No need to reopen/create a new issue anyway, device is supported.