dresden-elektronik/deconz-rest-plugin #3415: how to start Deconz-GUI minimized on Raspberry pi

Deficiency1984, 2020-10-14:
Hi.

How can i start DECONZ-GUI minimized in Raspberry pi. I use Raspbian.
or is there a non UI Mode available for DECONZ?

Thanks
Florian

jose-d, 2020-10-14:
non-UI mode is actually implemented in this binary `/usr/bin/deCONZ`.

Smanar, 2020-10-14:
Minimized or headless ?

For headless
```
$ sudo systemctl disable deconz-gui
$ sudo systemctl stop deconz-gui
$ sudo systemctl enable deconz
```

closed 2020-11-24 by stale[bot]

felker, 2020-12-01:
@Smanar is there a way to open the GUI minimized and have it automatically select Conbee instead of Raspbee? 

When I setup the headless to start automatically, it prevented me from opening the GUI.

Smanar, 2020-12-01:
For minimze IDK, but even if at start deconz search for raspbee after 1 mn it switch for conbee itself.

tebeso, 2022-08-24:
Found this ticket here in the rest-plugin area.. not sure why, but anyways.. i would also like to have a minimized startup function (not headless)