dresden-elektronik/deconz-rest-plugin #3419: Philips hue outdoor 1600 lumens detected but turns unavailable

maxandersen, 2020-10-15:
## Describe the question or issue you are having

I got 3 new bulbs philips white lighting and I added them and they all have the same problem.  

They get detected but very soon after they go unavailable. 

## Screenshots

## Environment

 - Host system: raspberry pi
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: (26350500)
 - deCONZ version: (2.05.84)
 - Device: ConBee I 
 - Do you use an USB extension cable: no 

### deCONZ Logs

## Additional context

Mimiix, 2020-10-15:
Hi. 

Missing information here. Any logs?

maxandersen, 2020-10-16:
sorry - i was trying to locate info on how i get those logs when its via homeassistant deconz plugin.

Any pointers for that ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-switches seem to assume command line access which I haven't been able to locate yet.

Smanar, 2020-10-16:
They go unavailable, but can you still use them (with deconz or using broadcast command) ?
Do they come back after a power cycle ?

maxandersen, 2020-10-16:
They go unavailable and stays unavailable. 

Power on/off of lamps nor deconz makes a difference. 

Repairing I get control enough to have them blink when identifying but then go unavailable eventually.

Smanar, 2020-10-16:
Where are you seing the bulb joining he network ? on phoscon or deconz ?

maxandersen, 2020-10-17:
Phoscon

Smanar, 2020-10-17:
As you have the conbee 1, can you make a ty with the touchlink feature ? in the old web app to reset the bulb ?

closed 2020-12-19 by stale[bot]