dresden-elektronik/deconz-rest-plugin #3424: Tuya TS0012 switches

bastshoes, 2020-10-16:
## Describe the bug

After upgrade to 2.05.85-beta Tuya TS0012 switches disappeared from the list. API is also do not show them. They are still joined to network and can be operated via PhosconApp.

## Steps to reproduce the behavior

Install new version.

## Expected behavior

Switches should persist after upgrade.

## Screenshots

[image]
[image]

## Environment

 - Host system: PC
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.05.84
 - Device: ConBee II 
 - Do you use an USB extension cable: yes

### deCONZ Logs

## Additional context

Smanar, 2020-10-16:
Hello how many device have you (To know the database size for log)
And if you haven't too much of them can you share the log at startup ? when the code parse the database.

Was wired siwtch ?

wich one version have you before ?

bastshoes, 2020-10-17:
Hello. Yes this is wired switch without netural line. And yes I was on. 84

Smanar, 2020-10-17:
ok, so it s my fault sorry.

Your device is a "light" but declared as sensor by itself.

I have some problem because lot of tuya "sensor" are reconized as "light" so I have decreased the "detection tolerance", and your device haven't succed with it.

bastshoes, 2020-10-17:
Yes, Tuya devices are tricky. Do need any additional info for establishing right criteria?

Smanar, 2020-10-18:
No it s ok, I know exactly the problem.

I think I will "autorise all" and use a "blacklist" to prevent special device.
The problem is I don't know if there more "zigbee compliant" device or the reverse ^^.

You want the correction for your installation ? But I don't thing you can recompile the code on your side ?

bastshoes, 2020-10-18:
I'll wait for release. Thanks.

closed 2020-11-15 by manup