dresden-elektronik/deconz-rest-plugin #3426: Deconz in Docker

swissmaster1, 2020-10-16:
Hi I'm using deconz in Docker on my Synology Diskstation DS1815+ (latest DSM) for a while now. I used alway latest marthoc/deconz build. Yesterday the deconz image in docker was updated to 2.05.73. Since then it doesnt start anymore because of an error:

 error while loading shared libraries: libQt5Core.so.5: cannot open shared object file: No such file or directory

I've read that this error is because of an old linux kernel used by my Synology Diskstation. I now installed deconz image version 2.05.70 which is running again without problems. Is there anything that I can use latest deconz images without error? (updating linux kernel of Synology Diskstation is not possible).

SAOPP, 2020-10-16:
U confused the repos for your question, but today Martin released a test build with a fix for Syno fans.

Check it: https://github.com/marthoc/docker-deconz/issues/292#issuecomment-710122578

swissmaster1, 2020-10-17:
You are right. Didnt see that. Thanks

closed 2020-10-17 by swissmaster1