dresden-elektronik/deconz-rest-plugin #3429: ConBee II Stick has the status "Joining ..." (Newly purchased / Latest firmware)

kkillius, 2020-10-16:
## Describe the bug

My newly purchased ConBee II Stick does not run on any platform. My main test runs on a Raspberry 3. I also completely reinstalled it. I completely followed the instructions below:
- https://phoscon.de/de/conbee2/install#raspbian
- https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

## Expected behavior

ConBee II Stick has the status "Joining ..."

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi 3
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.05.85
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs

pi@WETRIX-PI:~ $ deCONZ --dbg-error=2 --dbg-info=2 --dbg-ota=0 --dbg-zdp=0 --dbg-zcl=0 --db-aps=0 --dbg-http=0
QCoreApplication::arguments: Please instantiate the QApplication object first
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
libpng warning: iCCP: known incorrect sRGB profile
23:19:45:417 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
23:19:45:473 CTRL. 3.27.223:19:45:790 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
23:19:45:790 dev /dev/ttyAMA0 (/dev/ttyAMA0)
23:19:45:790 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
23:19:45:791 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
23:19:46:171 parent process bash
23:19:46:171 gw run mode: normal
23:19:46:171 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
23:19:46:172 sd-card cid: 1b534d303030303010ffffffff00f919
23:19:46:174 DB sqlite version 3.27.2
23:19:46:175 DB PRAGMA page_count: 33
23:19:46:175 DB PRAGMA page_size: 4096
23:19:46:175 DB PRAGMA freelist_count: 0
23:19:46:175 DB file size 135168 bytes, free pages 0
23:19:46:176 DB PRAGMA user_version: 7
23:19:46:176 DB cleanup
23:19:46:176 DB create temporary views
23:19:46:177 DB view [0] created
23:19:46:178 DB view [1] created
23:19:46:178 DB view [2] created
23:19:46:179 DB view [3] created
23:19:46:179 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
23:19:46:180 sql exec SELECT key FROM config2
23:19:46:180 sql exec SELECT key,value FROM config2
23:19:46:180 Load config UTC: 2020-10-16T21:11:59 from db.
23:19:46:180 Load config announceinterval: 45 from db.
23:19:46:180 Load config announceurl: https://phoscon.de/discover from db.
23:19:46:180 Load config apiversion: 2.05.85 from db.
23:19:46:181 Load config bridgeid: 00212EFFFF05FA1A from db.
23:19:46:181 Load config datastoreversion: 93 from db.
23:19:46:181 Load config dhcp: true from db.
23:19:46:181 Load config disablePermitJoinAutoOff: false from db.
23:19:46:181 Load config discovery: false from db.
23:19:46:181 Load config factorynew: false from db.
23:19:46:181 Load config fwneedupdate: false from db.
23:19:46:181 Load config fwupdatestate: idle from db.
23:19:46:181 Load config fwversion: 0x26660700 from db.
23:19:46:181 Load config gateway: 127.0.0.1 from db.
23:19:46:181 Load config group0: 65520 from db.
[... 327 lines cut ...]
21:20:46:051 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x0000
21:20:46:066 Master: read param with arg 0x19
21:20:46:119 Current channel 11
21:20:46:148 CTRL got nwk update id 0
21:20:46:161 CTRL ANT_CTRL 0x03
21:20:46:260 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
21:20:47:424 API error 1, /, unauthorized user
21:20:51:015 DB save zll database items 0x00000884
21:20:51:015 DB sql exec REPLACE INTO auth (apikey, devicetype, createdate, lastusedate, useragent) VALUES ('5159C37F23', 'Phoscon#B1024x1169', '2020-10-16T21:13:22', '2020-10-16T21:20:47', 'Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.75 Safari/537.36')
21:20:51:018 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('1', 'Daylight', 'Daylight', 'PHDL00', 'Philips', '00:21:2e:ff:ff:05:fa:1a-01', '1.0', '{"dark":true,"daylight":false,"lastupdated":"2020-10-16T21:19:46.190","status":230,"sunrise":"2020-10-16T06:00:24","sunset":"2020-10-16T16:39:46"}', '{"configured":true,"lat":"51.3553","long":"6.6366","on":true,"sunriseoffset":30,"sunsetoffset":-30}', '', 'normal', '1')
21:20:51:019 DB sql exec UPDATE devices SET nwk = 0 WHERE mac = '00:21:2e:ff:ff:05:fa:1a';INSERT INTO devices (mac,nwk,timestamp) SELECT '00:21:2e:ff:ff:05:fa:1a', 0, strftime('%s','now') WHERE (SELECT changes() = 0);
21:20:51:074 DB saved in 58 ms
21:20:51:315 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
21:20:51:315 dev /dev/ttyAMA0 (/dev/ttyAMA0)
21:20:51:315 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
21:20:51:316 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
21:20:51:316 GW firmware version: 0x26660700
21:20:51:316 GW firmware version is up to date: 0x26660700
21:20:52:215 API error 1, /, unauthorized user
21:20:56:259 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
21:20:57:435 API error 1, /, unauthorized user
21:21:02:464 API error 1, /, unauthorized user
21:21:06:259 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
21:21:06:962 API error 1, /, unauthorized user
21:21:12:217 API error 1, /, unauthorized user

Mimiix, 2020-10-19:
Hi @kkillius 
It seems that it is not detected:
21:20:32:258 void zmMaster::handleStateIdle(zmMaster::MasterEvent) **not connected** goto OFF state

Also, What OS is this? Just raspbian or something else?  What Extension cable are you using? Are the other devices attached?

manup, 2020-10-19:
If the stick doesn't manage to create the network and move to status "In Network" it can be an indication that the Zigbee channel is too occupied or some heavy interference is happening.

Could you please try to change the Zigbee channel from within the Phoscon App
Also do you have any other USB devices connected to the Raspberry Pi (SSD or Bluetooth devices)?

kkillius, 2020-10-21:
@Mimiix :
> Also, What OS is this? Just raspbian or something else? 
I tried Raspian, Debian and Windows 10
> What Extension cable are you using?
0,5m and 3,0m
> Are the other devices attached?
No no other Devices and no Wifi and Bluetooth are disabled. 

@manup :
> Could you please try to change the Zigbee channel from within the Phoscon App
I couldn't change a setting. He couldn't save any changes to the stick.
> Also do you have any other USB devices connected to the Raspberry Pi (SSD or Bluetooth devices)?
No nothing.

@Mimiix and @manup :
I ordered a new stick from Amazon and sent the other one back.
With the new stick there were no problems straight away and the status "In Network" was displayed.
So it was probably a hardware defect.

Thanks for your help anyway.

closed 2020-10-21 by kkillius